vshymanskyy/TinyGSM #215: Seeeduino GPRS SIM800H

tstanescu1, 2018-10-24:
TinyGSM version: latest
Hardware: Arduino Leonardo SIM800H GPRS: http://wiki.seeedstudio.com/Seeeduino_GPRS/

### Scenario, steps to reproduce
I've been using TinyGSM with other boards but I can't seem to init the moden on the Seeeduino GPRS. I believe the issue is that on the leonardo, we use "#define SerialAT Serial1". The Arduino Leonardo board uses Serial1 to communicate via TTL (5V) serial on pins 0 (RX) and 1 (TX) (https://www.arduino.cc/reference/en/language/functions/communication/serial/) and the Seeeduino GPRS uses pins TX 8, and RX 7. I believe if there is a way to redefine the pins to 8, and 7, I would be able to connect to the GPRS module.

### Expected result
Init of SIM800H modem

### Actual result
No init, not recognizing modem, must be because of pins

Thank you all.
T

SRGDamia1, 2018-10-24:
Um.. the line ```#define SerialAT Serial1``` in the example is meant to be replaced with whatever real connection you have to the board.  Just change the Serial1 to the appropriate serial port.  So if the SIM800 Rx/Tx pins are connected to pins 7/8 then create a software serial port there and use it:  ```SoftwareSerial SerialAT(8,7);```

closed 2018-10-24 by SRGDamia1

tstanescu1, 2018-10-25:
Thanks! I thought so too. Just to let everyone know that the SEEEDUINO GPRS has an issue/bug with the D9 digital pin and must be set to HIGH, LOW, HIGH like the following:

  digitalWrite(9, HIGH);
  delay(200);
  digitalWrite(9, LOW);
  delay(2000);
  digitalWrite(9, HIGH);