vshymanskyy/TinyGSM #216: TCP connection fails

theincredibleschnigges, 2018-10-27:
TinyGSM version: 0.3.5
Hardware: SIM808, Arduino

I am trying to use the PubSubClient to connect to an MQTT broker. This worked in the past, no idea what is going wrong now. Connecting to the network and gprs seems to work fine, but connecting to the server does not (I can connect to the same address/port fine from my PC).

Here is a log with what's happening (AT command, waitResponse parameter, waitResponse result):

	[16488] ### AT: +CIPSHUT
	[16504] ### ..: OK , ERROR ,  ,  ,
	[16543] ### ..: Result index: 1
	[16588] ### AT: +CGATT=0
	[16604] ### ..: OK , ERROR ,  ,  ,
	[16643] ### ..: Result index: 1
	[16708] ### AT: +SAPBR=3,1,"Contype","GPRS"
	[16723] ### ..: OK , ERROR ,  ,  ,
	[16762] ### ..: Result index: 1
	[16837] ### AT: +SAPBR=3,1,"APN"," internet.eplus.de "
	[16855] ### ..: OK , ERROR ,  ,  ,
	[16893] ### ..: Result index: 1
	[16955] ### AT: +SAPBR=3,1,"USER"," xxxx "
	[16973] ### ..: OK , ERROR ,  ,  ,
	[17011] ### ..: Result index: 1
	[17073] ### AT: +SAPBR=3,1,"PWD"," xxxx "
	[17090] ### ..: OK , ERROR ,  ,  ,
	[17129] ### ..: Result index: 1
	[17203] ### AT: +CGDCONT=1,"IP"," internet.eplus.de "
	[17220] ### ..: OK , ERROR ,  ,  ,
	[17259] ### ..: Result index: 1
	[17307] ### AT: +CGACT=1,1
	[17323] ### ..: OK , ERROR ,  ,  ,
	[32728] ### ..: Result index: 1
	[32743] ### AT: +SAPBR=1,1
	[32743] ### ..: OK , ERROR ,  ,  ,
	[33631] ### ..: Result index: 1
	[33645] ### AT: +SAPBR=2,1
	[33646] ### ..: OK , ERROR ,  ,  ,
	[33690] ### ..: Result index: 1
	[33712] ### AT: +CGATT=1
	[33727] ### ..: OK , ERROR ,  ,  ,
	[33766] ### ..: Result index: 1
	[33812] ### AT: +CIPMUX=1
	[33828] ### ..: OK , ERROR ,  ,  ,
	[33866] ### ..: Result index: 1
	[33915] ### AT: +CIPQSEND=1
	[33931] ### ..: OK , ERROR ,  ,  ,
	[33970] ### ..: Result index: 1
	[34018] ### AT: +CIPRXGET=1
	[34034] ### ..: OK , ERROR ,  ,  ,
	[34072] ### ..: Result index: 1
	[34150] ### AT: +CSTT=" internet.eplus.de "," xxxx "," xxxx "
	[34172] ### ..: OK , ERROR ,  ,  ,
	[34210] ### ..: Result index: 1
	[34254] ### AT: +CIICR
	[34270] ### ..: OK , ERROR ,  ,  ,
	[34309] ### ..: Result index: 1
	[34356] ### AT: +CIFSR;E0
	[34371] ### ..: OK , ERROR ,  ,  ,
	[34412] ### ..: Result index: 1
	[34476] ### AT: +CDNSCFG="8.8.8.8","8.8.4.4"
	[34491] ### ..: OK , ERROR ,  ,  ,
	[34530] ### ..: Result index: 1
	[34617] ### AT: +CIPCLOSE= 1
	[34634] ### ..: OK , ERROR ,  ,  ,
	[34673] ### ..: Result index: 2
	[34719] ### AT: +CIPSSL= 0
	[34736] ### ..: OK , ERROR ,  ,  ,
	[34775] ### ..: Result index: 2
	[34856] ### AT: +CIPSTART= 1 , "TCP "," m14.cloudmqtt.com ", 13855
	[34879] ### ..: CONNECT OK , CONNECT FAIL , ALREADY CONNECT , ERROR , CLOSE OK
	[48896] ### ..: Result index: 2

I don't know why I get a CONNECT FAIL with CIPSTART at the end. Also not sure why CIPCLOSE and CIPSSL result in ERROR.

theincredibleschnigges, 2018-10-27:
I tried the AT commands manually, and if I leave out the connection number, it works:

AT+CIPSTART="TCP","m14.cloudmqtt.com",13855 instead of 
AT+CIPSTART=**1,**"TCP","m14.cloudmqtt.com",13855

curlyz, 2019-01-12:
@theincredibleschnigges 
you use AT+CIPMUX=1 which tell the SIM800 that you will be multitask with 6 socket at the same time , if declared like this , every command related to socket activity must have the index number so that SIM800 know which socket are you talking about 
Socket is defined as 0 to 6

In your case , you did use CIPMUX = 1 , therefore , CIPSTART=1 (must have "1" or any number in (0-6))
If you only use 1 socket of the SIM800 , use CIPMUX = 0 , then , you doesnt need to provide the socket number

theincredibleschnigges, 2019-02-27:
This is not what I am doing, it is what the library is doing. In TinyGsmClientSIM800.h in bool modemConnect(...):
sendAT(GF("+CIPSTART="), mux, ',', GF("\"TCP"), GF("\",\""), host, GF("\","), port);

And "mux" seems to be 1.

theincredibleschnigges, 2019-02-27:
I ran the sketch to do a factory reset to the modem (TinyGSM/tools/FactoryReset) but it did not change anything.

theincredibleschnigges, 2019-02-27:
This is the sketch I am trying to run:

	#include <DebugUtils.h>
	#include <StringUtils.h>

	#include <PubSubClient.h>

	#define TINY_GSM_MODEM_SIM868
	#include <TinyGsmClient.h>

	#include <Secrets.h>

	#define SerialAT Serial1

	TinyGsm gsmModem(SerialAT);
	TinyGsmClient gsmClient(gsmModem, 0);
	PubSubClient mqtt;

	const char apn[]  = "internet.eplus.de";

	const unsigned long mqttReconnectIntervalInMs = 5 * 1000UL;
	unsigned long lastMqttReconnectAttemptInMs = 0;

	const char* mqttBroker = "m14.cloudmqtt.com";
	const uint16_t mqttBrokerPort = 13855;
	const char* mqtt_clientId = "carduinoBackend";

	// void handleIncomingMessage(char* topic, byte* payload, unsigned int length);

	void initialize() {
	  mqtt = PubSubClient(gsmClient);
	}

	void setup() {
	  initialize();

	  Serial.begin(115200);
	  delay(10);

	  SerialAT.begin(9600);
	  delay(3000);

	  if (!gsmModem.restart()) {
		DEBUG_PRINTLN("Could not initialize modem.");
		return;
	  }
	  
	  String modemInfo = gsmModem.getModemInfo();
	  DEBUG_PRINT("Modem: ");
	  DEBUG_PRINTLN(modemInfo);
	  
	  mqtt.setServer(mqttBroker, mqttBrokerPort);
	  // mqtt.setCallback(handleIncomingMessage);
	}

	bool connectMqtt() {
	  DEBUG_PRINT("Connecting to ");
	  DEBUG_PRINT(mqttBroker);
	  DEBUG_PRINT(":");
	  DEBUG_PRINT(mqttBrokerPort);  
	  if (!mqtt.connect(mqtt_clientId, Mqtt_user, Mqtt_password)) {
		DEBUG_PRINT(" failed, rc=");
		DEBUG_PRINTLN(mqtt.state());
		return false;
	  }
	  DEBUG_PRINTLN(" OK");
	  return mqtt.connected();
	}

	bool retryConnectMqtt() {
	  if (mqtt.connected()) {
		return true;
	  }
	  else {
		if ((millis() - lastMqttReconnectAttemptInMs) > mqttReconnectIntervalInMs) {
		  if (connectMqtt()) {
			lastMqttReconnectAttemptInMs = 0;
			return true;
		  }
		  else {
			lastMqttReconnectAttemptInMs = millis();
		  }
		}
	  }
	  return false;
	}

	bool connectModem() {
	  if (!gsmModem.isNetworkConnected()) { 
		DEBUG_PRINTLN("Trying to connect to network and GPRS");
		if (gsmModem.getSimStatus() == SIM_LOCKED) {
		  if (!gsmModem.simUnlock(SimPin)) {
			DEBUG_PRINTLN("Could not unlock SIM");
			return false;
		  }
		}

		if (!gsmModem.waitForNetwork()) {
		  DEBUG_PRINTLN("Could not connect to network");
		  return false;
		}
	  }

	  if (!gsmModem.isGprsConnected()) {
		if (!gsmModem.gprsConnect(apn, Apn_user, Apn_pass)) {
		  DEBUG_PRINTLN("Could not connect GPRS");
		  return false;
		}
	  }
	  
	  return true;
	}

	void loop() {
	  if (connectModem() && retryConnectMqtt()) {
		DEBUG_PRINT("Starting loop after connect.");
		mqtt.loop();
	  }
	}

theincredibleschnigges, 2019-02-27:
Here's the output if a pass a StreamDebugger to TinyGsm (mixed with some of my own debug output):

	AT
	AT
	OK
	AT+CLTS=1
	AT+CLTS=1
	OK
	AT&W
	AT&W
	OK
	AT+CFUN=0
	AT+CFUN=0
	+CPIN: SIM PIN

	+CPIN: NOT READY

	OK
	AT+CFUN=1,1
	AT+CFUN=1,1
	OK
	AT
	AT
	OK
	AT&FZ
	ATE0Z
	ATE0
	OK
	AT+CPMS="SM","SM","SM"

	ERROR
	AT+CPIN?

	+CPIN: SIM PIN

	+CPIN: SIM PIN

	OK
	ATI

	SIM808 R14.18

	OK
	Modem: SIM808 R14.18
	AT+CREG?

	+CREG: 0,2

	OK
	Trying to connect to network and GPRS
	AT+CPIN?

	+CPIN: SIM PIN

	OK
	AT+CPIN="6414"

	OK
	AT+CREG?

	+CREG: 0,2

	OK
	AT+CREG?

	+CPIN: READY

	+CREG: 0,2

	...

	OK
	AT+CREG?

	+CREG: 0,1

	OK
	AT+CGATT?

	+CGATT: 0

	OK
	AT+CIPSHUT

	SHUT OK
	AT+CGATT=0

	OK
	AT+SAPBR=3,1,"Contype","GPRS"

	OK
	AT+SAPBR=3,1,"APN","internet.eplus.de"

	OK
	AT+SAPBR=3,1,"USER","blau"

	OK
	AT+SAPBR=3,1,"PWD","blau"

	OK
	AT+CGDCONT=1,"IP","internet.eplus.de"

	OK
	AT+CGACT=1,1

	OK
	AT+SAPBR=1,1

	OK
	AT+SAPBR=2,1

	+SAPBR: 1,1,"10.214.180.163"

	OK
	AT+CGATT=1

	OK
	AT+CIPMUX=1

	OK
	AT+CIPQSEND=1

	OK
	AT+CIPRXGET=1

	OK
	AT+CSTT="internet.eplus.de","blau","blau"

	OK
	AT+CIICR

	OK
	AT+CIFSR;E0

	10.214.180.163

	OK
	AT+CDNSCFG="8.8.8.8","8.8.4.4"

	OK
	Connecting to m14.cloudmqtt.com:13855
	AT+CIPCLOSE=1

	ERROR
	AT+CIPSSL=0

	ERROR
	AT+CIPSTART=1,"TCP","m14.cloudmqtt.com",13855

	OK

	1, CONNECT FAIL
	 failed, rc=-2
	AT+CREG?

	+CREG: 0,1

	OK
	AT+CGATT?

	+CGATT: 1

	OK
	AT+CIFSR;E0

	10.215.6.68

	OK
	AT+CREG?

	+CREG: 0,1

	OK

	...

theincredibleschnigges, 2019-03-03:
Sorry, never mind, I had an issue with my phone provider.

closed 2019-03-03 by theincredibleschnigges

kamrulroky, 2019-07-02:
@theincredibleschnigges   can you elaborate what was the issue with your provider? I am facing this issue with one particular operator even though others work nicely, they couldn't give me any information. my finding is both tinygsm and pubsubclient are fine , but couldn't figure out what is causing problem with that provider

mihai-lazar, 2020-05-10:
Hi, im having the same problem, before it worked and now I get the exact same results as you! What did you do?

genotix, 2021-01-21:
Hmm I seem to have the same issue here.
Would like to know what was up and how it was fixed.

jakobbraun, 2021-04-18:
I would also be interested...

theincredibleschnigges, 2021-04-18:
Sorry for the late reply, if I remember correctly, it was simply that I had a prepaid data plan and I could no longer use it because you had to put in more money each month (even if you didn’t use up your existing credit), because otherwise your plan gets sort of put in sleep mode.

jakobbraun, 2021-04-18:
Thanks a lot for the reply!