vshymanskyy/TinyGSM #220: Can`t compile for ESP8266 (Nodemcu) as main board

saloid, 2018-11-15:
TinyGSM version: 0.3.5
ESP SDK: actual git clone from https://github.com/esp8266/Arduino

Hardware: Main board is ESP8266 (Nodemcu) with SIM800 connected to it.

### Scenario, steps to reproduce
Try to compile any example sketch with TinyGsmClient for Nodemcu board

### Expected result
Compilation success
### Actual result
Compilation error:
```
exit status 1
cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClient'
```

saloid, 2018-11-16:
Fixed it here: #221

vshymanskyy, 2019-02-06:
I think this is an issue, introduced in ESP8266 core 2.5.0.
Need to investigate it further

laxman-spidey, 2019-02-18:
I'm facing the same issue, is this solved yet? Does it work if i downgrade ESP8266 core to <2.5.0

saloid, 2019-02-18:
Yes, look at my fix at the top

SRGDamia1, 2019-03-12:
I think the correct solution will be to wait for the ESP8266 core to fix the extra virtual functions it introduced:
https://github.com/esp8266/Arduino/issues/5749

emi1988, 2019-03-17:
> I'm facing the same issue, is this solved yet? Does it work if i downgrade ESP8266 core to <2.5.0

Yes, for me it works if I downgrade the ESP8266 core to Version 2.4.2

d-a-v, 2019-04-09:
esp8266/Arduino#5969 adresses it

hardmax, 2019-04-21:
Hi, the same problem with ESP32, I have to downgrade to 1.0.1 to works...

SRGDamia1, 2019-05-22:
The compiler problems with the ESP32 and ESP8266 should now be fixed.

closed 2019-05-22 by SRGDamia1