vshymanskyy/TinyGSM #224: Having trouble adding support for Quectel MC20

reydelleon, 2018-11-21:
Hi there,

I've been trying to modify TinyGSM to work with Quectel MC20 GNSS/GPRS module for a few days now. I've updated every single AT command used in the library to use the command set specified in the manual for the module. I got it to where I can get the module connected to the GSM network and I can open a connection and send data, but I'm having trouble in the reading of the data. 

I made the modification because I'm trying to connect to an MQTT server with the [arduino-mqtt](https://github.com/256dpi/arduino-mqtt) library. I'm currently having an issue were the MQTT library will send a connect package and the MQTT server will recognize it and send a CONNACK back, but the MQTT library is not been able to read that package, which triggers a reconnection. 

At this point, I just want to know if you, @vshymanskyy have the time and the ability to help me out by reviewing the code and pointing out possible failure points. I'm willing to do all the work. Most of it is done I think, which is translate the AT commands, but I don't have the overall knowledge you have to understand how it all fits together and why certain parts might be failing. 

If you are not able to help I will understand and try something else. Just wanted to offer my help to get support for the MC20 added, since I'm working on it anyway.

Let me know your thoughts.

reydelleon, 2018-11-21:
Hey!

To confirm my suspicions and to provide more context, I went ahead and run the `Diagnostics` script. It got stuck in this section...

```
// Wait for data to arrive
  while (client.connected() && !client.available()) {
    delay(100);
    SerialMon.print('.');
```

Here is the log from the Serial Monitor (removed some repeated outputs for space sake)...

```
AT
AT
AT
AT
AT
AT 
Initializing modem...AT

OK
AT+CFUN=1,1

OK
AT

OK
AT+IPR=0&W

OK
AT+QIFGCNT=0

OK
AT+QICSGP=1,"wireless.twilio.com","",""

OK
AT+QIMUX=1

OK
AT+QIDNSIP=1
 [OK]
ATI

OK
Modem:
Waiting for network...AT+CREG?

Quectel_Ltd
Quectel_MC20
Revision: MC20CAR01A08

OK

+⸮⸮[24453] ### Unhandled: Quectel_Ltd
Quectel_MC20
Revision: MC20CAR01A08

OK

+⸮⸮
AT+CREG?

+CREG: 0,2

OK

Call Ready

+CREG: 0,2

OK

SMS Ready

+CREG: 0,2

OK

AT+CREG?

+CREG: 0,1

OK
 [OK]
Connecting to wireless.twilio.comAT+QIDEACT

DEACT OKAT+CPIN?

+CPIN: READY

OK
AT+CREG?

+CREG: 0,1

OK
AT+CGREG?

+CGREG: 0,2

OK

AT+CGREG?

+CGREG: 0,2

OK
AT+CGREG?

+CGREG: 0,1

OK
AT+QIREGAPP

OK
AT+QIACT

OK
AT+QILOCIP

172.50.25.44
 [OK]
AT+CGPADDR=1

+CGPADDR: 1

OK
Local IP: 0.0.0.0
Connecting to vsh.pp.uaAT+QIOPEN=1,"TCP","vsh.pp.ua",80

OK

1, CONNECT OK [OK]

AT+QISEND=1,32

>GET /TinyGSM/logo.txt HTTP/1.0

SEND OK
AT+QISEND=1,17

>Host: vsh.pp.ua

SEND OK
AT+QISEND=1,21

>Connection: close

SEND OK

+RECEIVE: 1, 633
.HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Wed, 21 N[49514] ### Unhandled: HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Wed, 21 N
.............................................................................................................................................
```

It gets stuck in receiving that data. I waited for a long time, but nothing.

SRGDamia1, 2018-11-26:
Have you tried increasing your rx buffer size?

reydelleon, 2018-11-26:
I fact, I got it to work altogether. There was a combination of issues, including a timeout for reads that was too short and some needed updates to the `waitResponse()` to fit the MC20 command set. I also had to set the reads to go to the buffer instead of directly into UART, etc. 

Now I'm trying to add support for SSL.

SRGDamia1, 2018-11-27:
Oh, great!

schmeckm, 2020-01-10:
Can you share the code or library for MC20?

reydelleon, 2020-01-12:
@schmeckm You should be able to access here: https://github.com/reydelleon/TinyGSM

I have since moved on from this code base, so I won't be able to answer any questions about the code. Hope it helps.

piyush9999, 2022-09-21:
Hello @SRGDamia1 can you please help me to know how to send websocket headers, I'm using tinygsm library for BG96. TIA