vshymanskyy/TinyGSM #226: TinyGSM SIM800L Modem stops working.

Bambofy, 2018-11-22:
TinyGSM version: latest
Hardware: arduino pro mini, SIM800L

### Scenario, steps to reproduce
Try to send a file of data in chunks to a web server.

### Expected result
Expected consistent packet requests to the web server containing the data from a file.

### Actual result
After a while the module breaks and cannot connect to the network?

### Debug output
```
Initializing modem...
Modem: SIM800 R14.18
20:47:00.054 -> Waiting for network... OK
20:47:06.323 -> Connecting to giffgaff.com OK
initialization done.
20:47:14.910 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:16.392 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:19.126 -> Connecting to 31.7.184.861 OK
Server disconnected
Connecting to 31.7.184.861 OK
Server disconnected
20:47:24.852 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:26.784 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:29.131 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:31.666 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:34.278 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:36.682 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:38.960 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:41.190 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:43.457 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:45.794 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:47.954 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:50.262 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:53.079 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:55.717 -> Connecting to 31.7.184.861 OK
Server disconnected
20:47:58.187 -> Connecting to 31.7.184.861 OK
Server disconnected
20:48:01.471 -> Connecting to 31.7.184.861 OK
Server disconnected
20:48:04.045 -> Connecting to 31.7.184.861 OK
Server disconnected
20:48:06.243 -> Connecting to 31.7.184.861 OK
Server disconnected
20:48:08.655 -> Connecting to 31.7.184.861 OK
Server disconnected
20:48:10.827 -> Connecting to 31.7.184.861 OK
Server disconnected
20:48:13.099 -> Connecting to 31.7.184.861 OK
Server disconnected
20:48:15.374 -> Connecting to 31.7.184.861 OK
[... 233 lines cut ...]
20:53:19.961 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:22.301 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:24.671 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:26.912 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:29.319 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:31.556 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:33.757 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:36.089 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:43.658 -> Connecting to 31.7.184.861 OK
Server disconnected
20:53:45.347 -> Connecting to 31.7.184.86 fail
Initializing modem...
Modem: 
20:55:23.408 -> Waiting for network... fail
Could not connect to the network, retrying...
Waiting for network...
```

jeancaffou, 2019-04-15:
This happens to me every time after I use `client.connect();` and the server does not close the connection.
It will break the modem and it will not even connect to the network after that, until you turn off the power to reset.

Described in #265

SRGDamia1, 2020-04-15:
I'm going to close this older issue.  Please re-open it if you're still having problems.

closed 2020-04-15 by SRGDamia1