vshymanskyy/TinyGSM #232: mqtt.connected(); flag update issue

mayurharge, 2019-01-02:
I am using SIM800 module.
mqtt.connected(); is returning true value while mqtt is disconnected in this case

> ERROR
[299152] I m alive
[299399] MQTT inLoop
[299399] 265
[299399] MQTT in conditional Loop
AT+CIPCLOSE=1
ERROR
[299418] I m alive
[299665] MQTT inLoop
[299665] 265
[299665] MQTT in conditional Loop
AT+CIPCLOSE=1
ERROR

When modem responds `ERROR ` for `AT+CIPCLOSE=1` or `AT+CIPSTATUS=1`, `mqtt.connected();` returns `true` while mqtt is disconnected.
here is the code for above response
`void mqttLoop(){
  DBG("MQTT inLoop");
  DBG(millis()-MQTTconnectedMillis);
  if(millis()-MQTTconnectedMillis>700000){
    DBG("SYS RESET");
    resetModule();
    DBG("This should not get printed");
  }  
  if(millis()-MQTTconnectedMillis>240000){
    if(!Init){
      Init=true;
      Modem_init(); 
    }
  }
  if(!GPRSConnected()) return;
  if(mqttConnected()){
    DBG("MQTT in conditional Loop");
    MQTTconnectedMillis=millis();
    Init=false;
      mqtt.loop();
  }else{
    _mqttConnect();
  }
  if(millis()-HbeatMillis>HEARTBEAT_INTERVAL){
    HbeatMillis=millis();
    Hbeat();
  }
}`
Thanks,
Mayur

SRGDamia1, 2019-03-12:
After sending ```AT+CIPCLOSE``` to the SIM800 the socket is flagged as closed regardless of the response (OK/ERROR/??).  That 'closed' flag could be ignored if there's anything left over in the Rx buffer - either the TinyGSM Rx buffer __OR__ the SIM800's internal Rx buffer.  You have to read everything out of both buffers before the library really knows a socket is closed.  It's a little clunky because the SIM800 doesn't always "remember" to notify about incoming data.

I have a fix for the stop on the EnviroDIY fork which I'm polishing up a bit - https://github.com/vshymanskyy/TinyGSM/pull/205

SRGDamia1, 2020-04-15:
This should be fixed.

closed 2020-04-15 by SRGDamia1