vshymanskyy/TinyGSM #239: Goouuu Tech IOT-GA6

friistf, 2019-01-29:
TinyGSM version: 
Hardware: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### AT command log

friistf, 2019-01-29:
Hi,
I know this was the subject of a now closed subject, but I am new to Goouuu Tech IOT-GA6 (just bought one) and the  earlier treatment left me confused. Is there now a conclusion?

I am trying the "AT" command as follows:

[Code]
#PICAXE 14M2
#Terminal 19200

Symbol TlfSet           = B.4                   'RTS pin on module
Symbol TXpin            = B.3                   'to U_RXD on module (there is also a H_RXD (which gives "No response"))
Symbol RXpin            = B.2                   'to U_TXD on module (there is also a U_TXD (which gives "No response"))

Symbol TlfBaudRate     = t9600_16               'Tlf Baud rate

pause 1000

PowerOnReset:
	High TXpin
	SetFreq M16
	SerTxd( Cr, LF, "Started PowerOnReset", CR, LF )
	pause 4000
	
MAIN:
 	
	    	b1 ="?" : b2 = "?" : b3 = "?" : b4 = "?"
		b5 ="?" : b6 = "?" : b7 = "?" : b8 = "?"
		
		SerOut TXpin, TlfBaudRate, ( "AT" )
			    
		Serin [8000], RXpin, TlfBaudrate, b1,b2,b3,b4,b5,b6,b7,b8
		    
		if b1 = "?" OR b1 = "E" Then
			SerTxd( "No response", CR, LF )
		Else
		      SerTxd( "R:", b1,b2,b3,b4,b5,b6,b7,b8, CR, LF )
		End if
		
		pause 4000
		goto MAIN
[/Code]

I have the red light (pwr) on and the blue light flashing every 3 sec. I have connected to U_RXD and U_TXD, but I get "No response" when sending "AT".
Can anyone help?
best regards
torben

friistf, 2019-01-29:
Hi,
I think I should'nt have closed.
torben

HebaEbrahem, 2019-02-28:
Heeeeey , Does anyone tried to communicate with this module (GOOUUU IOT-GA6) using" AT-command tester" or any terminal software , and found a response ? I'm really struggling with it
-

friistf, 2019-03-01:
Hi HebaEbrahem,
I did. I could not make it work. I got diff. results from test to test with
the same pgm. Maybe I was clumsy - I dont know.
torben

friistf, 2019-03-01:
Hi HebaEbrahem,
You should check your connections and note that the freq. at which you
communicate with the Goouuu Tech IOT-GA6 is 115.200 baud.
Let me know if you get results - I want to try again.
torben

vshymanskyy, 2019-03-06:
Please use the existing ticket which contains lots of info alrready: https://github.com/vshymanskyy/TinyGSM/issues/164

closed 2019-03-06 by vshymanskyy