vshymanskyy/TinyGSM #242: AT+CMGF=1

friistf, 2019-02-13:
<Hi,
I am using Goouuu Tech IOT-GA6.
I have issued the following commands:

Command             Reply
----------------------------------
AT&F                     OK
AT                          OK
AT-CMGF=1          ERROR

Can anyone explain why the third command gives an error?

I have tried AT+CSCS "GMS" which works, but AT-CMGF=1  stills gives an error.
Likewise AT+CMEE gives an error.
Best regards
torben

eabase, 2019-07-17:
There is no AT "minus" SOMETHING. 
i think you can close this non issue

SRGDamia1, 2020-04-15:
Are you still having problems with this?  I'm going to close this older issue.  Please re-open it if you're still having problems.

closed 2020-04-15 by SRGDamia1

Mich2899, 2021-12-02:
I am using Sim800C hat for raspberry pi. However, the image is not raspbian, it is buildroot and i am trying to send a text message using it. Commands and their response:
AT
OK
ATI
SIM800 R14.18
OK
AT+CMGF=1
ERROR
How can i solve this?
On raspbian with the same sim card i could send sms but on buildroot it gives error

kemploe, 2025-04-05:
@Mich2899 

I occasionally have the exact same error on SIM900A as well