vshymanskyy/TinyGSM #243: SEnding SMS

friistf, 2019-02-21:
Hi,
I am using Picaxe and Goouuu Tech IOT-GA6, which I bought 2 weeks ago.

I have issued the commands "AT", "AT+CMEE=1", "AT+CMGF=1","AT+CPIN=1234" - all OK.
I have issued "AT+CMGS=+458116757" and > is echoed back. But should'nt there be more?
I have next issued ("Bla, bla")$26 (where $26 = hex 1A) and I get back:

Bla, bla&

Here the message should have been sent, but nothing is received.

Can anyone help?
best regards
torben

friistf, 2019-02-21:
Hi,
It should have been ("Bla, bla",$1A), and with that I get the error code 500 - ie. Unknown error
torben

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1