vshymanskyy/TinyGSM #244: SARA-R4 APN set fix.

ldab, 2019-02-25:
https://github.com/vshymanskyy/TinyGSM/blob/477b4f2d1b3fa8b6c3bc00b47f6592663a83fd72/src/TinyGsmClientUBLOX.h#L455

[image]

[image]

I have been using the following to set the APN

AT+CGDCONT=1,"IP","APN_name""

SRGDamia1, 2019-02-25:
The SARA R4/N4 modules don't support any of the UPSx commands.  I've changed that and made a few other tells for those modules here: https://github.com/EnviroDIY/TinyGSM/blob/master/src/TinyGsmClientUBLOX.h

ldab, 2019-03-02:
Well.. that's what I said, why don't you push the changes and merge?

dansass, 2019-03-16:
Hi idab,
I have the sparkfun 4g shield and i am having a hard time getting it connected to blynk... 

hello im having trouble getting this shield connected to blynk. I uploaded and successfully connected the shield to hologram using the 00_Register_Operator sketch. On tinyGSM wiki it says that my modem is supported. When i enable blynk debug this is what pops up on my serial monitor

Initializing modem...
[8005] ### Unhandled: Manufacturer: u-blox
Model: SARA-R410M-02B
Revision: L0.0.00.00.05.06 [Feb 03 2018 13:00:41
V
M:55963
O
Modem:
[8083]
___ __ __
/ _ )/ /_ _____ / /__
/ _ / / // / _ / '/
///_, /////_
/__/ v0.6.1 on Arduino Uno

[8242] Modem init...
[8412] ### Modem: u-blox SARA-R410M-02B
[8550] Connecting to network...
[8659] Network: 302 720 Hologram
[8660] Connecting to hologram ...
[8744] ### Unhandled: Operation not allowed
[8810] Connecting to blynk-cloud.com:80
[8888] ### Unhandled: Operation not allowed
[13794] Connecting to blynk-cloud.com:80
[13873] ### Unhandled: Operation not allowed

Have u had any luck getting the thing connected??

Thanks, Daniel

SRGDamia1, 2019-03-16:
Can you post the AT log?
```cpp
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
```

dansass, 2019-03-16:
No problem!

Initializing modem...
ATI

Manufacturer: u-blox
Model: SARA-R410M-02B
Revision: L0.0.00.00.05.06 [Feb 03 2018 13:00:]S:2II55963
O
[8005] ### Unhandled: Manufacturer: u-blox
Model: SARA-R410M-02B
Revision: L0.0.00.00.05.06 [Feb 03 2018 13:00:]S:2II55963
O
Modem: 
[8081] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.6.1 on Arduino Uno

[8240] Modem init...
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+CGMI

u-blox

OK
AT+GMM

SARA-R410M-02B

OK
[8418] ### Modem: u-blox SARA-R410M-02B
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
[8558] Connecting to network...
AT+CEREG?

+CEREG: 0,5

OK
AT+COPS?

+COPS: 1,0,"302 720 Hologram",8

OK
[8669] Network: 302 720 Hologram
[8669] Connecting to hologram ...
AT+USOCL=0,1

+CME ERROR: Operation not allowed

[8778] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[8866] Connecting to blynk-cloud.com:80
AT+USOCR=6
 Operation not allowed

+CME ERROR: Operation not allowed

[8956] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[13829] Connecting to blynk-cloud.com:80
AT+USOCR=6
 Operation not allowed

+CME ERROR: Operation not allowed

[13906] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[18830] Connecting to blynk-cloud.com:80
AT+USOCR=6
 Operation not allowed

+CME ERROR: Operation not allowed

[18908] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[23829] Connecting to blynk-cloud.com:80
AT+USOCR=6
 Operation not allowed

+CME ERROR: Operation not allowed

[23909] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR: Operation not allowed

[28782] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[28848] Connecting to blynk-cloud.com:80
AT+USOCR=6
 Operation not allowed

+CME ERROR: Operation not allowed

[28931] ### Unhandled: Operation not allowed
AT+USOCL=0,1

ldab, 2019-03-16:
your +COPS is set to manual, did you change something on your modem before using this lib?

dansass, 2019-03-16:
Not that i know of. Im quite a newbie when it comes to at commands (i did a couple times in the past) but i can try to change it. Could u tell how i would do that?

ldab, 2019-03-16:
I believe your modem is not connected (missing +CGACT and +CGDCONT). please double check your Lib version as some extra features have been added by @SRGDamia1 recently.

ldab, 2019-03-16:
also make sure you have ```#define TINY_GSM_MODEM_UBLOX``` on top of your main code.

I'm closing the issue as it has been resolved. If you still facing some problems please open a new one.

closed 2019-03-16 by ldab