vshymanskyy/TinyGSM #245: TinyGSM for SIM5320

Najab, 2019-02-27:
Is there any library for SIM5320 3G GPS/GSM Module?
Please provide or help me for using SIM5320 with library.

HebaEbrahem, 2019-02-28:
I didn't encounter one , but you can check fro SIM900h,SIM800 they do have libraries in Arduino , and you can see the functions and apply them .
and you can always use "Serial" library to communicate with whatever model.

Najab, 2019-03-04:
I'm successfully done with SIM900h,SIM800 with TinyGSM library.
But now i want to use SIM5320 with Library.

Najab, 2019-03-06:
bool gprsConnect(const char* apn, const char* user = NULL, const char* pwd = NULL) {
    gprsDisconnect();

    // Set the Bearer for the IP
    sendAT(GF("+SAPBR=3,1,\"Contype\",\"GPRS\""));  // Set the connection type to GPRS
    waitResponse();

    sendAT(GF("+SAPBR=3,1,\"APN\",\""), apn, '"');  // Set the APN
    waitResponse();

    if (user && strlen(user) > 0) {
      sendAT(GF("+SAPBR=3,1,\"USER\",\""), user, '"');  // Set the user name
      waitResponse();
    }
    if (pwd && strlen(pwd) > 0) {
      sendAT(GF("+SAPBR=3,1,\"PWD\",\""), pwd, '"');  // Set the password
      waitResponse();
    }

    // Define the PDP context
    sendAT(GF("+CGDCONT=1,\"IP\",\""), apn, '"');
    waitResponse();

    // Activate the PDP context
    sendAT(GF("+CGACT=1,1"));
    waitResponse(60000L);

    // Open the definied GPRS bearer context
    sendAT(GF("+SAPBR=1,1"));
    waitResponse(85000L);
    // Query the GPRS bearer context status
    sendAT(GF("+SAPBR=2,1"));
    if (waitResponse(30000L) != 1)
      return false;

    // Attach to GPRS
    sendAT(GF("+CGATT=1"));
    if (waitResponse(60000L) != 1)
      return false;

    // TODO: wait AT+CGATT?

    // Set to multi-IP
    sendAT(GF("+CIPMUX=1"));
    if (waitResponse() != 1) {
      return false;
    }

    // Put in "quick send" mode (thus no extra "Send OK")
    sendAT(GF("+CIPQSEND=1"));
    if (waitResponse() != 1) {
      return false;
    }

    // Set to get data manually
    sendAT(GF("+CIPRXGET=1"));
    if (waitResponse() != 1) {
      return false;
    }

    // Start Task and Set APN, USER NAME, PASSWORD
    sendAT(GF("+CSTT=\""), apn, GF("\",\""), user, GF("\",\""), pwd, GF("\""));
    if (waitResponse(60000L) != 1) {
      return false;
    }

    // Bring Up Wireless Connection with GPRS or CSD
    sendAT(GF("+CIICR"));
    if (waitResponse(60000L) != 1) {
      return false;
    }

    // Get Local IP Address, only assigned after connection
    sendAT(GF("+CIFSR;E0"));
    if (waitResponse(10000L) != 1) {
      return false;
    }

    // Configure Domain Name Server (DNS)
    sendAT(GF("+CDNSCFG=\"8.8.8.8\",\"8.8.4.4\""));
    if (waitResponse() != 1) {
      return false;
    }

    return true;
  }

I just want the replace this function for SIM5320 but i tried something please check and correct it now my function is not working 

bool gprsConnect(const char* apn, const char* user = NULL, const char* pwd = NULL) {
    //gprsDisconnect();

    //// Set the Bearer for the IP
    //sendAT(GF("+SAPBR=3,1,\"Contype\",\"GPRS\""));  // Set the connection type to GPRS
    //waitResponse();

    //sendAT(GF("+SAPBR=3,1,\"APN\",\""), apn, '"');  // Set the APN
    //waitResponse();

    //if (user && strlen(user) > 0) {
    //  sendAT(GF("+SAPBR=3,1,\"USER\",\""), user, '"');  // Set the user name
    //  waitResponse();
    //}
    //if (pwd && strlen(pwd) > 0) {
    //  sendAT(GF("+SAPBR=3,1,\"PWD\",\""), pwd, '"');  // Set the password
    //  waitResponse();
    //}

	sendAT(GF("+CGATT=1"));  // Set the connection type to GPRS
	waitResponse();

	sendAT(GF("+CGACT=1,1"));  // Set the connection type to GPRS
	waitResponse();

	//sendAT(GF("+CGATT=1"));  // Set the connection type to GPRS
	//waitResponse();

    // Define the PDP context
	sendAT(GF("+CGDCONT=1,\"IP\",\""), apn, GF("\",\"0.0.0.0\""));
    //sendAT(GF("+CGDCONT=1,\"IP\",\""+String(apn)+"\",\"0.0.0.0\""));
    waitResponse();

    // Activate the PDP context
    sendAT(GF("+CGACT=1,1"));
    waitResponse(60000L);

    //// Open the definied GPRS bearer context
    //sendAT(GF("+SAPBR=1,1"));
    //waitResponse(85000L);
    //// Query the GPRS bearer context status
    //sendAT(GF("+SAPBR=2,1"));
    //if (waitResponse(30000L) != 1)
    //  return false;

	sendAT(GF("+CGPADDR=1"));
	if (waitResponse(30000L) != 1)
		return false;

    // Attach to GPRS
    sendAT(GF("+CGATT=1"));
    if (waitResponse(60000L) != 1)
      return false;

    // TODO: wait AT+CGATT?

    //// Set to multi-IP
    //sendAT(GF("+CIPMUX=1"));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    //// Put in "quick send" mode (thus no extra "Send OK")
    //sendAT(GF("+CIPQSEND=1"));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    //// Set to get data manually
    //sendAT(GF("+CIPRXGET=1"));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    //// Start Task and Set APN, USER NAME, PASSWORD
    //sendAT(GF("+CSTT=\""), apn, GF("\",\""), user, GF("\",\""), pwd, GF("\""));
    //if (waitResponse(60000L) != 1) {
    //  return false;
    //}

    //// Bring Up Wireless Connection with GPRS or CSD
    //sendAT(GF("+CIICR"));
    //if (waitResponse(60000L) != 1) {
    //  return false;
    //}

    //// Get Local IP Address, only assigned after connection
    //sendAT(GF("+CIFSR;E0"));
    //if (waitResponse(10000L) != 1) {
    //  return false;
    //}

    //// Configure Domain Name Server (DNS)
    //sendAT(GF("+CDNSCFG=\"8.8.8.8\",\"8.8.4.4\""));
    //if (waitResponse() != 1) {
    //  return false;
    //}

    return true;
  }

SRGDamia1, 2019-03-06:
Have you read the AT command manual and TCP-IP application notes?

Najab, 2019-03-08:
Yes, but i don't understand which commands executes sequential :(

eabase, 2019-07-09:
Please enclose you code within triple back ticks and `c` for c-code, like this:  **```c**.

You need to find the AT manual to make sure you're doing the right command in the right order.

eabase, 2019-07-09:
See #189. Should be similar to yours I think.

SRGDamia1, 2019-08-02:
@najab - this should be working now

closed 2020-04-15 by SRGDamia1