vshymanskyy/TinyGSM #247: Unhandled in A6

oleslavs, 2019-03-04:
Hello! 
When try to get SIM CCD wrotes this:

[106179] ### Unhandled: +CCID:89701015388545979303

OK
[106179] CCID: 

It hard to connect GPRS:

[20623] Operator: MTS RUS
[20623] Connecting to internet.mts.ru
[80701] ### Unhandled: COMMAND NO RESPONSE!
[90701] Initializing modem...
[98399] ### Unhandled: +CME ERROR:SIM not inserted
[100400] ### Unhandled: +CME ERROR:SIM not inserted

[106887] Operator: MTS RUS
[106887] Connecting to internet.mts.ru
[166889] ### Unhandled: +CME ERROR:Excute command failure
[226890] ### Unhandled: COMMAND NO RESPONSE!

But sometime it success.

[267148] Operator: MTS RUS
[267149] Connecting to internet.mts.ru
[327151] ### Unhandled: AT+CIPSHUT

+CME ERROR:50
[333387] GPRS status: connected
[333449] Local IP: 10.133.100.202
[333508] Signal quality: 23
[333568] Battery lavel: 81

After all, I`m don`t understand following:

[401302] Poweroff.
[414775] ### Unhandled: ^CINIT: 1, 0, 0
[415026] ### Unhandled: ^CINIT: 2, 32, 41891
[416081] ### Unhandled: ^STN: 38

^CINIT: 4, 8192, 38
[416874] ### Unhandled: ^CINIT: 8, 2048, 1
[417195] ### Unhandled: ^CINIT: 16, 0, 1638405
[417265] ### Unhandled: ^CINIT: 32, 0, 0

+CREG: 0
[418923] ### Unhandled: +CTZV:19/03/04,19:30:46,+03
[419761] ### Unhandled: +CIEV: service,  1
+CIEV: roam, 0

+CREG: 1

TinyGSM version: 
Hardware: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### AT command log

SRGDamia1, 2019-05-07:
I don't quite understand the problem.  It seems like sometimes your board might not be recognizing the sim, but I'm not sure what else is going on.

What program are you running?

SRGDamia1, 2020-04-15:
I'm closing this old issue. Re-open if you're still having trouble.

closed 2020-04-15 by SRGDamia1