vshymanskyy/TinyGSM #249: Usage of TinyGsmClient and Blynk together

vitzaoral, 2019-03-07:
TinyGSM version: latest
Hardware: SIM800L + ESP32 (ESP32-WROOM-32)

### Scenario, steps to reproduce
I want to use Blynk in the standard way, and I also want to perform HTTP GET requests with TinyGsmClient as is described in [WebClient example](https://github.com/vshymanskyy/TinyGSM/blob/master/examples/WebClient/WebClient.ino).

### Expected result
Both Blynk and TinyGsmClient should work equally and not affect themselves

### Actual result
When I configure Blynk and client:

```
....
TinyGsm modem(gsmSerial);
TinyGsmClient client(modem);
....
Blynk.begin(blynkAuth, modem, "apn", "", "");
```

Everything works well. But when I want make some HTTP GET request, it doesn't work (timeout):

```
// OK
if (modem.isGprsConnected())
    {
// OK
        if (client.connect("example.com", 80))
        {
            // Make a HTTP request:
            client.print(String("GET ") + "/blah.txt" + " HTTP/1.0\r\n");
            client.print(String("Host: ") + "example.com" + "\r\n");
            client.print("Connection: close\r\n\r\n");

            // HERE - TIMEOUT
            unsigned long timeout = millis();
            while (client.connected() && millis() - timeout < 10000L)
            {
                while (client.available())
                {
                    char c = client.read();
                    timeout = millis();
                }
            }
        }
    }
```

But when I commented out _Blynk.begin_ and just connect to the internet with:

`modem.gprsConnect("internet", "", "");`

then HTTP requests works well. So _Blynk.begin_ kills somehow _TinyGsmClient_. Is there a way how to makethe HTTP request and using Blynk together?
Thank you!

SRGDamia1, 2020-04-15:
I'm sorry I never addressed this.  Did you ever get it working?  If so, can you please close this issue.

closed 2020-04-15 by SRGDamia1

vitzaoral, 2020-04-16:
Actually I can't remember :) I think so but I don't know how I managed it..