vshymanskyy/TinyGSM #251: Data length SIM808 & PubSubClient

ChrSchultz, 2019-03-09:
TinyGSM version: 
Hardware: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### AT command log

Hello,
I send via mqtt.publish a json-string (length larger than 100 byte) and nothing  was send.  is there any possibility to trim datas in packets to send. (sorry for bad english) 
My Json-string: `{"_type":"location","tst":1052854452,"lon":48.71865,"lat":uCXhHvLeCWR18.99171,"alt":104.3,"vel":0.24} `
i want to add[code `"batt": 4,123´ to the string, but then the string has a length of 115+ Byte and couldn't be sent.
thanks

SRGDamia1, 2019-03-09:
PubSubClient (not Tiny GSM) has a default max packet size of 128. You can increase this in PubSubClient.h - it's MQTT_MAX_PACKET_SIZE.

ChrSchultz, 2019-03-09:
yes, I know, but if i publish the full string - within `"batt";4.123"` over Mqtt via wifi it works the whole string will be sent. if i publish via GPRS(Tiny-GSM) nothing will be sent.

SRGDamia1, 2019-03-09:
What are you using for "over MQTT via wifi?" You mean using the same PubSubClient library but a different arduino client library for wifi works but TinyGSM doesn't? Or do you mean you can send the request through another means and it is received? Did you try increasing the packet size in PubSubClient.h? Keep in mind that the packet size includes the MQTT header and topic, not just the data.

ChrSchultz, 2019-03-09:
first I use PubSubClient with esp8266 and wifi and it works, with the large string (eg 115 Bytes), if I use   ESPClient(esp)TinyGSMClient sim808(esp) and PubSubClient mqtt(sim808) and publishing the large string (< 100 Bytes) with mqtt.publish(topic, payload).. nothing will be sent. 

I guess that either the communication between ESP and SIM808 limits the data, or the TinyGSM library...
there  TINY_GSM_RX_BUFFER 64 (bytes) defined....

ChrSchultz, 2019-03-09:
fixed, i set TINY_GSM_RX_BUFFER to 128 and all works fine :-)

closed 2019-03-09 by ChrSchultz

SRGDamia1, 2019-03-09:
The rx buffer is for text coming from the SIM808 not going to it. I'm not sure I understand how it would fix your issue. But if increasing that fixes it for you, that's great!

ChrSchultz, 2019-03-09:
RX means receicer and the rx- buffer Store datas coming From the Esp8622 which ist Computing the datas...
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/vshymanskyy/TinyGSM/issues/251#issuecomment-471217987",
"url": "https://github.com/vshymanskyy/TinyGSM/issues/251#issuecomment-471217987",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]