vshymanskyy/TinyGSM #254: Getting 4g lte sparkfun shield to work with blynk

dansass, 2019-03-16:
hello im having trouble getting this shield connected to blynk. I uploaded and successfully connected the shield to hologram using the 00_Register_Operator sketch. On tinyGSM wiki it says that my modem is supported (SARA-R410M-02B). The discussion started here 

[https://github.com/vshymanskyy/TinyGSM/issues/244](url)

my code looks like this,

`#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#define TINY_GSM_DEBUG Serial
// Default heartbeat interval for GSM is 60
// If you want override this value, uncomment and set this option:
//#define BLYNK_HEARTBEAT 30

// Select your modem:
//#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM7000
#define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE

#include <TinyGsmClient.h>
#include <TinyGsmClientUBLOX.h>
#include <BlynkSimpleTinyGSM.h>

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Hardware Serial on Mega, Leonardo, Micro
//#define SerialAT Serial1

// or Software Serial on Uno, Nano
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(8, 9); // RX, TX

#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "hologram";
const char user[] = "";
const char pass[] = "";

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
const char auth[] = "****************";

//TinyGsm modem(SerialAT);

/*#include <SparkFun_LTE_Shield_Arduino_Library.h>
// Create a LTE_Shield object to be used throughout the sketch:
#define POWER_PIN 5
#define RESET_PIN 6
const mobile_network_operator_t MOBILE_NETWORK_OPERATOR = MNO_SW_DEFAULT;
const String APN = "hologram";
LTE_Shield lte(POWER_PIN, RESET_PIN);
*/

void setup()
{
 
  /*pinMode(5,OUTPUT);
  digitalWrite(5,HIGH);
  delay(3200);
  digitalWrite(5,LOW);
  delay(10000);*/
  SerialMon.begin(9600);
  delay(2000);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(5000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  

  Blynk.begin(auth, modem, apn, user, pass);
  delay(5000);
}

void loop()
{
 Blynk.run();
}`

And this is how the Serial monitor looks

Initializing modem...
ATI

Manufacturer: u-blox
Model: SARA-R410M-02B
Revision: L0.0.00.00.05.06 [Feb 03 2018 13:00:]S:2II55963
O
[8005] ### Unhandled: Manufacturer: u-blox
Model: SARA-R410M-02B
Revision: L0.0.00.00.05.06 [Feb 03 2018 13:00:]S:2II55963
O
Modem: 
[8081] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.6.1 on Arduino Uno

[8240] Modem init...
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+CGMI

u-blox

OK
AT+GMM

SARA-R410M-02B

OK
[8417] ### Modem: u-blox SARA-R410M-02B
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
[8554] Connecting to network...
AT+CEREG?

+CEREG: 0,5

OK
AT+COPS?

+COPS: 1,0,"302 720 Hologram",8

OK
[8665] Network: 302 720 Hologram
[8666] Connecting to hologram ...
AT+CGACT=1,0

ERROR
AT+CGATT=1

OK
AT+CGDCONT=1,"IP","hologram"

OK
AT+CGACT=1,1

OK
[8863] Connected to GPRS
AT+USOCL=0,1

+CME ERROR: Operation not allowed

[8944] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[9010] Connecting to blynk-cloud.com:80
AT+USOCR=6
 Operation not allowed

+CME ERROR: Operation not allowed

[9093] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[13995] Connecting to blynk-cloud.com:80
AT+USOCR=6
 Operation not allowed

+CME ERROR: Operation not allowed

[14073] ### Unhandled: Operation not allowed
AT+USOCL=0,1

+CME ERROR:[18996] Connecting to blynk-cloud.com:80

SRGDamia1, 2019-03-16:
Hm.  Those operations *should* be allowed.  I've sometimes gotten error messages on the OSOCL (close socket) when there wasn't a socket open first, so that's suprising.  Opening the socket (USOCR) will fail if you're not really properly connected/attached to the EPS service.  Have you tried factory resetting the modem? (tools/FactoryReset or just sent AT&F)  That should also clear out your strange COPS settings.  After the factory reset, you may find you can connect to Hologram faster if you set the profile to AT&T and the radio access technology to LTE-M only (AT+COPS=2, AT+UMNOPROF=2 AT+URAT=7, AT+COPS=0, AT+CFUN=15)

dansass, 2019-03-16:
When i used blynk's sketch it acually got connected for maybe 20s! here is what came on serial mon;

i could not fit everything on here but i can explain some error codes

AT+USORD=0,0
+USORD: 0,0
OK
AT+USOCTL=0,10

+USOCTL: 0,10,4

OK
AT+USOWR=0,5

@AT+USOCL=0,1

+USOWR: 0,5

OK

+UUSORD: 0,5
AT+USOCL=0,1

OK
[111156] Connecting to 45.55.96.146:80
AT+USORD=0,0

ERROR
AT+USOCL=0,1

ERROR
AT+USOCR=6

+USOCR: 1

OK

dansass, 2019-03-16:
but than if i close it and open the serial mon again this is what i get...
```cpp
Initializing modem...
[3009] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.6.1 on Arduino Uno

[3126] Modem init...
AT

OK
ATE0

OK
AT+CMEE=0

OK
AT+CGMI

u-blox

OK
AT+GMM

SARA-R410M-02B

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
[3437] Connecting to network...
AT+CEREG?

+CEREG: 0,5

OK
AT+COPS?

+COPS: 1,0,"302 720 Things Mobile",8

OK
[3549] Network: 302 720 Things Mobile
[3549] Connecting to tm ...
AT+CGACT=1,0

ERROR
AT+CGATT=1

OK
AT+CGDCONT=1,"IP","tm"

OK
AT+CGACT=1,1

OK
[3735] Connected to GPRS
AT+USOCL=0,1

ERROR
AT+USOCL=0,1

ERROR
[3805] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
[8805] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
[13806] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
[18805] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
AT+USOCL=0,1

ERROR
[23809] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
[28810] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
[33811] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
[38812] Connecting to blynk-cloud.com:80
AT+USOCR=6

ERROR
AT+USOCL=0,1

ERROR
AT+USOCL=0,1

ERROR
```

SRGDamia1, 2019-03-16:
I think the factory reset is likely to help. I don't think there's much it would screw up - not unless there is something else you'd already successfully used this modem for and had saved settings that you think you could restore. It's odd to use manual COPS like you are and it's possible there are other strange settings saved. If that's not something you understand and did intentionally, I think the reset would help clear it up.

dansass, 2019-03-16:
ok... But this is what came up

```cpp
AT

OK
ATE0

OK
AT+CMEE=0

OK
AT+CGMI

u-blox

OK
AT+GMM

SARA-R410M-02B

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
***********************************************************
 Return settings to Factory Defaults: FAIL
***********************************************************
```

SRGDamia1, 2019-03-17:
Oops! I don't actually implement the factory reset command for the LTE-M models! Can you try using steam debugger and just send the command "AT&F"?

dansass, 2019-03-18:
AT&F worked yup but nothing better,
```
Initializing modem...
[3010] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.6.1 on Arduino Uno

[3127] Modem init...
AT

OK
ATE0

OK
AT+CMEE=0

OK
AT+CGMI

u-blox

OK
AT+GMM

SARA-R410M-02B

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
[3440] Connecting to network...
AT+CEREG?

+CEREG: 0,5

OK
AT+COPS?

+COPS: 0,0,"302 720 Things Mobile",8

OK
[3555] Network: 302 720 Things Mobile
[3556] Connecting to tm ...
AT+CGACT=1,0

ERROR
AT+CGATT=1

OK
AT+CGDCONT=1,"IP","tm"

OK
AT+CGACT=1,1

ERROR
[3745] Connect GPRS failed
```

SRGDamia1, 2019-03-18:
Have you ever successfully used this modem/sim combination to make a GPRS/TCP connection outside of TinyGSM?  Please verify in the u blox m-center program that you're able to successfully attach to the network and that you can see activity from your sim on the hologram dashboard.  The problems you're having seem to be related to your sim and getting it on the network not this library per-se.

dansass, 2019-03-19:
With this modem, no but i have got it successfully connected using a sim900. In the m center software i was trying to figure out if the sim is connected and allowing data, so i tried to do a firmware update and it did not seem to work

[image]

Would there be any other way to figure out if the sim is connected and allowing data? Or what that pdp context even means?

SRGDamia1, 2019-03-19:
You can't do the FOTA that simply with m-center.  

Open the network tab and the AT terminal.  Update the info and post the screenshot showing both.

dansass, 2019-03-19:
U can also gladly talk to me personally 519 288 3445

[image]
[image]

SRGDamia1, 2019-03-19:
You're successfully registered on the network.

Use the AT terminal in m-center to send the same commands TinyGSM attempts:
AT+CGATT=1
AT+CGDCONT=1,"IP","tm"
AT+CGACT=1,1
AT+USOCR=6

dansass, 2019-03-19:
everything went smooth. But when i ran the blynk sketch again i noticed that writes  AT+CGACT=1,0 instead of AT+CGACT=1,1

so i opened the TinyGsmClientUBLOX and there it said 
```
// LTE-M and NB-IoT modules do not support UPSx commands
    if (isCatM) {
      sendAT(GF("+CGACT=1,0"));  // Deactivate PDP context 1 
      if (waitResponse(40000L) != 1) {
        return false;
      }
    }
```
before we go further i should address that i do live in ontario canada so i dont know if all the lte bands are available

SRGDamia1, 2019-03-19:
```AT+CGACT=1,0``` is to detact/disconnect from GPRS.  It's correct in the library.

dansass, 2019-03-19:
So what now? U think i should give up for now or is there anything else i could try?

dansass, 2019-03-19:
this is the error that pops up now 

```
[12465] Redirecting to 45.55.96.146:80
AT+USOCL=6,1

OK
AT+USOCL=6,1

+CME ERROR:[12561] Connecting to 45.55.96.146:80
 Operation not allowed

[12584] ### Unhandled: Operation not allowed
AT+USOCL=6,1

+CME ERROR:[12696] Waiting for +UUSOCL URC on 6
 Operation not allowed

[12728] ### Unhandled: Operation not allowed

+UUSOCL: 6
[43597] ### Closed: 6
AT+USOCR=6

+USOCR: 6

OK
AT+USOSO=6,6,1,1

OK
AT+USOCO=6,"45.55.96.146",80

OK
AT+USORD=6,0
```

SRGDamia1, 2019-03-19:
The errors after ```AT+USOCL=6,1``` are not a problem.  Before opening a new connection on a client, the library stops any old connection.  If there wasn't a socket open, the ublox returns an error at the stop.  That's not a problem.  It looks like you made the TCP connection just fine.

dansass, 2019-03-19:
ok... would there be anything else i could do/try?

SRGDamia1, 2019-03-20:
Well the last thing you posted should have succeeded, the tcp socket was correctly opened. You didn't post the log beyond that. The errors in what you posted are not concerning.

You have posted results showing that you've tried to set different APN's, so you're obviously not running exactly the same code each time.  **You need to make sure you have the correct APN**. (hologram? tm? Something else?) You can sometimes get the network to allow one or two connections with the wrong apn, but that's it. After you're completely certain you have the right apn and all your libraries are up to date, post your code and log all the way to the end of the Blynk data post.

Also be aware that the first time you attempt to connect to the network it can sometimes take a loooong time to connect.

Anyway, I really think your problems are APN and network related, not the library.

closed 2019-03-20 by SRGDamia1