vshymanskyy/TinyGSM #255: `getGPS` vs `getGPSraw` accuracy issue/bug

BracketJohn, 2019-03-18:
TinyGSM version: 0.3.5
Hardware: SIM868 R14.18

### Scenario, steps to reproduce
Trying to get the exact position:
```c
modem.enableGPS();

String rawGps = modem.getGPSraw();

float lat, lng, speed;
int alt, vsat, usat;
modem.getGPS(&lat, &lng, &speed, &alt, &vsat, &usat);

Serial.println(rawGps);

Serial.print(lat);
Serial.print(", ");
Serial.println(lng);
```

### Expected result
```
> 1,1,20190318170547.000,50.XXXXXX,6.XXXXXX,227.847,0.00,157.3,1,,1.6,1.9,1.0,,12,6,,,30,,
> 50.XXXXXX, 6.XXXXXX
```

### Actual result
```
> 1,1,20190318170547.000,50.XXXXXX,6.XXXXXX,227.847,0.00,157.3,1,,1.6,1.9,1.0,,12,6,,,30,,
> 50.XX, 6.XX
```

(note the changed number of decimal places `X`)

## Possible solution/Source of issue

The [Arduino](https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/tofloat/) `toFloat` method used truncates `floats` rather strongly. Using `atof` or something similar should fix the issue. This issue should also arise for the SIM7000 module code should also be affected.

If this is an actual issue, and if my suspicion is confirmed by one of the maintainers, I'm more than open to create a PR addressing this, as I also need the higher accuracy at word 😄

I also suggest to improve `getGPS` further, e.g., `C/N0_MAX` is not retrieved by `getGPS`. `C/N0_MAX` is an important ratio that can tell the user how "close" he is to locking on GPS signal.

djcobylevy, 2019-12-23:
try:

Serial.print(lat, 6);
Serial.print(", ");
Serial.println(lng, 6);

BracketJohn, 2019-12-23:
This will produce:
```
> 50.XX0000, 6.XX0000
```
The problem is not the printing, but the underlying use of `toFloat`, which, as documented on the page that I linked, rounds to 2 places.

Although you're correct, that this is problematic in my problem description, will update this, thanks!

djcobylevy, 2019-12-23:
it does works for me.
this is my output:
19:40:17.482 -> 32.077618, 34.825180

djcobylevy, 2019-12-23:
it does works for me.
this is my output:
19:40:17.482 -> 37.077618, 31.825180

BracketJohn, 2019-12-23:
Which module are you using?

BracketJohn, 2019-12-23:
And, can you show me the code that produces this?

djcobylevy, 2019-12-23:
sim868

BracketJohn, 2019-12-23:
Also, maybe you should edit your comments, so that your exact location is hidden, you call!

djcobylevy, 2019-12-23:
> Also, maybe you should edit your comments, so that your exact location is hidden, you call!
i deleted the post and riposted it earlier haha.

the relevant code is (based on yours):
	  float lat, lng, speed;
	  int alt, vsat, usat;
          String GpsLat,GpsLong;
	  
          modem.getGPS(&lat, &lng, &speed, &alt, &vsat, &usat);

          //convert float to string
	  GpsLat=String(lat, 6);
	  GpsLong=String(lng, 6);

	  Serial.print(GpsLat);
	  Serial.print(", ");
	  Serial.println(GpsLong);

BracketJohn, 2019-12-23:
I think what's going on here, is that you're casting un-initialized memory to `String`. The Arduino String lib is *very* buggy and has many problems (heap allocation, arguably, could be one of them).

Is the location that is printed out to you, your actual location?

BracketJohn, 2019-12-23:
@djcobylevy please have a look at https://github.com/vshymanskyy/TinyGSM/issues/352 for more on this topic. I think you should totally continue to use this library, just be aware of its shortcomings.

djcobylevy, 2019-12-23:
> Is the location that is printed out to you, your actual location?
yes, its correct but the last digit (rounds up the 6th decimal 0.00000X).
i checked on the map and it comes down to a few meter which is good enough for my usage.

SRGDamia1, 2020-04-15:
I believe this has been addressed in the update to 0.10.

closed 2020-04-15 by SRGDamia1