vshymanskyy/TinyGSM #256: [ESP32][SIM808] getGsmLocation and USSD

DaveCalaway, 2019-03-18:
TinyGSM version: 0.6 ( https://goo.gl/qJZRtQ )
Hardware: ESP32 and SIM808 [board](https://goo.gl/HDfTCf)

### Scenario, steps to reproduce
I connected the ESP32 Serial(2) - 16,17 - to the RT and TX of the board.
No other pins are connected. 
I have [read](https://goo.gl/wNjVXt) that the RI pin is needed to verify then a call or SMS is incoming.

I have loaded the AllFunctions.ino example.

I disabled that function because that give me 601 ERROR. If i don't disable that, getGSMDataTime doesn't work. 
```
String gsmLoc = modem.getGsmLocation(); // -------- > 601 ERROR
DBG("GSM location:", gsmLoc);
```

### Expected result
No problems with the basic example.
I would like to use the SMS send/receive.

### Actual result
```
[3050] Trying baud rate 38400 ...
[6070] Modem responded at rate 38400
[6070] Initializing modem...
[10126] ### Unhandled: AT&FZ
[10133] ### Modem: SIMCom SIM808
[12174] Modem: SIM808 R14.18
[12195] Waiting for network...
[16399] Network connected
[16399] Connecting to internet.wind
[76412] ### Unhandled: ��
␔����j�����t�v�
SMS Ready
p�������w
[79854] GPRS status: connected
[79870] CCID: OK
[79883] IMEI: OK
[79900] Operator: Wind Telecom SpA
[79914] Local IP: OK
[79925] Signal quality: 23
[79938] Battery lavel: 62
[82938] Battery voltage: 3.90
[82954] GSM Time: 13:06:22+04
[82970] GSM Date: 19/03/18
[85203] Balance (USSD): UNKNOWN APPLICATION
[87774] Phone number (USSD): UNKNOWN APPLICATION
[88805] ### Unhandled: LK
�M��^/>������������k����u�~�����������
[88805] GPS raw data: 0,,,,,,,,,,,,,,,,,,,,
[89449] GPRS disconnected
[90787] ### Unhandled: '�
```

- During modem.restart():
sendAT(GF("+CFUN=0")) strange respond

- During waitForNetwork() -> isNetworkConnected() -> getRegistrationStatus():
it's a for loop, sometime return "Call Ready", is it normal?

- During gprsConnect():
what is "SMS Ready" ?

- During getGSMDateTime():
the time is correct, but one year back

- During modem.sendUSSD("*111#") -> TinyGsmDecodeHex8bit(hex):
it's return [26614] Balance (USSD): UNKNOWN APPLICATION
and for modem.sendUSSD("*161#"):
[29277] Phone number (USSD): UNKNOWN APPLICATION
i really not understand what it doing....

The GPS works great after 4/5 cycles. 

### AT command log
DUMP_AT_COMMANDS
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
[3051] Trying baud rate 38400 ...
[4066] Modem responded at rate 38400
[4066] Initializing modem...
AT

OK
AT+CLTS=1

OK
AT&W

OK
AT+CFUN=0

+SAPBR 1: DEACT
⸮⸮⸮⸮⸮7 ⸮N⸮⸮⸮⸮⸮⸮7⸮⸮~⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮n⸮⸮⸮⸮⸮
+C ⸮N: NOT READY
⸮~⸮⸮⸮⸮⸮⸮⸮ ⸮n⸮⸮⸮⸮⸮⸮⸮⸮⸮~⸮⸮⸮⸮⸮⸮⸮⸮⸮n⸮⸮⸮⸮⸮1⸮T⸮⸮⸮⸮l⸮⸮⸮z⸮⸮⸮⸮TӅ׾⸮⸮_⸮⸮⸮⸮<?⸮⸮⸮8⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮X
OK
AT+CFUN=1,1

OK
AT

OK
AT&FZ
ATE0
ATE0

OK
[10613] ### Modem: SIMCom SIM808
AT+CPIN?

ERROR
AT+CPIN?

+CPIN: READY

+CPIN: READY

OK
ATI

SIM808 R14.18

OK
[11645] Modem: SIM808 R14.18
AT+CPIN?

+CPIN: READY

OK
AT+CPIN=""

ERROR
[11668] Waiting for network...
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?
⸮
+CREG: 0,2

OK
AT+CREG?
⸮⸮⸮⸮]⸮t⸮m⸮v.⸮
+CREG: 0,2

OK
AT+CREG?
⸮⸮⸮g⸮".⸮⸮⸮⸮?⸮⸮
+CREG: 0,2

OK
AT+CREG?
⸮⸮⸮*⸮S⸮⸮⸮6⸮
+CREG: 0,2

OK
AT+CREG?
⸮L⸮7⸮⸮f⸮=⸮
Call Ready
⸮$J⸮4⸮?⸮⸮
+CRE @0,2

OK
}⸮⸮⸮⸮⸮ղ⸮⸮7⸮⸮N⸮6?=⸮⸮⸮⸮k/Ԫ⸮⸮⸮⸮⸮=?⸮}(⸮*⸮_⸮W⸮{ü⸮⸮⸮Z[16331] ### Unhandled: ⸮L⸮7⸮⸮f⸮=⸮
Call Ready
⸮$J⸮4⸮?⸮⸮
+CRE@0,2

OK
}⸮⸮⸮⸮⸮ղ⸮⸮7⸮⸮N⸮6?=⸮⸮⸮⸮k/Ԫ⸮⸮⸮⸮⸮=?⸮}(⸮*⸮_⸮W⸮{ü⸮⸮⸮Z
AT+CREG?
⸮⸮Q⸮⸮⸮⸮⸮⸮
+CREG: 0,1

OK
AT+CREG?

+CREG: 0,1

OK
[16604] Network connected
[16604] Connecting to internet.wind
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","internet.wind"

OK
AT+CGDCONT=1,"IP","internet.wind"
⸮M⸮⸮⸮
 CK

SMS Ready
[17664] ### Unhandled: ⸮M⸮⸮⸮
CK

SMS Ready
AT+CGACT=1,1
⸮⸮߮⸮6⸮⸮⸮ٶ⸮⸮{]⸮⸮⸮⸮6⸮N⸮1⸮
⸮?m⸮⸮j⸮޿⸮6⸮⸮O⸮p⸮⸮*⸮S$⸮\⸮j߮⸮s⸮⸮⸮P⸮⸮
⸮s$⸮\⸮⸮V⸮'⸮
OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.6.252.95"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="internet.wind","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

10.6.252.95

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

10.6.252.95

OK
[21384] GPRS status: connected
AT+ICCID
⸮⸮
+ICCID: 8939883681000183251f

OK
[21401] CCID: 8939883681000183251f
AT+GSN

865067020808575

OK
[21414] IMEI: 865067020808575
AT+COPS?

+COPS: 0,0,"Wind Telecom SpA"

OK
[21430] Operator: Wind Telecom SpA
AT+CIFSR;E0

10.6.252.95

OK
[21444] Local IP: 10.6.252.95
AT+CSQ

+CSQ: 23,0

OK
[21456] Signal quality: 23
AT+CBC

+CBC: 0,62,3898

OK
[21468] Battery lavel: 62
AT+CBC

+CBC: 0,62,3898

OK
7⸮[23480] ### Unhandled: 7⸮
[23480] Battery voltage: 3.90
AT+CCLK?

+CCLK: "19/03/18,13:42:24+04"[23497] GSM Time: 13:42:24+04
AT+CCLK?

OK

+CCLK: "19/03/18,[23513] GSM Date: 19/03/18
AT+CMGF=1
13:42:24+04"

OK
AT+CSCS="HEX"

OK
AT+CUSD=1,"*111#"

OK

OK
w⸮eծv/⸮⸮_⸮⸮⸮m⸮⸮⸮#⸮⸮|⸮⸮*.⸮⸮v⸮]⸮_#⸮^.⸮}_⸮⸮⸮⸮⸮⸮⸮⸮⸮>oi⸮v⸮⸮ע⸮⸮⸮⸮⸮⸮⸮׶;⸮⸮⸮⸮⸮⸮⸮⸮⸮t⸮⸮>⸮߂⸮⸮⸮⸮?⸮S⸮⸮⸮⸮;윞⸮r⸮⸮⸮⸮⸮⸮~⸮⸮^ն⸮/⸮⸮⸮⸮&e⸮⸮⸮ޢ⸮}⸮⸮⸮A,⸮⸮⸮h⸮?⸮-⸮⸮g⸮>⸮>⸮⸮T⸮⸮⸮⸮⸮⸮⸮z⸮
+CUSD: 2, "554E4B4E4F574E204150504C49434154494F4E", 15
[26614] Balance (USSD): UNKNOWN APPLICATION
AT+CMGF=1

OK
AT+CSCS="HEX"

OK
AT+CUSD=1,"*161#"

OK
#⸮⸮⸮⸮_t⸮zL⸮|⸮⸮s3⸮⸮o⸮⸮<⸮,⸮ORM[c&⸮⸮⸮⸮N⸮⸮⸮⸮{⸮⸮O⸮⸮⸮\7⸮<⸮ ;
⸮⸮⸮yO3⸮W⸮⸮r⸮?⸮⸮⸮ݪx⸮⸮)⸮⸮⸮\⸮⸮⸮⸮⸮⸮⸮p⸮}⸮x⸮⸮⸮I҂⸮⸮⸮y⸮⸮n⸮⸮j⸮I⸮⸮⸮⸮⸮m⸮⸮⸮⸮'⸮$⸮⸮]R⸮⸮(⸮]n⸮⸮]⸮⸮W⸮
+CUSD: 2, "554E4B4E4F574E204150504C49434154494F4E", 15
[29277] Phone number (USSD): UNKNOWN APPLICATION
AT+CGNSPWR=1

OK
AT+CGNSINF

+CGNSINF: 0,,,,,,,,,,,,,,,,,,,,

OK
AT+CGNSPWR=0
⸮C⸮j⸮H⸮}w[⸮n(⸮⸮⸮⸮⸮⸮_~⸮⸮?o4Wh⸮⸮⸮⸮7⸮[30308] ### Unhandled: ⸮C⸮j⸮H⸮}w[⸮n(⸮⸮⸮⸮⸮⸮_~⸮⸮?o4Wh⸮⸮⸮⸮7⸮
[30308] GPS raw data: 0,,,,,,,,,,,,,,,,,,,,
AT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT
⸮⸮⸮@⸮=⸮⸮⸮⸮⸮⸮D⸮=⸮⸮⸮⸮⸮⸮⸮⸮^⸮⸮⸮⸮⸮⸮⸮⸮⸮^⸮⸮⸮⸮⸮⸮⸮⸮⸮ھ⸮⸮⸮⸮@⸮^⸮⸮⸮⸮⸮⸮D⸮⸮⸮⸮⸮⸮⸮
OK
AT+CGATT?

+CGATT: 0

OK
[30896] GPRS disconnected
⸮⸮[31238] ### Unhandled: ⸮⸮
```

ShwethaVasan, 2019-08-01:
Did you get what was happening?
If not, try setting the baud rate to 115200..or pass a CR('\r') to auto detect the BR before sending the AT commands

SRGDamia1, 2020-04-15:
I'm closing this old issue. Re-open if you're still having trouble.

closed 2020-04-15 by SRGDamia1