vshymanskyy/TinyGSM #257: AT+CFUN=16 not supported on Ublox SARA-R412M

jpmeijers, 2019-03-19:
TinyGSM version: commit 5858a83
Hardware: Sodaq SARA AFF with ublox SARA-R412M

### Scenario, steps to reproduce
Using HttpsClient example can't connect to network.

Using Diagnostics example:
```
Initializing modem...AT
AT

OK
AT+CFUN=16
AT+CFUN=16

ERROR
 [fail]
************************
 Is your modem connected properly?
 Is your serial speed (baud rate) correct?
 Is your modem powered on?
 Do you use a good, stable power source?
 Try useing File -> Examples -> TinyGSM -> tools -> AT_Debug to find correct configuration
************************
```

### Expected result
`AT+CFUN=16` should return with OK not ERROR.

### Actual result
`AT+CFUN=16` responds with ERROR.

### AT command log
```
AT

OK
AT+CFUN=16

ERROR
AT+CFUN=?

+CFUN: (0,1,4,15),(0-1)

OK
ATI

Manufacturer: u-blox
Model: SARA-R412M-02B
Revision: M0.05.00 [Jul 06 2018 10:01:55]
SVN: 03
IMEI: 354679090045173

OK
```

SRGDamia1, 2019-03-19:
This has already been fixed with fdd0279f70d7aa7d890c6db182dbabfef5ca6ce7.  I didn't bump the version after that fix.  I'll do that.

closed 2019-03-19 by SRGDamia1

jpmeijers, 2019-03-22:
I'm working on a few tweaks to get the SARA-R412M working as expected. For now the issue described here is still there. The problem is in the examples `modem.restart()` is called before `modem.init()`. During init the isCatM flag is detected and set. If one do a reset without first calling init, the library thinks that isCatM is false, and tries to reset the modem using `AT+CFUN=16`.

If I get this module working correctly using this library I will send a PR.

jpmeijers, 2019-03-22:
The tricky thing is that the library handles the SARA range of modules in two ways:
* isCatM == false
* isCatM == true

The SARA-R412M supports CatM, so is correctly detected as isCatM == true. However this specific module has support to fall back to 2G/GSM/GPRS when CatM is not available. This all depends on wat radio access technology (RAT) has been configured.

When the module is configured to use GPRS, the library still detects it as isCatM == true. The following function is an issue:
```
RegStatus getRegistrationStatus() {
    if (isCatM) {  // Check EPS registration for LTE modules
      sendAT(GF("+CEREG?"));
      if (waitResponse(GF(GSM_NL "+CEREG:")) != 1) {
        return REG_UNKNOWN;
      }
    }
    else {
      sendAT(GF("+CGREG?"));  // Check GPRS registration for others
      if (waitResponse(GF(GSM_NL "+CGREG:")) != 1) {
        return REG_UNKNOWN;
      }
    }
    streamSkipUntil(','); // Skip format (0)
    int status = stream.readStringUntil('\n').toInt();
    waitResponse();
    return (RegStatus)status;
  }
```
Because the module isCatM, but it is configured or falls back to GPRS, the `AT+CEREG` will always return `+CEREG: 0,4` and therefore the `getRegistrationStatus` will fail, making the library think that it is not connected to a network. In the meantime the module did successfully connect to GPRS `+CGREG: 0,1`.

To support this module the library need to handle U-blox modules in three ways:

```
if ( isCatM && RAT==9 ) {
  // GPRS mode. Use CGREG.
}
else if ( isCatM && (RAT == 7 || RAT == 8) ) {
  // CatM mode. Use CEREG.
} else {
  // GPRS mode. Use CGREG.
}
```

What one can also do is use the `CREG` command to get the status of either EPS or GPRS. I'm trying this approach now.