vshymanskyy/TinyGSM #265: Client interface connects once, then hangs modem

jeancaffou, 2019-04-14:
I am experiencing the same issue as in (horribly) described ticket #127

Using modem SIM900

The problem is consistently reproducible, 

After quite a few hours wasted for debugging I figured out that this happens on persistent connections.

Steps to reproduce:
- Set up a TCP server that doesn't disconnect clients (persistent connection)
- Arduino code:
```
client.connect(server, port);
while(client.available()) client.read(); // optional
client.flush(); // optional
client.stop(); // optional
```

Only using `client.connect();` will hang the modem and modem will not respond anymore, even to commands like `reset()` or `poweroff()`

After connection data can be sent from the client and the server will receive it.
Using `client.stop();` does not work or solve the problem (which it probably should)

After using `client.connect()` the modem will respond false in `isNetworkConnected()`

Only after hard reset of the modem (cutting the power) the modem will work again and connect to the network and GPRS again.

**Another workaround is if the server disconnects the client off the connection.**
It seems there is something blocking the modem after connection.

Interestingly, using a higher level protocol, like using ArduinoHTTPClient is more reliable.
Probably because of the same reason - after the body is sent, HTTP server disconeccts the client.

SRGDamia1, 2019-05-07:
Have you seen this with anything other than the SIM800?  You referenced a few issues tied to the A6/A7 as well as the SIM800.

You said the results are very reproducible with TinyGSM.  Do you see the same trouble if you're making the connection with the modem using a terminal and AT commands?  Is it definitely the order of the TinyGSM commands that crash it?

I have a SIM800, but I don't have access to a place that still has 2G service to do much testing.  It's rapidly going the way of the dinosaurs here in the US.

jeancaffou, 2019-05-08:
SIM900, no I havent tested with other modems. But it is reproducible with the code provided.
No custom AT commands. If the server does not disconnect the client, then the modem will hang.

SRGDamia1, 2019-05-22:
I can't replicate this with a SIM800.

SRGDamia1, 2019-05-25:
I got a SIM900 from a coworker, and I still can't replicate this.  I can connect with keep-alive, disconnect, and re-connect again.  I can flip between two different servers.  I can disconnect from GPRS and then reconnect.

I guess I don't understand what you mean by the modem hanging and requiring a power cycle.  I assume that means that no AT commands yield any response and everything times out.  But you say that you can still send and receive data after connecting, which clearly wouldn't be possible if the modem was non-responsive.

Can you try 0.7.7 and give a more detailed log of what happens?

jeancaffou, 2019-06-13:
Tried it with SIM800F. Works flawlessly. Seems like there is a problem with the modem.

closed 2019-06-13 by jeancaffou