vshymanskyy/TinyGSM #266: read() screws sock_available

karioja, 2019-04-16:
https://github.com/vshymanskyy/TinyGSM/blob/5de558352b24a13cbac564b7717e373da13e9ff9/src/TinyGsmClientSIM800.h#L106

read() should not touch sock_available. sock_available is maintained in maintain() and modemRead() and after the call to modemRead sock_available already reflects the latest amount of stuff in the modem internal buffer. No need to subtract anything. After changing the line to  
`at->modemRead(TinyGsmMin((uint16_t)rx.free(), sock_available), mux);`
I can get a perfect MQTT connection and managed to get the FileDownload example also working. Before that the code would get into an endless RXGET loop.

SRGDamia1, 2019-05-06:
Hm.  Yes, I think you're right.  I had added that because I was having issues with the socket always reading available.  I think the real trouble might be in modemRead where it was waiting until the requested length of characters have been returned.  Since the "confirmed" length can be less than the requested length, that may have been causing the freeze.

SRGDamia1, 2019-05-07:
Ok.  Yes.  You're definitely right.  What happened was that for the ublox and quentcel modems modemRead *wasn't* setting sock_available, so read and stop were working around that.  When I was trying to smooth changes between the modems, I missed it and copied the incorrect work-around back from the ublox to the SIM800.

I think it should be fixed with #276 .

closed 2019-05-07 by SRGDamia1