vshymanskyy/TinyGSM #268: [E][WString.cpp:185] changeBuffer(): realloc failed! Buffer unchanged

zuqualla, 2019-04-17:
TinyGSM version: 0.3.5
Hardware: esp32 DEVkitC
Modem: SIM800 R14.18

### Scenario, steps to reproduce
i use the httpclient example, slightly modified for HardwareSerial on ESP32. I just get the content of a webpage. It works most of the time, but once in a while it gives:
`Performing HTTP GET request... -3`
followed by endless repitition of:
`[E][WString.cpp:185] changeBuffer(): realloc failed! Buffer unchanged`
It hangs the whole esp32 basically...

My full code:
`/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * For this example, you need to install ArduinoHttpClient library:
 *   https://github.com/arduino-libraries/ArduinoHttpClient
 *   or from http://librarymanager/all#ArduinoHttpClient
 *
 * TinyGSM Getting Started guide:
 *   http://tiny.cc/tiny-gsm-readme
 *
 * For more HTTP API examples, see ArduinoHttpClient library
 *
 **************************************************************/

// Select your modem:
#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_ESP8266

// Increase RX buffer if needed
#define TINY_GSM_RX_BUFFER 64

#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>

// Uncomment this if you want to see all AT commands
//#define DUMP_AT_COMMANDS

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Use Hardware Serial on Mega, Leonardo, Micro
//#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX

#include <HardwareSerial.h>
HardwareSerial SerialAT(1);

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "default";
const char user[] = "";
const char pass[] = "";

// Server details
const char server[] = "airquality.addisabeba.info";
//const char resource[] = "/TinyGSM/logo.txt";
const char resource[] = "/myphp/sim.html";
const int  port = 80;

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

TinyGsmClient client(modem);
HttpClient http(client, server, port);

int loopCount;

void setup() {
  float msSetup = millis();
  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(115200, SERIAL_8N1, 26, 27); // Rx | Tx
  //SerialAT.begin(115200);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println(F("Initializing modem..."));
  modem.restart();

  String modemInfo = modem.getModemInfo();
  SerialMon.print(F("Modem: "));
  SerialMon.println(modemInfo);

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");
  SerialMon.println("setup done in (sec):" + String((millis()- msSetup)/1000));
}

void loop() {
  loopCount++;
  SerialMon.println("start loop #" + String(loopCount));
  delay(100);
  float msLoop = millis();
  SerialMon.print(F("Waiting for network..."));
  if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  SerialMon.print(F("Connecting to "));
  SerialMon.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  SerialMon.print(F("Performing HTTP GET request... "));
  int err = http.get(resource);
  if (err != 0) {
    SerialMon.println(F("failed to connect"));
    delay(10000);
    return;
  }

  int status = http.responseStatusCode();
  SerialMon.println(status);
  if (!status) {
    delay(10000);
    return;
  }

  while (http.headerAvailable()) {
    String headerName = http.readHeaderName();
    String headerValue = http.readHeaderValue();
    //SerialMon.println(headerName + " : " + headerValue);
  }

  int length = http.contentLength();
  if (length >= 0) {
    SerialMon.print(F("Content length is: "));
    SerialMon.println(length);
  }
  if (http.isResponseChunked()) {
    SerialMon.println(F("The response is chunked"));
  }

  String body = http.responseBody();
  SerialMon.println(F("Response:"));
  SerialMon.println(body);

  SerialMon.print(F("Body length is: "));
  SerialMon.println(body.length());

  // Shutdown

  http.stop();
  SerialMon.println(F("Server disconnected"));

  modem.gprsDisconnect();
  SerialMon.println(F("GPRS disconnected"));
  
  SerialMon.println("loop done in (sec):" + String((millis()- msLoop)/1000));
  // Do nothing forevermore
//  while (true) {
    delay(10000);
//  }
}
`

SRGDamia1, 2019-05-08:
It looks like you're getting a time-out on the http request.  (That's the -3.)

If the ESP32 developers say that's a memory error, I'd assume they're right.  TinyGSM isn't directly calling new/malloc/realloc.  It does use the Arduino String class, but so do many libraries.

SRGDamia1, 2020-04-15:
I'm closing this old issue.  Re-open if you're still having trouble.

closed 2020-04-15 by SRGDamia1