vshymanskyy/TinyGSM #269: During comilation example WebClient.ino on ESP32 got error "cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClient'

Warlib1975, 2019-04-18:
During compilation on ESP32 board the example WebClient.ino got the error

GSM_SIM800L.cpp:39:15: error: cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClient'

 TinyGsmClient client(modemGSM);

               ^

In file included from C:\Users\Fedorov_A\Documents\Arduino\libraries\TinyGSM_ID1287\src/TinyGsmClient.h:15:0,

                 from sketch\GSM_SIM800L.h:7,

                 from sketch\GSM_SIM800L.cpp:1:

C:\Users\Fedorov_A\Documents\Arduino\libraries\TinyGSM_ID1287\src/TinyGsmClientSIM800.h:50:7: note:   because the following virtual functions are pure within 'TinyGsmSim800::GsmClient':

 class GsmClient : public Client

       ^

In file included from C:\Users\Fedorov_A\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:157:0,

                 from sketch\GSM_SIM800L.h:4,

                 from sketch\GSM_SIM800L.cpp:1:

C:\Users\Fedorov_A\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:31:17: note: 	virtual int Client::connect(IPAddress, uint16_t, int)

     virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;

                 ^

C:\Users\Fedorov_A\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:32:17: note: 	virtual int Client::connect(const char*, uint16_t, int)

     virtual int connect(const char *host, uint16_t port, int timeout) =0;

                 ^

Multiple libraries were found for "WiFiClient.h"

I couldn't find any solution here. The issue https://github.com/vshymanskyy/TinyGSM/issues/220  is similar for ESP8266

Warlib1975, 2019-04-18:
The problem occurs only when using GSMClient class. Everything work ok without using it. 

MH ET Live ESP32 Devboard

hardmax, 2019-04-22:
I have the same problem, for now I change my Arduino ESP32 Board Version form 1.0.2 to 1.0.1...With this I can compile now

hardmax, 2019-04-23:
I comment in C:\Users\Max\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\Client.h

This two lines:

//virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;
//virtual int connect(const char *host, uint16_t port, int timeout) =0;

After that I can compile

Warlib1975, 2019-04-23:
> After that I can compile, but It is not the correct solution...

Thank you for the solution. But, I suppose it's a too rough way to solve the problem. Every time you need to remember, that after ESP32 lib updating you need to fix Client.h

Warlib1975, 2019-04-23:
I solved the problem with compilation adding this block of code into TinyGsmClientSIM800.h

  #ifdef ESP32 || ESP8266
  virtual int connect(IPAddress ip, uint16_t port, int timeout)
  {
    throw "Method [virtual int connect(IPAddress ip, uint16_t port, int timeout)] is not implemented in TinyGsmClientSIM800.h";
  }

  virtual int connect(const char *host, uint16_t port, int timeout)
  {
    throw "Method [virtual int connect(const char *host, uint16_t port, int timeout)] is not implemented in TinyGsmClientSIM800.h";
  }
  #endif

In Arduino everything compiles ok.

detamend, 2019-05-01:
Same problem here while compiling on a ESP8266 / WeMos D1. I tried Warlib1975 fix to no avail. Compiliation breaks at the TinyGsmClient client(modem) - Line 81 in the webclient.ino.

Warlib1975, 2019-05-02:
> Same problem here while compiling on a ESP8266 / WeMos D1. I tried Warlib1975 fix to no avail. Compiliation breaks at the TinyGsmClient client(modem) - Line 81 in the webclient.ino.

Try to remove ifdef section of the fix. I'm not sure that in case of different ESP8266 boards this clause works properly.

detamend, 2019-05-03:
Thanks Warlib1975 for the idea - sadly no luck there :(

SRGDamia1, 2019-05-07:
This really *should* be fixed on the ESP32 side, not by TinyGSM.  They added/modified virtual functions to the Client.h that are out of line with the standard Arduino API.  They should have been aware of this, but I created a new issue on their side just in case (https://github.com/espressif/arduino-esp32/issues/2755).

The ESP8266 core created the same problem in version 2.5.0.  On the ESP8266 side, it appears to be fixed for the yet-to-be-released 2.5.1.

SRGDamia1, 2019-05-07:
As mentioned in https://github.com/vshymanskyy/TinyGSM/issues/220, you should be able to compile by rolling back to core version 1.0.1 for the ESP32.

For the ESP8266 you could roll back to 2.4.2 or use pull the core from git.

SRGDamia1, 2019-05-22:
The compiler problems with the ESP32 and ESP8266 should now be fixed.  The timeout functions are now fully implemented for all modems.

closed 2019-05-22 by SRGDamia1