vshymanskyy/TinyGSM #270: Problems with memory management, fixed by increasing data.reserve(64);

alches, 2019-04-18:
Arduino MKRZERO.
With default configuration of data.reserve(64); in TinyGSMClientSim800.h at line 940, unable to register to network. After increasing reserved memory amount, hanging after different AT commands execution. Some stability is reached at data.reserve(256);
Suppose main problem is using of String class with it's known heap fragmentation issues.
Any ideas/recommendations?

TinyGSM version: 0.6.1
Hardware: SIM800 R14.18

### Scenario, steps to reproduce
Trying to register to network using modem.init(); 

### Expected result
modem.isNetworkConnected() should return true;

### Actual result
returning false;

### AT command log
00:01:51.342 -> AT
00:01:51.623 -> AT
00:01:51.623 -> BUAT
00:01:52.232 -> AT
00:01:52.533 -> AT
00:01:52.816 -> AT
00:01:52.816 -> ⸮AT
00:01:53.424 -> AT
00:01:53.752 -> AT
00:01:54.033 -> AT
00:01:54.363 -> AT
00:01:54.644 -> AT
00:01:54.644 -> ⸮AT

00:01:54.644 -> OK
00:01:54.644 -> AT&FZ
00:01:54.644 -> AT&FZ

00:01:54.644 -> OK
00:01:54.644 -> ATE0
00:01:54.644 -> ATE0

00:01:54.644 -> OK
00:01:54.644 -> AT+CPIN?
00:01:54.644 -> 
00:01:54.644 -> ERROR
00:01:55.663 -> AT+CPIN?
00:01:55.663 -> 
00:01:55.663 -> RDY
00:01:55.663 -> 
00:01:55.663 -> +CFUN: 1
00:01:55.663 -> 
00:01:55.663 -> ERROR
00:01:56.682 -> AT+CPIN?
00:01:56.682 -> 
00:01:56.682 -> +CPIN: READY
00:01:56.682 -> 
00:01:56.682 -> +CPIN: READY
00:01:56.682 -> 
00:01:56.682 -> OK
00:01:56.682 -> Modem: ATI
00:01:56.682 -> 
00:01:56.682 -> SIM800 R14.18
00:01:56.682 -> 
00:01:56.682 -> OK
00:01:56.682 -> SIM800 R14.18
00:02:06.656 -> AT+CREG?
00:02:06.656 -> 
00:02:06.656 -> Call Ready
00:02:06.656 -> 
00:02:06.656 -> SMS Ready
00:02:06.656 -> 
00:02:06.656 -> +CREG: 0,1
00:02:06.656 -> 
00:02:06.656 -> OK

SRGDamia1, 2019-05-30:
Exactly what code are you running?  Are you close to your board's memory limits?  I don't quite see how increasing the data.reserve helps you.

Also, the modem.init() function doesn't do anything to register on the network.  The SIM800 should do that on its own.  But you do have to wait for the registration to happen.  Depending on how good or bad your service is, the SIM800 might take several minutes before it registers and isNetworkConnected() returns true.  You should run waitForNetwork() first.

alches, 2019-05-31:
Seems that Scheduler library by Arduino v4.4.0 used is causing problem with String-s used in TinyGSM.
Closing this thread.
Thanks and sorry for wasting your time

closed 2019-05-31 by alches