vshymanskyy/TinyGSM #273: if (!modem.isGprsConnected()) : error: invalid use of non-static member function

Mluis31, 2019-04-30:
Hi, i'm using a SIM800L and a ESP32 with Arduino IDE, I'm using this line:" if (!modem.isGprsConnected()) ESP.restart; " in order to restart the system when gsm connection is lost. But i'm getting this "error: invalid use of non-static member function" does anyone know why this happens?

SRGDamia1, 2019-05-06:
Do you mean ```ESP.restart():```?

Mluis31, 2019-05-06:
Yes.

Mluis31, 2019-05-06:
Yes. But this way it works 
[image]

SRGDamia1, 2019-05-22:
Yes, you need the open and close parenthesis after ```ESP.restart()```.  It shouldn't matter if it's on a new line with curly braces, as long as that would be the only statement inside the curly braces.

closed 2019-05-22 by SRGDamia1