vshymanskyy/TinyGSM #275: how to make check on payload to perform action in subscribe topic

shadababe04, 2019-05-07:
TinyGSM version: 
Hardware: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### AT command log

SRGDamia1, 2019-05-07:
I don't understand your issue.

shadababe04, 2019-05-08:
in the given example, led status change on/off if the matching topic is received but I want to change the led status on payload attached with the topic.like
when i get ON in payload then led on and off with off.

SRGDamia1, 2019-05-08:
Now I understand what you're trying to do, but the support for that is way outside the scope of this library.

This library only provides the client architecture.  MQTT support would come from another library that rides on top of that architecture.  Topic parsing and LED control would be yet another layer above the MQTT support.

closed 2019-05-08 by SRGDamia1