vshymanskyy/TinyGSM #278: Add support for SIMCOM 7020 LTE-M/NB IoT module

SRGDamia1, 2019-05-08:
Per request in #277

geologic, 2019-09-20:
That would be great, thumb up!

jonnor, 2019-10-25:
Has anyone tested this? The manufacturer claim to be somewhat compatible with SIM800C.

geologic, 2019-10-25:
> 
> 
> Has anyone tested this? The manufacturer claim to be somewhat compatible with SIM800C.

Well, not sure what manufacturer claims, but it is not. Some AT commands are different and the HTTP stack is also different. For example,

retrieve IMEI:
SIM800 - AT+CGSN
SIM7020 - AT+GSN

registration status:
SIM800 - AT+CREG
SIM7020 - AT+CGREG

register APN
SIM800 - AT+CSTT="apn","user","pass"
SIM7020 - AT+CSGACT=1,1,"apn"

SRGDamia1, 2019-10-25:
I've skimmed the documentation.  I think the entire set of TCP commands is different.

captFuture, 2019-12-13:
I was trying to implement this yesterday, the basic stuff works, but i am completely struggling with the http stuff (trying to test this in AT Command tester before actually coding and am unable to get a working http connection with it)

The base set of SIM800 commands seems to be working for it, apart from some small differences but i have not found a working alternative for the SAPBR stuff.

geologic, 2019-12-18:
SIM7020 seems to have a different HTTP stack, to make a HTTP POST request i use this sequence (not all AT commands are mandatory):

AT
AT+CGSN
AT+CSQ
AT+CGREG?
AT+COPS?
AT*MCGDEFCONT="IP"," apn_here"
AT+CHTTPCREATE="http://myserver.com"
AT+CHTTPCON=0
AT+CHTTPSEND=0,1,"/data/mysensor.php",436f6e6e656374696f6e3a204b6565702d416c6976650d0a,"application/octet-stream",68656c6c6f20776f726c64
AT+CHTTPDISCON=0
AT+CHTTPDESTROY=0

The first HEX part of  CHTTPSEND is the header and must be terminated with 0D0A (CRLF)
The second HEX part is the data you want to send.

The response for that command is sent in 2 parts, like this:

+CHTTPNMIH:0,0,800,Date: Tue, 10 Apr 2018 07:24:25 GMT
Server: Apache/2.0.58 (Win32) PHP/5.2.11
Last-Modified: Fri, 16 May 2014 01:01:31 GMT
ETag: "282e-45-f4410fef"
Accept-Ranges: bytes
Content-Length: 69
Content-Type: text/html

+CHTTPNMIC:0,0,69,138,3c736372697074206c616e67756167653d6a6176617363726970743e6c6f636174696f6e2e687265663d27657370636d732f696e6465782e706870273c2f7363726970743e

If you manage to make it work let me know.

9H1LO, 2020-01-21:
support for sim7020 seems to be only for gprs it seems....no nb-iot support

roysG, 2021-04-04:
any update on this?

roysG, 2021-04-10:
??

islandmonkey, 2021-04-14:
I'd be interested too

wjax, 2022-03-24:
Would love to see it working too. M5Stack Atom DTU uses it!
Great work and thanks a lot