vshymanskyy/TinyGSM #279: [SIM800L] Waiting for network

DaveCalaway, 2019-05-09:
Version: 0.6.2
Hardware: SIM800L

### Scenario, steps to reproduce
Connection, RX TX with level shifter.
Power the module by 5V Arduino Uno, through a diode, to Vcc SIM800L.

### Actual result
The Serial stay at "Waiting for network..."
```
[3009] Trying baud rate 38400 ...
[6028] Modem responded at rate 38400
[6028] Initializing modem...
[9118] ### TinyGSM Version: 0.6.2
[9319] ### Unhandled: ⸮
[9440] ### Modem: SIMCom SIM800
[10480] Modem: SIM800 R14.18
[20579] Waiting for network...
[90679] Initializing modem...
```

SRGDamia1, 2019-05-22:
How confident are you in your level shifters?  It looks like some of the characters are being botched.

DaveCalaway, 2019-05-29:
Probably my 800l doesn't work.... i will try with the 900A

closed 2019-06-05 by DaveCalaway