vshymanskyy/TinyGSM #280: Excessive memory consumption in 0.6.2 (AVR)

AlexIII, 2019-05-10:
TinyGSM version: 0.6.2
Hardware: Arduino Nano ATmega328P + SIM800C

### Scenario, steps to reproduce
Upgrade from 0.3.5 to 0.6.2. Compile example "AllFunctions" with uncommented SMS_TARGET and CALL_TARGET.

### Actual result
After version upgrade 
ROM consumption rises from 18554 to 21868 (+3314),
RAM consumption rises from 805 to 905  (+100)

For boards with <=32k ROM and <=2k RAM every 100 bytes is a big dial. Many sketches for such boards are tightly packed in the available space and don't have so much memory to spare. I realize that one have to pay for the new functionality, but consider adding an option to disable memory hungry functions.

Brandon2255p, 2019-05-10:
I had this too. Had to fix it at version 0.3.6 to actually be able to upload my firmware.

SRGDamia1, 2019-05-13:
I'm sorry.  I looked into it and it's definitely from me adding a parent class forcing the generation of memory-consuming vtables.  I'm thinking about a way to improve it.

Brandon2255p, 2019-05-16:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=43745

Seems this "issue" will not be resolved on AVR compiler. 

What direction are you thinking of heading regarding this @SRGDamia1 ?

AlexIII, 2019-05-16:
Brandon2255p, this is known problem, yes, but there's a way to mitigate it. I.e. when you have an abstract class but in whole your program you instantiate only one derived class and never downcast it to the base type, the optimizer can see through it and throw out all vtabes. Also there's 'static' inheritance template-based techniques (not applicable in this specific situation).

SRGDamia1, 2019-05-16:
@AlexIII - sounds nice, how?

SRGDamia1, 2019-05-22:
Okay.  v0.7.4 completely removed the virtual class.  The compile size should only be very slightly larger than it was in v0.3.5.

closed 2019-05-22 by SRGDamia1

AlexIII, 2019-05-23:
@SRGDamia1 Are you aware of static inheritance trick? https://en.wikipedia.org/wiki/Template_metaprogramming#Static_polymorphism

SRGDamia1, 2019-05-23:
I'm _aware_ of CRTP, but I haven't yet successfully used it.

For now, I'm satisfied with the pre-processor macros I've set up.  But if you have time to sit and re-factor it with better templates, I'd love a pull request.