vshymanskyy/TinyGSM #281: Connection refused using HTTPClient.h

kotakomputer, 2019-05-12:
TinyGSM version: 
Hardware: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### AT command log

kotakomputer, 2019-05-12:
```
#define TINY_GSM_MODEM_SIM800
#define TINY_GSM_RX_BUFFER 650

HTTPClient http;
http.begin("https://www.yahoo.com");   //Specify destination for HTTP request
int httpCode = http.GET(); 

```
Error: connection refused (http code is -1)
Above code running well using Wifi but error using GSM (while sms working fine). What have caused "Connection Refused"?

SRGDamia1, 2019-05-22:
I can't help without more details.

xvinny, 2019-05-29:
I'm getting the same error. Follow the details:

1. I connect the GPRS:

`
    if (!modem.restart())
    {
        ESP_LOGE("TEST", "Failure restarting modem.");
        return ESP_ERR_INVALID_STATE;
    }

    String modemInfo = modem.getModemInfo();
    ESP_LOGD("TEST", "Modem: %s", modemInfo.c_str());
    ESP_LOGD("TEST", "Waiting network...");

    if (!modem.waitForNetwork())
    {
        ESP_LOGE("TEST", "Network not available...");
        return ESP_ERR_INVALID_STATE;
    }

    if (modem.isNetworkConnected())
    {
        ESP_LOGD("TEST", "Network connected!");
    }
    
    if (!modem.gprsConnect(apn, user, pass))
    {
        ESP_LOGE("TRACKER", "Falha ao conectar no APN [%s].", apn);
        return ESP_ERR_INVALID_STATE;
    }`

2. Then I send the http POST:
`HTTPClient http;
 http.begin(url);
 int httpCode = http.POST(str);`

3. I get the error:
`[D][main.cpp:168] ConnectionMonitor(): Testing the Http POST...
[V][HTTPClient.cpp:224] beginInternal(): url: http://mytestserver.southcentralus.cloudapp.azure.com/
[D][HTTPClient.cpp:265] beginInternal(): host: mytestserver.southcentralus.cloudapp.azure.com port: 80 url: /
[D][main.cpp:40] postPosition(): Testing HTTP Post...
[E][WiFiGeneric.cpp:652] hostByName(): DNS Failed for gapos.centralus.cloudapp.azure.com
[D][HTTPClient.cpp:995] connect(): failed connect to gapos.centralus.cloudapp.azure.com:80
[W][HTTPClient.cpp:1287] returnError(): error(-1): connection refused
`

It looks like HTTPClient is not suitable to use with GSM connection. Maybe there is another requirement. The strange thing is: why the WiFiGeneric.cpp is trying to resolve the DNS if I am using GSM?

SRGDamia1, 2019-05-30:
The ESP32's core [HTTPClient library](https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPClient) is **NOT** the [Arduino HttpClient library](https://github.com/arduino-libraries/ArduinoHttpClient).  The case of the name in the include statement is not the only difference; they have different functionalities.  To add to the confusion, the powers that be in the ESP32 (and ESP8266) Arduino core world haven't complied very well with the general Arduino standards, which has caused a bunch of issues with TinyGSM.

I've upgraded TinyGSM so it is fully compatible with the current ESP32 core, but the you're still going to have problems.  If you want to use the Arduino HttpClient library on an ESP32, along with downloading the correct library, you have to downgrade your ESP32 core to 1.0.1.  Also, please comment on the issue on the ESP32 GitHub page:  https://github.com/espressif/arduino-esp32/issues/2755.

closed 2019-05-30 by SRGDamia1

xvinny, 2019-09-18:
The problem seems to be solved with the new ESP32-Arduino version. Anyone can test and report if it is now working?