vshymanskyy/TinyGSM #283: HTTPClient.h and ArduinoHttpClient.h side by side

tkralj, 2019-05-18:
Hello. I would like to use HTTPClient.h and ArduinoHttpClient.h side by side. It seems that this two libs cannot co-exist together. I'm using HTTPClient.h with ESP32 to GET / POST but unfortunately I cannot use TinyGsm with this HTTPClient also. The problem is when I include ArduinoHttpClient.h i get compile errors:

The code is:
TinyGsmClientSecure client_gsm(modem);
const char server[] = "vsh.pp.ua";
const char resource[] = "/TinyGSM/logo.txt";
const int  port = 443;
HttpClient http = HttpClient(client_gsm, server, port);

And the error i get:

networkgsm:24:54: error: invalid cast to abstract class type 'HttpClient'
 HttpClient http = HttpClient(client_gsm, server, port);

Is there a way to get around it ?

SRGDamia1, 2019-05-18:
What version of the ESP32 core are you using?

It is this problem:  https://github.com/vshymanskyy/TinyGSM/issues/269

tkralj, 2019-05-18:
Hello, no. I get slightly different error (i got the one you suggested but i applied fix toTinyGsmClientSIM800.h) after that i now get:

sketch_may18a:88:12: error: cannot declare variable 'http' to be of abstract type 'HttpClient'

HttpClient http(client, server, port);

I copy-pasted HTTPClient.ino example from GIT

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

SRGDamia1, 2019-05-22:
The compiler problems with the ESP32 and ESP8266 should now be fixed.

closed 2019-05-22 by SRGDamia1

uuperi, 2019-05-23:
> 
> 
> The compiler problems with the ESP32 and ESP8266 should now be fixed.

I'm still getting the error with my ESP32 and SIM868
> cannot declare variable 'http' to be of abstract type 'HttpClient'

SRGDamia1, 2019-05-23:
Ok, sorry, I'm seeing the issue now.  This isn't something TinyGSM can fix.  The ArduinoHTTPClient library just isn't compatible with ESP32 core v1.0.2.  The powers in the ESP32 world added extra virtual functions to their core Client.h that are not in the Arduino standard.  That breaks compatibility with every other type of Arduino library that uses a Client.  I overhauled TinyGSM to make it compatible with v1.0.2, but ArduinoHTTPClient isn't.  You'll have to roll back to 1.0.1, comment out the extra virtual functions as [here](https://github.com/vshymanskyy/TinyGSM/issues/269#issuecomment-485608237), or hand-do the http requests using a web client.

Also, I'd suggest commenting on the issue for the ESP32 core:  https://github.com/espressif/arduino-esp32/issues/2755

tkralj, 2019-05-23:
Hello, or maybe it's something else because i did roll back to esp32 1.0.1 core and  commented out two lines in Client.h and i use tinygsm 0.7.4 but it still does not work 😊 I guess i will have to do it using web client.
But TinyGSM seemed so promissing and realy well to me.

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

Also, I'd suggest commenting on the issue for the ESP32 core: espressif/arduino-esp32#2755<https://github.com/espressif/arduino-esp32/issues/2755>

SRGDamia1, 2019-05-23:
It builds for me TinyGSM 0.7.4, ESP32 core v1.0.1, and ArduinoHttpClient 0.4.0.  It also builds for me if I use ESP32 core v1.0.2 but comment out the two offending lines in Cores/esp32/Client.h.

tkralj, 2019-05-23:
Hello Sara, it builds for me also but it does not work well. It's not problem with build. I can build and run code but code does not run as it should. Instead it loops without getting response even though TinyGSM downloaded the file because i can see it in debug mode.

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

tkralj, 2019-05-24:
Hello Sara, would it be too much trouble to try to make TinyGSM compatible with ESP32 HTTPClient.h also along with ? ArduinoHTTPClient ?
So for example i could code something like this:

TinyGsmClient client(modem);

HTTPClient http(client, server, port); // this is HTTPclient from esp32 libraries folder.

Best regards

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

uuperi, 2019-05-24:
> Hello Sara, would it be too much trouble to try to make TinyGSM compatible with ESP32 HTTPClient.h also along with ? ArduinoHTTPClient ? So for example i could code something like this: [...]

I had the same problem. The build was fine with 1.0.1 and some of the ArduinoHttpClient's functions worked just as desired but when I tried to do a basic http GET-request with TinyGSM's example-sketch it gets stuck and never prints out the response. According to the debug output I believe it has something to do with the fact that ArduinoHttpClient is designed for 16 bit integers and ESP32 uses 32 bit integers.

uuperi, 2019-05-24:
> 
> Hello Sara, it builds for me also but it does not work well. It's not problem with build. I can build and run code but code does not run as it should. Instead it loops without getting response even though TinyGSM downloaded the file because i can see it in debug mode.

I had the same problem. The build was fine with 1.0.1 and some of the ArduinoHttpClient's functions worked just as desired but when I tried to do a basic http GET-request with TinyGSM's example-sketch it gets stuck and never prints out the response. According to the debug output I believe it has something to do with the fact that ArduinoHttpClient is designed for 16 bit integers and ESP32 uses 32 bit integers.

SRGDamia1, 2019-05-24:
I'm sorry, I don't have time for that right now. I am working to solve the bugs with the SIM800.

uuperi, 2019-05-27:
> It's not problem with build. I can build and run code but code does not run as it should. Instead it loops without getting response even though TinyGSM downloaded the file because i can see it in debug mode.

It seems like the problem is not just with the HttpClient, but also with the TinyGSM library since in the Diagnostics-sketch the code uses only TinyGSMClient to do the http-request but still can't print out the response body. Debug log shows the output just as supposed but you can't save it to any variable.

uuperi, 2019-05-27:
> It seems like the problem is not just with the HttpClient, but also with the TinyGSM library since in the Diagnostics-sketch the code uses only TinyGSMClient to do the http-request but still can't [...]

Nevermind, I updated the code

eabase, 2019-07-19:
@uuperi  Which library did you update? I get the same error today, using what is supposed to be the latest of both TinyGSM and ArduinoHTTPClient for an ESP32 based project, testing on the HttpClient.ino (from TinyGSM).

uuperi, 2019-07-19:
@eabase the TinyGSM library. You are using ESP32 board version 1.0.1 right?

eabase, 2019-07-22:
@uuperi 
yes

alexandreberg, 2020-11-19:
Hi,

I am testing with platformIO: PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1]
With the libs
-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0      
|   |   |-- <WiFi> 1.0
|-- <TinyGSM> 0.10.9

And it doesn´t work.
GPRS connected
httpRequestData atualizaGrafico: api_key=xxxxxxxxp&value1=Sensor-01&value2=Sensor-02&value3=Sensor-03
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for example.com
-1

Does anyone have any sugestion to posting http data over GPRS without using the  HTTPClient.h library for ESP32 without have to downgrade the esp32 core?

Thanks.

SRGDamia1, 2020-11-19:
@alexandreberg The issue being discussed here last summer was only an issue with core version 1.0.2, and the issue was related to compiling the library, not using it.

Based on the little bit you posted (ie, the reference to "WiFiGeneric.cpp) I'm guessing you're improperly trying to supply the esp core's wifi libraries with a GSM client.  That will simply not work: https://github.com/vshymanskyy/TinyGSM/issues/422#issuecomment-662543187

genotix, 2021-01-19:
It would be very much appreciated if someone would have a combined example of:
- Usage of the SIM800L
- Doing a POST
- Over HTTPS

I've been trying to combine the ArduinoHTTP requests and the one from TinyGSM but the latter only has a GET and the first seems to have issues working with the TinyGSMSecure variant.

Some help would be greatly appreciated.

alexandreberg, 2021-01-19:
> It would be very much appreciated if someone would have a combined example of:
>
> * Usage of the SIM800L
> * Doing a POST
> * Over HTTPS
>
> I've been trying to combine the ArduinoHTTP requests and the one [...]

Hi this example is working and makes a post with HTTPd, not HTTPS and sends data do a mysql database stored in BlueHost.
It runs on STM32 Bluepill, but works too on ESP32.
You can see de server side following the link bellow in RandomNerdTutorials that explains in detail.

The sketch:
#include <Arduino.h>
/*
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp32-sim800l-publish-data-to-cloud/
*/

// Teste com buzzer
#define buzzer PB5

// Your GPRS credentials (leave empty, if not needed)
const char apn[]      = "timbrasil.br"; // APN (example: internet.vodafone.pt) use https://wiki.apnchanger.org
const char gprsUser[] = "tim"; // GPRS User
const char gprsPass[] = "tim"; // GPRS Password

// SIM card PIN (leave empty, if not defined)
const char simPIN[]   = ""; 

// Server details
// The server variable can be just a domain name or it can have a subdomain. It depends on the service you are using
const char server[] = "yourDomain.com.br"; // domain name: example.com, maker.ifttt.com, etc
const char resource[] = "/data/post-data.php";         // resource path, for example: /post-data.php
const int  port = 80;                             // server port number

// Keep this API Key value to be compatible with the PHP code provided in the project page. 
// If you change the apiKeyValue value, the PHP file /post-data.php also needs to have the same key 
String apiKeyValue = "xxxxxxxxxxxxxx";

#define MODEM_RX PB10
#define MODEM_TX PB11
HardwareSerial Serial3(MODEM_TX, MODEM_RX);

// Set serial for debug console (to Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to SIM800 module)
#define SerialAT Serial3

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

// Define the serial console for debug prints, if needed
#define DUMP_AT_COMMANDS

#include <Wire.h>
#include <TinyGsmClient.h>

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

// TinyGSM Client for Internet connection
TinyGsmClient client(modem);

[... 92 lines cut ...]
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:bluepill_f103c8]   ;STM32duino
platform = ststm32 ;STM32duino
board = bluepill_f103c8 ;STM32duino
framework = arduino
monitor_port = COM5
monitor_speed = 115200
upload_port = COM5
upload_speed = 921600
upload_protocol = serial ;para FTDI (RX - A9, TX - A10, GND - GND, 5V - 5V)
lib_ldf_mode = deep+

lib_deps =
  vshymanskyy/TinyGSM @ 0.10.9
  vshymanskyy/StreamDebugger @ 1.0.1
  adafruit/RTClib @ ~1.12.4

Hope it may help you.

Toumal, 2021-02-05:
> It would be very much appreciated if someone would have a combined example of:
>
> * Usage of the SIM800L
> * Doing a POST
> * Over HTTPS
>
> I've been trying to combine the ArduinoHTTP requests and the one [...]

I have to echo this request. I am running against walls trying to get HTTPS to work in any way.
Yes, HTTP works. Please don't post examples for http.

genotix, 2021-02-05:
> I have to echo this request. I am running against walls trying to get HTTPS to work in any way.
> Yes, HTTP works. Please don't post examples for http.

I have spent over 2,5 week finding a way to get this to work properly.
Eventually I have integrated the SSLClient library with the TinyGSM code.
I advise you to do the same.

Since I developed this for my company, I'm not at liberty to share any code with you.

datamann, 2021-02-05:
@genotix Nice of you to take advantage of other people's work but not to share back. That's the way to keep open source alive and kicking....

genotix, 2021-02-05:
> @genotix Nice of you to take advantage of other people's work but not to share back. That's the way to keep open source alive and kicking....

It isn’t my own time spent on this, it’s my companies time. I release everything I build on my own Github but am not a C++ coder.
Looking at your repo’s you will have this running within the hour with my tip.
I spent 2,5 weeks searching for a proper solution. SSLClient provides one, very easy.

genotix, 2021-02-05:
@datamann, as you can see on my profile I will share sourcecode when I develop something myself or when I extend sources of others. I have been doing so way before Github even existed and I am very well aware of the GNU/GPL.

This is a contribution I did in 2003 on the Gentoo platform:
https://forums.gentoo.org/viewtopic-t-113911.html
Yes, USB was new to Linux then.

As I said, in this case I am not allowed to submit sourcecode, I simply am not at liberty. I contribute my part to the community, just you worry about your own contributions. I have spent reading about 2000 articles for 2,5 weeks and ended up writing a very specific implementation in 2,5 days with the tip I provided above.
(Adding SPIFFS certificate storage and a shitload of other things; the implementation was merely a small percentage)

I just saved you about 15 days, get out of your diapers and start coding.