vshymanskyy/TinyGSM #284: client.read() returns nothing

tkralj, 2019-05-18:
Hello. I am able to connect to  the server and i get client.available() but client.read() reads nothing. Can someone help ?

Here is sample code (connect returns true, available()also but read not):

  TinyGsmClient *client = new TinyGsmClient(modem);

  if (client->connect(downloadServerName.c_str(),80)){
      Serial.println("connected");
      client->setTimeout(cfg.tout);
      client->println("GET "+downloadPath+"cert.der HTTP/1.1");
      client->println("Host: "+downloadServerName);
      client->println("Connection: close");
      client->println();

      delay(2000);

      String fileData ="";
      while (client->available()){
         char c = client->read();    
         Serial.print(c);
         //pazi da nismo dobili kakvo smeće po HTTPu
         if(!isAscii(c))
            continue;

         fileData+=c;
      }

  // Shutdown
  client->stop();
  modem.gprsDisconnect();
  
  Serial.println(F("Server disconnected"));

SRGDamia1, 2019-05-18:
What does your setup look like? What modem are you using? Please post the debugging output and AT log.

tkralj, 2019-05-19:
Hello, here is my setup:
GSMC.begin(115200, SERIAL_8N1, 36, 4); //Baud rate, parity mode, RX, TX SIM800 
StreamDebugger debugger(GSMC, Serial);
TinyGsm modem(debugger);

    if(!modem.restart()){
      Serial.println("Failed to restart GSM modem!");
      return;
    }

    String modemInfo = modem.getModemInfo();
    Serial.print("Modem: ");
    Serial.println(modemInfo);

    if(cfg.pin!=0 && cfg.pin!="")
      modem.simUnlock(cfg.pin);

    if (modem.waitForNetwork()){
      Serial.println("Network found");
      networkFound=true;
      return;
    }
      Serial.print("Connecting to: ");  
      Serial.println(cfg.apn);
      
      if (!modem.gprsConnect(cfg.apn)) {
        eth_connected=false;
      }else{
        eth_connected=true;
        Serial.print("GSM modem connected to APN: ");
        Serial.println(cfg.apn);      
        return;
      }

  // Make a HTTP GET request:
  TinyGsmClient *client = new TinyGsmClient(modem);

  if (client->connect(downloadServerName.c_str(),80)){
      Serial.println("connected");
      client->setTimeout(cfg.tout);
      client->println("GET "+downloadPath+"cert.der HTTP/1.1");
      client->println("Host: "+downloadServerName);
      client->println("Connection: close");
    
      // Wait for data to arrive   
      while (client->connected() && !client->available()) {    
        delay(100);    
        Serial.print('.');    
      };    
      client->println();
      // Skip all headers   
      client->find("\r\n\r\n");
      // Read data
      unsigned long timeout = millis();
      unsigned long bytesReceived = 0;
      while (client->connected() && millis() - timeout < 10000L) {
        while (client->available()) {
          char c = client->read();
          Serial.print(c);
          bytesReceived += 1;
          timeout = millis();
      }
  }

  client->stop();
  delete client;
    
  Serial.println(F("Server disconnected")); 

Now, here's the AT command output:

AT

OK
AT+CLTS=1

OK
AT&W

OK
AT+CFUN=0

+SAPBR 1: DEACT

+PDP: DEACT

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
AT
⸮AT

OK
AT&FZ

OK
ATE0
ATE0

OK
AT+CPIN?

ERROR
AT+CPIN?

RDY

+CFUN: 1

+CPIN: READY

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem: SIM800 R14.18
AT+CPIN=""

ERROR
Waiting for network...
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

Call Ready

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

SMS Ready

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,1

AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","internet"

OK
AT+CGDCONT=1,"IP","internet"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"100.80.71.42"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="internet","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

100.80.71.42

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
AT+CIPCLOSE=1,1

ERROR
AT+CIPSSL=0

OK
AT+CIPSTART=1,"TCP","iot-test.softlab.hr",80

OK

1, CONNECT OK
connected
AT+CIPSEND=1,31

>GET /download/cert.der HTTP/1.1 
DATA ACCEPT:1,31
AT+CIPSEND=1,2

>
 
DATA ACCEPT:1,2
AT+CIPSEND=1,25

>Host: iot-test.softlab.hr 
DATA ACCEPT:1,25
AT+CIPSEND=1,2

>
 
DATA ACCEPT:1,2
AT+CIPSEND=1,17

>Connection: close 
DATA ACCEPT:1,17
AT+CIPSEND=1,2

DATA ACCEPT:1,2
SN:3e7fc55a-451d-3e40-aff8-955aa3f84110
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CONNECTED"

OK
...AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

… and it goes and goes until …

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","REMOTE CLOSING"

OK

1, CLOSED
AT+CIPCLOSE=1,1

ERROR

And also i am having  a problem uploading certificate to the modem, here's the output:

AT

OK
AT&FZ

OK
ATE0
ATE0

OK
AT+CPIN?

+CPIN: READY

OK
AT+FSCREATE=CERT.der

OK
AT+FSWRITE=CERT.der,0,2048,10

>

OK
AT+SSLSETCERT="CERT.der"

OK

+SSLSETCERT: 1

****************************
Setting Certificate: FAILED
****************************

Best regards and hope this helps!

tkralj, 2019-05-19:
And this is with debugging option enabled.  What's really interesting here is line [68766] ### which prints out all headers and content of the file correctly (content of the file is text "hello"). This means that server returned correct response but TinyGSM didn't recognize it that way. Why are this ### unhandled debug outputs and how can I fix this ?

[8420] ### TinyGSM Version: 0.6.2
[9042] ### Modem: SIMCom SIM800
Modem: Call Ready  SIM800 R14.18
Waiting for network... OK
Connecting to wm OK
Performing HTTP GET request... [42542] ### Unhandled: 3
[44059] ### Unhandled: 1
[45596] ### Unhandled: 18
[47121] ### Unhandled: 9
[48639] ### Unhandled: 2
[50161] ### Unhandled: 6
[51699] ### Unhandled: 19
[53217] ### Unhandled: 2
[54746] ### Unhandled: 10
[56264] ### Unhandled: 2
[57796] ### Unhandled: 13
[59314] ### Unhandled: 2
[60843] ### Unhandled: 10
[62361] ### Unhandled: 2
[63882] ### Unhandled: 5
[65400] ### Unhandled: 2
[66918] ### Unhandled: 2
[68648] ### READ: 0 from 1
[68766] ### Unhandled: Content-Type: text/plain
Last-Modified: Sun, 19 May 2019 07:12:31 GMT
Accept-Ranges: bytes
ETag: "203acf3912ed51:0"
Server: Microsoft-IIS/7.5
X-Powered-By: ASP.NET
Date: Sun, 19 May 2019 18:48:59 GMT
Connection: close
Content-Length: 5

hello
OK
[70225] ### Unhandled: +CIPSTATUS: 1,0,"TCP","212.92.212.113","80","REMOTE CLOSING"

OK
[72771] ### Unhandled: +CIPSTATUS: 1,0,"TCP","212.92.212.113","80","REMOTE CLOSING"

OK
[75090] ### Closed:  1

SRGDamia1, 2019-05-22:
Can you try the newest version?

tkralj, 2019-05-23:
Hello, i installed version 0.7.4, & ArduinoHttpclient 0.4.0 but unfortunately, although it looked promissing from the beginning, it didn't work. The result is different than last time (there's no ### Unhadler errors) but library gets stucked in an infinte loop. Here is AT + Debug output of your HttpClient example:

Wait...
Initializing modem...
AT

OK
AT+CLTS=1

OK
AT&W

OK
AT+CFUN=0

+SAPBR 1: DEACT

+PDP: DEACT

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
[14188] ### TinyGSM Version: 0.7.4
AT
⸮[14398] ### Unhandled: ⸮
AT

OK
AT&FZ

OK
ATE0
ATE0

OK
[14560] ### Modem: SIMCom SIM800
AT+CPIN?

ERROR
AT+CPIN?

RDY

+CFUN: 1

+CPIN: READY

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem: SIM800 R14.18
Waiting for network...AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?
[... 492 lines cut ...]

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CLOSED"

OK
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CLOSED"

OK
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,0,"TCP","212.92.212.113","80","CLOSED"

OK
AT+CIPRXGET=4,1

SRGDamia1, 2019-05-25:
I think I've found the issue.  Can you try again with 0.7.7 and let close this if it works now?

tkralj, 2019-05-25:
Hi Sara , i tried newest version. This version cannot recognize the modem nor connect to the network so i cannot test if it gets something from URL. Here's the output from serial monitor (mind you i didnt change code from the last time at all , only updated TinyGSM):

AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
ATI
Modem:
Waiting for network...AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

tkralj, 2019-05-25:
I'm sorry Sara, it seems that mistake was on my side. TinyGSM now gets the data from my URL in expected format. I use ESP core 1.0.2 and ArduinoHttpClient 0.4.0 . Great Job Sara thank you so much !!!!!
Now i will test other samples and we'll se how it goes.For now, once again, great job !!!

OK
Network connected
Connecting to wm
OK
Connecting to iot-test.softlab.hr
OK
Performing HTTP GET request...

Response:
HTTP/1.1 200 OK
Content-Type: text/plain
Last-Modified: Sun, 19 May 2019 07:12:31 GMT
Accept-Ranges: bytes
ETag: "203acf3912ed51:0"
Server: Microsoft-IIS/7.5
X-Powered-By: ASP.NET
Date: Sat, 25 May 2019 06:07:51 GMT
Connection: close
Content-Length: 5

hello

Server disconnected
GPRS disconnected

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

SRGDamia1, 2019-05-30:
@tkralj - can I close this?

tkralj, 2019-05-30:
OK.

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

closed 2019-05-30 by SRGDamia1