vshymanskyy/TinyGSM #286: TinyGSM on ESP32 only working with #define TINY_GSM_RX_BUFFER 2

fbeek, 2019-05-22:
TinyGSM version: 0.7.4
Hardware: ESP32 Dev Kit V1 with Framework 1.0.2 (1.0.1 also tested), Sim800L EVB
HardwareSerial 1 for AT commands.

### Scenario, steps to reproduce
Iam trying to get the diagnostic and mqtt sketch to work. Lab Power Supply with 5V 4A is working nicely, so no VCC drops or resets.

### Expected result
Passes http transfer test with 121 Bytes. And connect to mqtt

### Actual result
I can only receive the complete data at the http test when I set #define TINY_GSM_RX_BUFFER 2. With larger buffer values I can only get a part of the 121 Bytes of the TINYGSM test logo file. I thought the ESP32 should have enough power and ram?

With this settings the MQTT Sketch is also not working.

In the command log there where no errors until the end. So i think it must be something in the ESP32 context.

Does Anyone have this lib working on an esp32

SRGDamia1, 2019-05-22:
You're not getting more than two bytes at a time? That seems pretty unreasonable.

What errors did you get at the end? Please post the AT/errors.

hardmax, 2019-05-22:
Hi,

I have the same problem with the similar hardware (ESP32 with SIM800L). When is conecting to MQTT and I view in server side, conection petition, but this is not finished, and conecction is dropped.

Change library to 0.3.5 and apply this workarround #269  to compile, and works all againg....

SRGDamia1, 2019-05-23:
AT log? Debug prints?

fbeek, 2019-05-23:
Hello together,

here is my full log with a Buffer Size of 256:

Initializing modem...[4031] ### TinyGSM Version: 0.7.4
AT

OK
AT&FZ

OK
ATE0

OK
[4066] ### Modem: SIMCom SIM800
AT+CPIN?
ATE0

OK

+CPIN: READY

OK
 [OK]
ATI

SIM800 R14.18

OK
Modem: SIM800 R14.18
Waiting for network...AT+CREG?

+CREG: 0,5

OK
 [OK]
Connecting to pinternet.interkom.deAT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","pinternet.interkom.de"

OK
AT+CGDCONT=1,"IP","pinternet.interkom.de"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.177.109.126"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

[... 711 lines cut ...]

+CIPSTATUS: 1,0,"TCP","212.237.59.104","80","REMOTE CLOSING"

OK
[32606] ### Unhandled: +CIPSTATUS: 1,0,"TCP","212.237.59.104","80","REMOTE CLOSING"

OK
AT+CIPCLOSE=1,1

1, CLOSE OK
Server disconnected
AT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

OK
GPRS disconnected

************************
 Received: 121 bytes
 Test:     PASSED
************************

fbeek, 2019-05-23:
I tried the solution from @hardmax and it works, but this cannot be the solution.

I think the newer version has some nice fixes and additions.

SRGDamia1, 2019-05-23:
I think this issue and #284 are different faces of the same problem.  I don't think it's something related to the ESP32.

I've modified the read to de-increment the sock_available as we read out, which should reduce/eliminate the number of ```fewer bytes confirmed than requested!``` problems.

I'd changed the read function to not attempt to read more characters than the response confirmed were available, but it seems the SIM800 occasionally lies about how much is available.  I still need to think about how to deal with it.  That didn't happen when I did some quick tests with the SIM800 before pushing the update.  I'll try and test again somewhere with 2G service.

tkralj, 2019-05-24:
Hello, i tried different approaches and all of them resulted in different output for the same code. I changed only TINY_GSM_RX_BUFFER from 1, 2, 16 and 1024 and each value produced different output from screen full of question marks to nothing at all . TinyGSM 0.7.4 ESP32 core 1.0.1. i used WebClient example for testing purposes.

this is the output if i set buffer to 1024:
Initializing modem...
Modem: SIM800 R14.18
Waiting for network... OK
Network connected
Connecting to wm
 OK
Connecting to iot-test.softlab.hr
 OK
Performing HTTP GET request...
⸮
Server disconnected
GPRS disconnected

this is the output if i set buffer to 2:
Initializing modem...
Modem: SIM800 R14.18
Waiting for network... OK
Network connected
Connecting to wm
 OK
Connecting to iot-test.softlab.hr
 OK
Performing HTTP GET request...
HTTP/1.1 200 OK
Content-Type: text/plain
Last-Modified: Sun, 19 May 2019 07:12:31 GMT
Accept-Ranges: bytes
ETag: "203acf3912ed51:0"
Server: Microsoft-IIS/7.5
X-Powered-By: ASP.NET
Date: Fri, 24 May 2019 11:20:33 GMT
Connection: close
Content-Length: 5

hell⸮
Server disconnected
GPRS disconnected

(almost perfect, but misses "o" as in hello)

This is the output if i set buffer to 16:

Initializing modem...
Modem: SIM800 R14.18
Waiting for network... OK
Network connected
Connecting to wm
 OK
Connecting to iot-test.softlab.hr
 OK
Performing HTTP GET request...
HTTP/1.1 200 OK
Content-Type: text/plain
Last-Modified: Sun, 19 May 2019 07:12:31 GMT
Accept-Ranges: bytes
ETag: "203acf3912ed51:0"
Server: Microsoft-IIS/7.5
X-Powered-By: ASP.NET
Date: Fri, 24 May 2019 11:22:32 GMT
Connection: close
Content-Len⸮
Server disconnected
GPRS disconnected

SRGDamia1, 2019-05-25:
I'm sorry, guys.  I've (hopefully) caught the issues now.  I think it was all in the SIM800 and probably not on the ESP32 side.  Can you try again with 0.7.7 and let me know how you fare?

fbeek, 2019-05-25:
Good Morning,

on monday i am back at home and will test the new version.

Thanks for your help.

Regards

hardmax, 2019-05-25:
Its Working........
Nice job...Thanks

Reagards

SRGDamia1, 2019-05-30:
@fbeek - can I close this?

fbeek, 2019-05-31:
Hello, 

sorry for my late response. 
You can close this issue, its working now.

Many thanks to you for your work.

Regards
fbeek

closed 2019-05-31 by SRGDamia1