vshymanskyy/TinyGSM #288: HTTPS POST Timeout

tkralj, 2019-05-25:
Hello. I am using TinyGSM lib v 0.79 ESP32 core 1.0.2. and ArduinoHttpClient 0.4.0. I am trying to post some data to my WebAPI. I am sending data over SSL. When i call WebAPI endpoint using GET method everything works fine, i get response from WebAPI. When i call WebAPI using POST method the request times out. I checked IIS log and it seems that POST never got to IIS.

Here's the code:

  HttpClient https(client, server, port);
  https.connectionKeepAlive(); // Currently, this is needed for HTTPS
  https.beginRequest();

    client.print(String(F("POST /ssm/MyEndPoint?tid=a347a95c-b91f-48ae-8a36-0fcd58ca2313&time=1970.05.09 08:18:59&amount=2.00&rsnd=false&code=0000 HTTP/1.1")));
    client.print(String(F("\r\nContent-Type: text/plain")));
    client.print(String(F("\r\ncontent-length: 0")));
    client.print(String(F("\r\nUser-Agent: MyAgent")));
    client.print(String(F("\r\nAccept: */*")));
    client.print(String(F("\r\nhost: iot-test.softlab.hr")));
    client.print(String(F("\r\napi-key: 112fc55a-451d-3e40-aff8-955aa3f84157")));
    client.print(String(F("\r\nAuthorization: "))+ AuthString);
    client.print(String(F("\r\n\r\n")));
    client.print(String(""));

  https.endRequest();

  int statusCode = https.responseStatusCode();
  res = https.responseBody();

  https.stop();

Here's TinyGSM debug output:
Initializing modem...
[7340] ### TinyGSM Version: 0.7.9
[7540] ### Unhandled: ⸮⸮
[7655] ### Modem: SIMCOM SIM800L
[7655] ### Modem: SIMCOM SIM800L
Modem: SIM800 R14.18
Waiting for network... OK
Network connected
Connecting to wm OK
Performing HTTPS POST request... [22518] ### Available: 0 on 1
[23538] ### Available: 0 on 1
[24558] ### Available: 0 on 1
[25578] ### Available: 0 on 1
[26596] ### Available: 0 on 1
[27614] ### Available: 0 on 1
[28633] ### Available: 0 on 1
[29650] ### Available: 0 on 1
[30668] ### Available: 0 on 1
[31685] ### Available: 0 on 1
[32704] ### Available: 0 on 1
[33721] ### Available: 0 on 1
[34739] ### Available: 0 on 1
[35757] ### Available: 0 on 1
[36775] ### Available: 0 on 1
[37792] ### Available: 0 on 1
[38809] ### Available: 0 on 1
[39827] ### Available: 0 on 1
[40845] ### Available: 0 on 1
[41863] ### Available: 0 on 1
[42881] ### Available: 0 on 1
[43899] ### Available: 0 on 1
[44917] ### Available: 0 on 1
[45934] ### Available: 0 on 1
[46953] ### Available: 0 on 1
[47971] ### Available: 0 on 1
[48988] ### Available: 0 on 1
[50006] ### Available: 0 on 1
[51023] ### Available: 0 on 1
[52041] ### Available: 0 on 1
[53059] ### Available: 0 on 1
[54077] ### Available: 0 on 1
[55095] ### Available: 0 on 1
[56113] ### Available: 0 on 1
[57131] ### Available: 0 on 1
[58148] ### Available: 0 on 1
[59165] ### Available: 0 on 1
[60183] ### Available: 0 on 1
[61200] ### Available: 0 on 1
[62218] ### Available: 0 on 1
[63236] ### Available: 0 on 1
[64253] ### Available: 0 on 1
[65271] ### Available: 0 on 1
[66289] ### Available: 0 on 1
[67307] ### Available: 0 on 1
[68325] ### Available: 0 on 1
[69343] ### Available: 0 on 1
[70361] ### Available: 0 on 1
[71379] ### Available: 0 on 1
[72397] ### Available: 0 on 1
[73415] ### Available: 0 on 1
[74433] ### Available: 0 on 1
[75450] ### Available: 0 on 1
[76468] ### Available: 0 on 1
[77485] ### Available: 0 on 1
[78503] ### Available: 0 on 1
[79521] ### Available: 0 on 1
[80539] ### Available: 0 on 1
[81557] ### Available: 0 on 1
[82574] ### Available: 0 on 1
[83591] ### Available: 0 on 1
[84092] ### Available: 0 on 1
Status:
-3
Response:

Server disconnected
GPRS disconnected

tkralj, 2019-05-25:
One thing I noticed regardless of method POST or GET. You should never send empty char in your query string. For example if you "GET /api/MyMethod?name=some name" the GET will fail. You should always url encode querystring so it replaces empty chars with %20 . Maybe it could be implemented in the TinyGSMClient so it does url encoding ?

tkralj, 2019-05-25:
It turned out to be my mistake. If you send POST request you have to set Content-Length header (even there is no content) for this to work.

closed 2019-05-25 by SRGDamia1

fedy0, 2019-06-08:
@tkralj  Please what firmware version are you using? (To check, use AT+GSV?)