vshymanskyy/TinyGSM #289: [SIM800] DateTime AutoSync doesn't work properly in restart() method

Warlib1975, 2019-05-26:
Hardware: SIM800L

### Scenario, steps to reproduce
- Execute restart() method of TimyGSM lib.
- Execute the command "AT+CCLK?".
- Got the default DateTime "04/01/01,00:00:40+12" instead of current one.  

### Expected result
After reset() the modem I supposed to get current DateTime. 

### Actual result
'''
AT+CCLK?
+CCLK: "04/01/01,00:00:40+12"
OK

Check either GSM modem in auto sync mode
AT+CLTS?
+CLTS: 0
OK
'''
### Solution
The code in the TinyGSM lib is:

'''
bool restart() {
    ...
    //Enable Local Time Stamp for getting network time
    // TODO: Find a better place for this
    sendAT(GF("+CLTS=1"));
    if (waitResponse(10000L) != 1) {
      return false;
    }
    sendAT(GF("&W"));
    waitResponse();
    sendAT(GF("+CFUN=0"));
    if (waitResponse(10000L) != 1) {
      return false;
    }
    sendAT(GF("+CFUN=1,1"));
    if (waitResponse(10000L) != 1) {
      return false;
    }
    ...
  }
'''

I wrote an additional method which I execute after the restart() and it works ok. 
'''
void SetAutoTimeSync()
{
  Serial.println("Enable GSM RTC network time sync.");
  sendAT(GF("+CLTS=1"));
  if (waitResponse(10000L) != 1) 
  {
      Serial.println("AT command \"AT+CLTS=1\" executing fault.");;
  }
  
  sendAT(GF("&W"));
  waitResponse();

  sendAT(GF("+CFUN=0"));
  if (waitResponse(10000L) != 1) 
  {
      Serial.println("AT command \"AT+CFUN=0\" executing fault.");;
  }
  
  //delay(3000);
  sendAT(GF("+CFUN=1"));
  //if (modemGSM.waitResponse(10000L) != 1) 
  if (waitResponse(10000L, GF(GSM_NL "DST:")) != 1)
  {
    Serial.println("AT command \"AT+CFUN=1\" executing fault.");;
  }

  Serial.println("DateTime: "+ GetDateTime());
}

String GetDateTime()
{
  sendAT(GF("+CCLK?"));
  if (waitResponse(10000L, GF(GSM_NL "+CCLK:")))
  {
      String body = stream.readStringUntil('\n');
      return body;
  }
  else
  {
    Serial.println("AT command \"+CCLK?\" executing fault.");
  } 
  return "";
}
'''

shodan8192, 2019-09-25:
Hello,
there's no need to have additional method, the cause and solution are described in #188. I have faced the same problem, and commenting out factory reset code in init() function in TinyGsmClientSIM800.h gives expected result.

leenowell, 2020-01-19:
Hi All,

I too have the same issue but have tried commenting out the 2 lines mentioned above and still have the same issue.  I have checked that the network is connected using isNetworkConnected and it confirms it is connect.   Just to be double sure, I updated the init method to look like this

`
```
  bool init(const char* pin = NULL) {
    DBG(GF("### TinyGSM Version:"), TINYGSM_VERSION);

    if (!testAT()) {
      return false;
    }
//    sendAT(GF("&FZ"));  // Factory + Reset
//    waitResponse();
    sendAT(GF("E0"));   // Echo Off
    if (waitResponse() != 1) {
      return false;
    }

    DBG(GF("### Modem:"), getModemName());

    int ret = getSimStatus();
    // if the sim isn't ready and a pin has been provided, try to unlock the sim
    if (ret != SIM_READY && pin != NULL && strlen(pin) > 0) {
      simUnlock(pin);
      return (getSimStatus() == SIM_READY);
    }
    // if the sim is ready, or it's locked but no pin has been provided, return true
    else {
      return (ret == SIM_READY || ret == SIM_LOCKED);
    }
  }
```
Version on the Anyone have any idea why this is not working for me? 

thanks in advance

Lee.

shodan8192, 2020-01-20:
@leenowell 
Hi, after commenting out those 2 lines, did You call at least once restart() method, or set manually AT+CLTS=1 and save settings by AT&W, then restart modem ?

leenowell, 2020-01-20:
@shodan8192 

Thanks very much for getting back to me. Essentially I power up the SIM800 and then call restart.  The initialisation code is as follows 

`
```
void initGSM()
{
  // Keep power when running from battery
  Wire.begin(I2C_SDA, I2C_SCL);
  bool isOk = setPowerBoostKeepOn(1);
  SerialMon.println(String("IP5306 KeepOn ") + (isOk ? "OK" : "FAIL"));

  // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(3000);

  // Restart SIM800 module, it takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();
  // use modem.init() if you don't need the complete restart

  SerialMon.println("Done restart");

  // Unlock your SIM card with a PIN if needed
  if (strlen(simPIN) && modem.getSimStatus() != 3 ) {
    modem.simUnlock(simPIN);
  }
  SerialMon.println("Done initGSM");

}
```
`

I then wait for the network to be connected and run my various tests.  Not sure if it is relevant or related but I also seem to always get 601 for the GSM Location which I believe is an error code for "network error"? 

Thanks again for your help with this. 
Lee.

shodan8192, 2020-01-20:
@leenowell 
To get location by CIPGSMLOC=1,1 function which is TinyGSM using, You have to be connected to GPRS, I believe.
It's rather not related to problem with getting time from cellular network.

leenowell, 2020-01-20:
@shodan8192 

Ah that might be the issue. I believe when I checked GPRS was not connected for some reason. This is also on my list to try and get to the bottom of. Doing a quick Google looks like 2g does support GPRS so should in theory work? 

Once I get the library working correctly I hope to the implement receive SMS (which I need for my project) and hopefully contribute it back 😀 Got to get this working first though.

SRGDamia1, 2020-04-15:
I think this has all been addressed.

closed 2020-04-15 by SRGDamia1

federicobriata, 2021-04-27:
@SRGDamia1 with the described scenario and step to reproduce I'm still able to reproduce this issue with current TinyGSM lib and a Sim800L.
I've also tried to downgrade to tag: v0.10.3 (commit fb3f8583a37423f97bc1b4216250fc92f72e9919) to avoid regression but still the same problem.
As NTP it's not working, due to UDP not implemented, https://github.com/vshymanskyy/TinyGSM/issues/116
Can you help to fix this issue? thanks

thanks

SRGDamia1, 2021-04-28:
The current `init` method for the SIM800 is calling `AT+CLTS=1` so it should be getting the local time.  See https://github.com/vshymanskyy/TinyGSM/blob/9498800b50baa46c1a0b3f6d0df68cbd3230c025/src/TinyGsmClientSIM800.h#L178.  If you turn on the AT dump, you should see that command go by.  I'm not sure how else to enable local time syncing.

federicobriata, 2021-04-28:
@SRGDamia1 I've checked again, with TINY_GSM_DEBUG turned on, testing getNetworkTime() and getGSMDateTime() from AllFunctions.ino I get only this information:

`[834] ### Unhandled: BU
[6639] ### TinyGSM Version: 0.10.9
[6639] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
[6950] ### Modem: SIMCOM SIM800L
[6950] ### Modem: SIMCOM SIM800L
[14562] Requesting current network time
[14568] Year: 2004 	Month: 1 	Day: 1
[14568] Hour: 0 	Minute: 0 	Second: 13
[14568] Timezone: 0.00
[14568] Retrieving time again as a string
[14578] Current Network Time: 04/01/01,00:00:13+00`

SRGDamia1, 2021-04-28:
I guess it's possible that your network/sim card doesn't support local time syncing.  I'm not sure how to force a manual time sync.  Check the AT manual.