vshymanskyy/TinyGSM #291: MQTT ISSUE - SIM808 R14.18 not updating !mqtt.connn

DynamiteDavePE, 2019-05-26:
TinyGSM version: 0.62 (problem seems to be the same in every other version after)
Hardware: SIM808 R14.18 in conjunction with the Arduino Mega, SIM808 board version 2.2.9.1

### Scenario, steps to reproduce
I am trying to use the SIM808 and the Mega to connect to our MQTT server. I am struggling to get the connection to reflect in the code. I have added the code from the example that I am working with. I end up getting stuck in what appears to be some kind of loop.

  // MQTT Broker setup
  mqtt.setServer(broker, port);
  mqtt.setCallback(mqttCallback);
}

boolean mqttConnect() {
  Serial.print("Connecting to ");
  Serial.print(broker);
 mqtt.connect(clientID,username,password);

  if (!mqtt.connect(clientID,username,password)){
    Serial.println("Fail");
    return false;
  }
  Serial.println(" OK");
  mqtt.publish(topic, "GsmClientTest started");
  //mqtt.subscribe(topic);
  return mqtt.connected();
}

void loop() {

  if (!mqttConnect()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
     //Reconnect every 10 seconds
   unsigned long t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
   delay(100);
    return;
  }

  mqtt.loop();
}

### Expected result
To connect without issue. Allow for the uploading of GPS data and subscribe to specific topics that will trigger events on the Arduino.

### Actual result
Stuck in an endless loop of failed connection that will not allow the code to progress.
[Log 2.txt](https://github.com/vshymanskyy/TinyGSM/files/3220277/Log.2.txt)

### AT command log
[Log.txt](https://github.com/vshymanskyy/TinyGSM/files/3220268/Log.txt).

SRGDamia1, 2019-05-30:
Please try again with 0.7.9 and post that log.  Make sure you do ```#define DUMP_AT_COMMANDS``` and ```#define TINY_GSM_DEBUG Serial```

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1