vshymanskyy/TinyGSM #292: Using TinyGsmClient and TinyGsmClientSecure at the same time

tkralj, 2019-05-26:
Hello how can i use both Http and Https client in the same ino file.

If i declare both clients like this GET and POST over SSL wont work any more:
TinyGsmClientSecure clientHttps(modem);
TinyGsmClient clientHttp  TinyGsmClient(modem);

SRGDamia1, 2019-05-26:
To have more than one client of any type, you need to create them on different sockets. All of the chips except the XBee support it.

```TinyGsmClientSecure clientHttps(modem, 0);
TinyGsmClient clientHttp TinyGsmClient(modem, 1);```

closed 2019-05-26 by SRGDamia1

eabase, 2019-07-23:
This should have been made more clear **somewhere**...
Is there a Wiki for helping people implement new devices (or parts of)?

eabase, 2019-07-23:
@SRGDamia1 
I tried to find the place to include this in the TinyGsmClientSIM5360.h, but I don't see where this should be placed.  Any suggestions?