vshymanskyy/TinyGSM #293: Not able to establish MQTT Connection - AI Thinker A6

Ohmion, 2019-05-26:
TinyGSM version: 0.7.9
Hardware: Arduino Mega 2560 + Ai Thinker Co.LTD A6  V03.06.20171127R

### Scenario, steps to reproduce
MqttClient example from TinyGSM

[Arduino Code ](https://github.com/Ohmion/A6issues/blob/master/MQTTClient.ino)

### Expected result
Establish connection with the MQTT server
After few days trying to solve, I can't find a solution, that's why I'm asking for help. 

### Actual result

https://github.com/Ohmion/A6issues/blob/master/ResponseCSTT

### AT command log - FTDI or Arduino Serial
I can establish connection with the MQTT server directly with AT Commands.

[AT Log WithoutCSTT](https://github.com/Ohmion/A6issues/blob/master/withoutCSTT)

My apn has user and password, I should use AT+CSTT to register it, but then this happends (following same squency from tinyGSM A6 library).

[AT Log WithCSTT](https://github.com/Ohmion/A6issues/blob/master/withCSTT)

SRGDamia1, 2019-05-30:
So your A6 locks up after getting the AT+CSTT command, whether you send it manually or it is sent by TinyGSM?

Ohmion, 2019-05-30:
It's sent by TinyGSM. The sequence is inside the gprsConnect function in the TinyGsmClientA6.h file

```
289    if (!user) user = "";
290    if (!pwd)  pwd = "";
291    sendAT(GF("+CSTT=\""), apn, GF("\",\""), user, GF("\",\""), pwd, GF("\""));
292    if (waitResponse(60000L) != 1) {
293      return false;
294    }
```

To check why it fails, I send it manually and verify that the CIPSTATUS = 0, IP START (when I use CSTT) and the CIPSTART command is not accepted (returning COMMAND NO RESPONSE!).

So my A6 locks up in both cases.

However, if I delete the CSTT command line, it establishes a connection perfectly, but it doesn't login in to my MQTT server.

SRGDamia1, 2019-05-30:
The user name and password are for your cellular service provider, not for the MQTT connection, right? 

I'm confused which of your logs goes with which program or set of modifications or manual commands. The first log you posted labeled "ResponseCSTT" doesn't have the CSTT command in it at all; it looks like the whole gprsConnect sequence is missing. The second log "Response No CSTT" looks much better, but it seems like it's miss-hearing an OK from the modern. It did succesfully open the TCP connection, but it didn't recognize it so it tried to reopen and failed. It never attempted to connect to MQTT. Your "Without CSTT" (manual?) log looks the almost identical; there's no MQTT connection, just TCP.

Ohmion, 2019-05-30:
You are correct, the username and password used in CSTT are from the provider, not for MQTT. According to the specifications, my mobile service provider requires a username and password, however, I have managed to establish the connection without using the credentials.

The "Response" files are from tinyGSM and "AT LOG" are from the manual AT sequence.

"ResponseNoCSTT" is the response I get from SerialMon when I delete the CSTT sequence from TinyGsmClientA6.h.

"ResponseCSTT" is the answer I get from SerialMon with the default library. Here, my A6 module locks up.

I also use username and password for MQTT. Using the AT commands manually I can send the connection packet (in hexadecimal) once I have established the TCP connection (it is not shown in the log).

However, as you can see in "ResponseNoCSTT" even though it manages to establish TCP connection, the PubSubClient library is not able to send the connection packet.

English is not my first language, so please excuse any mistakes.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1