vshymanskyy/TinyGSM #294: Ordering of calls responseStatusCode and responseBody

tkralj, 2019-05-28:
Hello, I'm not sure if it is a bug but i will report it anyway. It looks like it is not all the same if i call responseBody first and then responseStatusCode  or other way around. If I call responseBody first i get response body. If i call responseStatusCode   afer responseBody  it always returns -3.

If i first call responseStatusCode   then I get correct status . The responseBody  works as expected no matter when i call it.

SRGDamia1, 2019-05-28:
Are you sure this is an issue with TinyGSM and not the HttpClient library?

tkralj, 2019-05-28:
Not sure, it could be both ways either TinyGsmClient or HttpClient. I posted issues on boht githubs.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1