vshymanskyy/TinyGSM #295: http.get crashes if resource is unavailable

tkralj, 2019-05-28:
Hello, i encountered a problem with getting the file from server if server is unavailable or file does not exists on the server. http.get call returns 0 and crashes later on with guru meditation error:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060530  A0      : 0x800fd4e0  A1      : 0x3ffb1630  
A2      : 0x00000194  A3      : 0x00000190  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000025  
A10     : 0x3ffb1aa5  A11     : 0x3f402e68  A12     : 0x3ffb1aca  A13     : 0x3f403f6e  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000194  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffb1630 0x400fd4dd:0x3ffb1640 0x400fb57d:0x3ffb1950 0x400d7bfe:0x3ffb1a10 0x400d7d96:0x3ffb1df0 0x400d7fc3:0x3ffb1e50 0x400da5c6:0x3ffb1e70 0x400e0e3f:0x3ffb1fb0 0x40089fc1:0x3ffb1fd0

Here's the code:
    TinyGsmClient clientHttp(modem);
    HttpClient http(clientHttp, downloadServerName, 80);

    http.beginRequest();
    int err = http.get(downloadPath+fname);
    Serial.println("return");
    http.endRequest();
    Serial.print("END request:");
    Serial.println(err);

// here it crashes
    int statusCode = http.responseStatusCode();     
    String body = http.responseBody();

SRGDamia1, 2019-05-30:
I suspect this problem is on the HttpClient side, not TinyGSM.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1