vshymanskyy/TinyGSM #296: Ublox SARA-G450

jpmeijers, 2019-05-28:
Default example does not work. Still debugging this. Will update this issue when I found the cause.

TinyGSM version: 
Hardware: 

### Scenario, steps to reproduce

HttpClient example

### Expected result

### Actual result

### AT command log

```
Initializing modem...
AT
AT

OK
AT+CFUN=16
AT+CFUN=16

OK
AT

+UUPSDD: 0

 [8219] ## Wait timeout: 200
[8219] ### Unhandled: +UUPSDD: 0

 
AT
[8519] ## Wait timeout: 200
AT
[8819] ## Wait timeout: 200
AT
[9119] ## Wait timeout: 200
AT
[9419] ## Wait timeout: 200
AT
[9719] ## Wait timeout: 200
AT
AT

OK
ATE0
AT

OK
AT+CMEE=2
AT

OK
AT+CGMI
AT

OK
AT+GMM
AT

OK
[9826] ### Modem: AT AT
[9826] This is NOT a CatM
AT+CPIN?
AT

OK
ATE0

OK

OK

u-blox

OK

SARA-G450

OK
[10827] ## Wait timeout: 1000
[10827] ### Unhandled: AT

OK
ATE0

OK

OK

u-blox

OK

SARA-G450

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?
AT+CPIN?

+CPIN: READY

OK
ATI
AT+CPIN?

+CPIN: READY

OK
Modem: AT+CPIN?  +CPIN: READY
Waiting for network...AT+CGREG?
ATI

SARA-G450-00C-00

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK
 OK
Connecting to internetAT+UPSDA=0,4
AT+UPSDA=0,4

ERROR
AT+CGATT=1
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3
AT+UPSDA=0,3

OK
AT+UPSND=0,8
AT+UPSND=0,8

+UPSND: 0,8,0

OK
[23744] ## Wait timeout: 1000
[23744] ### Unhandled: AT+UPSND=0,8

+UPSND: 0,8,0

OK
 fail connecting gprs
Waiting for network...AT+CGREG?

+UUPSDA: 0,"100.65.214.75"

AT+CGREG?

+CGREG: 0,1

OK
 OK
Connecting to internetAT+UPSDA=0,4
AT+UPSDA=0,4

OK
AT+CGATT=0
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3
AT+UPSDA=0,3

OK
AT+UPSND=0,8
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"41.151.137.17"

[36374] ## Wait timeout: 1000
[36374] ### Unhandled: AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"41.151.137.17"
 fail connecting gprs

Waiting for network...AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK
 OK
Connecting to internetAT+UPSDA=0,4
AT+UPSDA=0,4

OK
AT+CGATT=0
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3
AT+UPSDA=0,3

OK
AT+UPSND=0,8
AT+UPSND=0,8

+UPSND: 0,8,0

OK
[49133] ## Wait timeout: 1000
[49133] ### Unhandled: AT+UPSND=0,8

+UPSND: 0,8,0

OK
 fail connecting gprs
Waiting for network...AT+CGREG?

+UUPSDA: 0,"100.69.86.150"

AT+CGREG?

+CGREG: 0,1

OK
 OK
Connecting to internetAT+UPSDA=0,4
AT+UPSDA=0,4

OK
AT+CGATT=0
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3
AT+UPSDA=0,3

OK
AT+UPSND=0,8
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"100.67.47.101"

[61839] ## Wait timeout: 1000
[61839] ### Unhandled: AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"100.67.47.101"
 fail connecting gprs
Waiting for network...AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK
 OK
Connecting to internetAT+UPSDA=0,4
AT+UPSDA=0,4

OK
AT+CGATT=0
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3
AT+UPSDA=0,3

OK
AT+UPSND=0,8
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"102.249.97.239"

[74579] ## Wait timeout: 1000
[74579] ### Unhandled: AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"102.249.97.239"
 fail connecting gprs

```

SRGDamia1, 2019-05-30:
Are you sure you have the right APN?

jpmeijers, 2019-05-30:
Yes, the APN is correct.

Issue 1:
When sending `AT+UPSDA=0,3` the modem responds with `OK` directly. Then one should wait for the unsolicited `+UUPSDA: 0,"100.89.183.48"` before sending `AT+UPSND=0,8`.

Issue 2:
After sending `AT+USOCO=` the modem directly responds with `OK` and responds with another `OK` when the TCP connection is connected. This library only waits for the first `OK`.

Issue 3:
After opening the TCP socket and writing data there is still an issue with reading the response. Still working on this.

Because it seems like the SARA-G450 responds slightly differently to the same AT commands than the other u-blox modems, I decided to create a dedicated `TinyGsmClientSaraG450.h` file to test with and modify the algorithm. I'll file a PR when its working.

jpmeijers, 2019-05-30:
Sometimes the modem restart sequence is strange. I see an extra `AT` that should not be in the log.
```
ATE0
AT

OK
```

Full log for `modem.restart();`
```
Switching on modem
AT
AT

OK
AT+CFUN=16
AT+CFUN=16

OK
[5019] ### TinyGSM Version: 0.7.9
AT
 AT
AT
AT
AT

OK
ATE0
AT

OK
AT+CMEE=2
AT

OK
AT+CGMI
AT

OK
AT+GMM
ATE0

OK
[6102] ### Modem: AT ATE0
[6103] ### WARNING:  You are using the wrong TinyGSM modem!
AT+CPIN?

OK

u-blox

OK

SARA-G450

OK
[7104] ### Unhandled: OK

u-blox

OK

SARA-G450

OK
AT+CPIN?

+CPIN: READY

OK
ATI
AT+CPIN?

+CPIN: READY

OK
Modem: AT+CPIN?  +CPIN: READY
ATI

SARA-G450-00C-00

OK
```

jpmeijers, 2019-05-30:
Increased TINY_GSM_YIELD to 100ms. Now also clearing my Serial RX buffer after doing testAT(). Seems to work ok, except for ATE0's affect being reversed when doing ATI.

```
Switching modem ON
AT

OK
AT+CFUN=16

OK
[7575] ### TinyGSM Version: 0.7.9
AT
 AT
AT
AT
AT
AT

OK
AT

OK
AT

OK
AT

OK
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+CGMI

u-blox

OK
AT+GMM

SARA-G450

OK
[11280] ### Modem: u-blox SARA-G450
AT+CPIN?

+CPIN: READY

OK
ATI
ATI

SARA-G450-00C-00

OK
Modem: ATI  SARA-G450-00C-00
```

See the ATI being printed twice above? Yes, that's the issue. The modem echo's ATI back even though we turned echoing off using ATE0.

jpmeijers, 2019-05-30:
Next issue is that the library does not seem to use the `+UUSORD` unsolicited response. But we are not in asynchronous mode, so should we be receiving this? But anyways, we get an error if we try to do `AT+USORD=0,0` after this because the socket is closed.

```
AT+USORD=0,0

+USORD: 0,0,""

OK
AT+USOCTL=0,10

+USOCTL: 0,10,4

OK
+UUSORD: 0,551

+UUSOCL: 0
AT+USORD=0,0

+CME ERROR: Operation not allowed
AT+USOCTL=0,10

+CME ERROR: Operation not allowed
[40019] ### Unhandled: Operation not allowed
AT+USORD=0,0

+CME ERROR: Operation not allowed
```

SRGDamia1, 2019-05-30:
The +UUSORD URC is used; see https://github.com/vshymanskyy/TinyGSM/blob/7c06a4cb9af822db599c46bfc9135aa976185c4f/src/TinyGsmClientUBLOX.h#L700

The extra echo's actually shouldn't be that much of a problem, as long as the 'OK' follows afterwards.  The extra characters will just be skipped in the waitResponse.

Yes, you'll get an error if you try to poll a closed socket (for status or buffer content), but, again, the library should just move on.

Do you have level shifters or translaters in your custom board?  Are you able to make connections properly when you manually enter the AT commands instead of using TinyGSM?

Per the manual, the G450 follows the "standard" command set, unlike the R4/N4/N2.

I'm looking into the missed check for +UUPSDA.

jpmeijers, 2019-05-30:
I got the library so far as to catch the +UUPSDA now, by calling waitForResponse() before modemGetAvailable() in maintain(). It also catches the +UUSOCL.

For some reason the modem responds with the  `+CME ERROR: Operation not allowed` when trying to read data from the socket after it was close. But fine, this is to be expected. So the question would be: Why does the socket get closed so early?

I made a setup where everything is done by the library up to the connection with GPRS. Then I send the HTTP GET AT commands by hand using serial passthrough. When I did this the HTTP GET works perfectly and I do not get the +UUSOCL after the +UUPSDA. So the socket remains connected. The funny thing is I used exactly the same sequence of AT commands as the library.

```
GPRS connect OK

OK
AT+USOCTL=0,10

+CME ERROR: Operation not allowed
AT+USOCR=6

+USOCR: 0

OK
AT+USOSO=0,6,1,1

OK
AT+USOCO=0,"google.co.za",80

OK

OK
AT+USOWR=0,3
@ 
> GET
+USOWR: 0,3

OK
AT+USOWR=0,83
@ 
>  / HTTP/1.1
>
> Host: google.co.za
> User-Agent: Arduino/2.2.0
> Connection: close
> 
> 

+USOWR: 0,83

OK
+UUSORD: 0,1400
AT+USORD=0,0

+USORD: 0,22400

OK
AT+USOCTL=0,10

+USOCTL: 0,10,4

OK
AT+USORD=0,1000

+USORD: 0,1000,"HTTP/1.1 200 OK
Date: Thu, 30 May 2019 16:49:58 GMT
Expires: -1
Cache-Control: private, max-age=0
Content-Type: text/html; charset=ISO-8859-1
P3P: CP="This is not a P3P policy! See g.co/p3phelp for more info."
Server: gws
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN
Set-Cookie: 1P_JAR=2019-05-30-16; expires=Sat, 29-Jun-2019 16:49:58 GMT; path=/; domain=.google.com
Set-Cookie: NID=184=oSlpYOBfDm1E1IC2sStjL-dRZQ4QYS__Bxn4m86OQMWlcfPPbgjze62bKHRBjGEar7PPahqus4efIK4m280QC0Hm_7WPquHo_--8DqIFkqLueNCNeQVStdzLSIJdTdOVFsDMESUJmgEfnQjCMyY4cyvSmVRLNfOwe2mxf40XiNo; expires=Fri, 29-Nov-2019 16:49:58 GMT; path=/; domain=.google.com; HttpOnly
Accept-Ranges: none
Vary: Accept-Encoding
Transfer-Encoding: chunked

5286
<!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="en-ZA"><head><meta content="text/html; charset=UTF-8" http-equiv="Content-Type"><meta content="/logos/doodles/2019/2019-icc-cricket-world-cup-begins-6508349483384832-law.gif" itemprop="image"><"

OK

+UUSORD: 0,22040

```

SRGDamia1, 2019-05-30:
Can you test out this version:  https://github.com/EnviroDIY/TinyGSM

Based on the response your posting, the response of the U201 I have, and the manual.. I can't figure out which modules will return immediate OK to the ```AT+UPSDA=0,3``` followed by ```+UUPSDA=0[,...]``` on activation and which will wait to return the first ```OK``` until the activation is complete and never send the URC.  The SARA U2 doesn't seem to bother with the URC.  But I think either way you should be able to query the PSD profile status with ```AT+UPSND=0,8``` right after the activation attempt and then wait for the URC based on whether the connection is already complete or not.

jpmeijers, 2019-05-30:
Using https://github.com/EnviroDIY/TinyGSM and the HttpClient example from that repo. Back at one of my original issues:

```
Switching modem ON
Wait...
Initializing modem...
AT

OK
AT+CFUN=16

OK
[8412] ### TinyGSM Version: 0.7.9
AT

+UUPSDD: 0

 [8713] ### Unhandled: +UUPSDD: 0
AT
AT
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+CGMI

OK
AT+GMM

OK
[10616] ### Modem:  
AT+CPIN?

OK

u-blox

OK

SARA-G450

OK
[11720] ### Unhandled: OK

u-blox

OK

SARA-G450

OK
AT+CPIN?

+CPIN: READY

OK
ATI

+CPIN: READY

OK
Modem: +CPIN: READY
Waiting for network...AT+CGREG?

SARA-G450-00C-00

OK

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CGREG?

+CGREG: 0,1

OK
 OK
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to internetAT+UPSDA=0,4

+CME ERROR:AT+CGATT=1
 Unspecified GPRS error

+CGATT:1

OK
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3

OK
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"100.70.64.95"

[24841] ### Unhandled: +UPSND: 0,8,0

OK

+UUPSDA: 0,"100.70.64.95"
 fail
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 OK
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to internetAT+UPSDA=0,4

OK
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3

OK
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"100.84.56.7"

[39158] ### Unhandled: +UPSND: 0,8,0

OK

+UUPSDA: 0,"100.84.56.7"
 fail
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 OK
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to internetAT+UPSDA=0,4

OK
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3

OK
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"100.89.214.21"

[54274] ### Unhandled: +UPSND: 0,8,0

OK

+UUPSDA: 0,"100.89.214.21"
 fail
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 OK
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to internetAT+UPSDA=0,4

OK
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3

OK
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"100.70.206.4"

[69391] ### Unhandled: +UPSND: 0,8,0

OK

+UUPSDA: 0,"100.70.206.4"
 fail
Waiting for network...AT+CGREG?

+CGREG: 0,1

OK
 OK
AT+CGREG?

+CGREG: 0,1

OK
Network connected
Connecting to internetAT+UPSDA=0,4

OK
AT+CGATT=0

+UUPSDD: 0

+CGATT:0

OK
AT+CGATT=1

+CGATT:1

OK
AT+UPSD=0,1,"internet"

OK
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3

OK
AT+UPSND=0,8

+UPSND: 0,8,0

OK

+UUPSDA: 0,"41.145.150.123"

[83708] ### Unhandled: +UPSND: 0,8,0

OK

+UUPSDA: 0,"41.145.150.123"
 fail
```

SRGDamia1, 2019-05-30:
That's not what I expected.  I'm sorry, can you uncomment the debug on lines 755 and 756 and re-run?  I'm not sure why it's missing the ,8,0 of the UPSND response.

jpmeijers, 2019-05-31:
Line 755 of TinyGsmClientUBLOX.h? That file only has 753 lines, so I'm not sure what you are referring to.

jpmeijers, 2019-05-31:
Scrap that. Seems like I was still using the upstream version :confused: 

Current output:
```
Switching modem ON
Wait...
Initializing modem...
AT

OK
[5811] < 1 > /OK/
AT+CFUN=16

OK
[6612] < 1 > /OK/
[9612] ### TinyGSM Version: 0.7.10
AT
 [9912] < 0 > 
AT
[10312] < 0 > 
AT

OK
[11212] < 1 > /OK/
ATE0

OK
[12012] < 1 > /OK/
AT+CMEE=2

OK
[12813] < 1 > /OK/
AT+CGMI

OK
[13614] < 1 > /OK/
AT+GMM

OK
[14415] < 1 > /OK/
[14415] ### Modem: /OK/ /OK/
AT+CPIN?

u-blox

OK

SARA-G450

OK

+CPIN:[18916] < 1 > /u-blox//OK//SARA-G450//OK//+CPIN:
 READY[19616] < 1 >  READY

OK
[20516] < 1 > //OK/
ATI

SARA-G450-00C-00

OK
[23316] < 1 > /SARA-G450-00C-00//OK/
Modem: /SARA-G450-00C-00//OK/
Waiting for network...AT+CGREG?

+CGREG:[24417] < 1 > /+CGREG:
 0,1

OK
[25117] < 1 > /OK/
 OK
AT+CGREG?

+CGREG:[26218] < 1 > /+CGREG:
 0,1

OK
[26918] < 1 > /OK/
Network connected
Connecting to internetAT+UPSDA=0,4

+CME ERROR:[28419] < 3 > /+CME ERROR:
AT+CGATT=1
 Unspecified GPRS error

+CGATT:1

OK
[32920] < 1 >  Unspecified GPRS error//+CGATT:1//OK/
AT+UPSD=0,1,"internet"

OK
[33722] < 1 > /OK/
AT+UPSD=0,7,"0.0.0.0"

OK
[34524] < 1 > /OK/
AT+UPSDA=0,3

OK
[35325] < 1 > /OK/
AT+UPSND=0,8

+UPSND: 0,8,0[37026] < 2 > /+UPSND: 0,8,0

OK
[37926] < 1 > //OK/

+UUPSDA: 0[39426] < 1 > //+UUPSDA: 0
,"100.68.179.224"
 OK
Performing HTTP GET request... 
[39826] < 0 > 
AT+USOCTL=0,10

+CME ERROR:[41427] < 3 > /+CME ERROR:
AT+USOCR=6
 Operation not allowed

+USOCR:[45028] < 1 >  Operation not allowed//+USOCR:
 0

OK
[45728] < 1 > /OK/
AT+USOSO=0,6,1,1

OK
[46529] < 1 > /OK/
AT+USOCO=0,"vsh.pp.ua",80

OK
[48431] < 1 > /OK/

OK
[49231] ### Unhandled: OK
[49231] < 0 > 
AT+USOWR=0,3
@[49532] < 1 > @
GET 
+USOWR:[50682] < 1 >  /+USOWR:
 0,3

OK
[51382] < 1 > /OK/
AT+USOWR=0,1
@[51783] < 1 > @
  
+USOWR:[52933] < 1 >  /+USOWR:
 0,1

OK
[53633] < 1 > /OK/
AT+USOWR=0,17
@[54034] < 1 > @
/TinyGSM/logo.txt 
+USOWR:[55185] < 1 >  /+USOWR:
 0,17

OK
[55885] < 1 > /OK/
AT+USOWR=0,9
@[56286] < 1 > @
 HTTP/1.1 
+USOWR:[57436] < 1 >  /+USOWR:
 0,9

OK
[58136] < 1 > /OK/
AT+USOWR=0,2
@[58537] < 1 > @

 
+USOWR:[59687] < 1 >  /+USOWR:
 0,2

OK
[60387] < 1 > /OK/
AT+USOWR=0,6
@[60788] < 1 > @
Host:  
+USOWR:[61938] < 1 >  /+USOWR:
 0,6

OK
[62638] < 1 > /OK/
AT+USOWR=0,9
@[63039] < 1 > @
vsh.pp.ua 
+USOWR:[64189] < 1 >  /+USOWR:
 0,9

OK
[64889] < 1 > /OK/
AT+USOWR=0,2
@[65290] < 1 > @

 
+USOWR:[66440] < 1 >  /+USOWR:
 0,2

OK
[67140] < 1 > /OK/
AT+USOWR=0,10
@[67541] < 1 > @
User-Agent 
+USOWR:[68691] < 1 >  /+USOWR:
 0,10

OK
[69391] < 1 > /OK/
AT+USOWR=0,2
@[69792] < 1 > @
:  
+USOWR:[70942] < 1 >  /+USOWR:
 0,2

OK
[71642] < 1 > /OK/
AT+USOWR=0,13
@[72043] < 1 > @
Arduino/2.2.0 
+USOWR:[73194] < 1 >  /+USOWR:
 0,13

OK
[73894] < 1 > /OK/
AT+USOWR=0,2
@[74295] < 1 > @

 
+USOWR:[75445] < 1 >  /+USOWR:
 0,2

OK
[76145] < 1 > /OK/
AT+USOWR=0,10
@[76546] < 1 > @
Connection 
+USOWR:[77696] < 1 >  /+USOWR:
 0,10

OK
[78396] < 1 > /OK/
AT+USOWR=0,2
@[78797] < 1 > @
:  
+USOWR:[79947] < 1 >  /+USOWR:
 0,2

OK
[80647] < 1 > /OK/
AT+USOWR=0,5
@[81048] < 1 > @
close 
+USOWR:[82198] < 1 >  /+USOWR:
 0,5

OK
[82898] < 1 > /OK/
AT+USOWR=0,2
@[83299] < 1 > @

 
+USOWR:[84449] < 1 >  /+USOWR:
 0,2

OK
[85149] < 1 > /OK/
AT+USOWR=0,2
@[85550] < 1 > @

 
+USOWR:[86700] < 1 >  /+USOWR:
 0,2

OK
[87400] < 1 > /OK/
AT+USORD=0,0
+UUSORD: 0,633

+UUSOCL: 0
[90301] ### URC Sock Closed:  0

+CME ERROR:[91601] < 3 > /+CME ERROR:
 Operation not allowed
AT+USOCTL=0,10

+CME ERROR:[93103] < 3 > /+CME ERROR:
 Operation not allowed
[95603] ### Unhandled: Operation not allowed
[95603] < 0 > 
AT+USORD=0,0

+CME ERROR:[98204] < 3 > /+CME ERROR:
 Operation not allowed
AT+USOCTL=0,10

+CME ERROR:[99706] < 3 > /+CME ERROR:
 Operation not allowed
[102206] ### Unhandled: Operation not allowed
[102206] < 0 > 
AT+USORD=0,0

+CME ERROR:[104807] < 3 > /+CME ERROR:
 Operation not allowed
AT+USOCTL=0,10

+CME ERROR:[106309] < 3 > /+CME ERROR:
 Operation not allowed
[108809] ### Unhandled: Operation not allowed
[108809] < 0 > 
AT+USORD=0,0
```

SRGDamia1, 2019-05-31:
Do you still have the a very long yield set? Otherwise, I'll look into the slow down.

jpmeijers, 2019-05-31:
```
// Add a reception delay, if needed
#define TINY_GSM_YIELD() { delay(100); }
```

But there is a strange delay of almost a second between the SerialMon printed characters. Looks like a type writer slowly typing the characters one by one.

jpmeijers, 2019-05-31:
Anyway, we have been moving way off topic now. I'm going to switch back to my own G450 specific code to find the cause of the real issue. Your fork is better than the main one, but it only solves the first issue, not the numerous other differences the standard Ublox code doesn't handle correctly.

SRGDamia1, 2019-05-31:
The u-blox code is written for the AT manual that includes the G450, so it really should work with it.

The type-writer effect is because of that delay.  That yield runs between each character to make sure nothing's come in before it exits the routine.  If that were to happen you might end up splicing the URC's and getting a two unhandled halves of it instead.  But you shouldn't need it at all unless you have a particularly fast processor and slow baud rate combination.  And even then, 1ms should be plenty.  All that delay is probably also the reason you didn't manage to read the logo out of the socket buffer before it closed.  The data did come in.

jpmeijers, 2019-06-01:
Your explanation makes a lot of sense, so using your version of the library I changed the delay to 1ms. The error still occurs.

I however do not think this is correct:
> All that delay is probably also the reason you didn't manage to read the logo out of the socket buffer before it closed.

Because when I manually send the same set of commands to the modem using a serial pass through sketch the timing is very slow but it does work. That makes me think it's rather too fast using the library and the socket maybe gets closed due to a firmware bug?

This G450 modem is quite new so I am really starting to think the issues I've been seeing are firmware bugs. I'm waiting for a response from our local u-blox agent with an update on this.

```
09:10:50.564 -> Switching modem ON
09:10:50.564 -> 
09:10:51.558 -> Wait...
09:10:54.549 -> Initializing modem...
09:10:54.549 -> AT
09:10:54.549 -> 
09:10:54.549 -> OK
09:10:54.549 -> [5032] < 1 > /OK/
09:10:54.549 -> AT+CFUN=16
09:10:54.582 -> 
09:10:54.582 -> OK
09:10:54.582 -> [5054] < 1 > /OK/
09:10:57.572 -> [8054] ### TinyGSM Version: 0.7.10
09:10:57.572 -> AT
09:10:57.572 -> 
09:10:57.605 -> 
09:10:57.605 -> +UUPSDD: 0
09:10:57.605 -> 
09:10:57.605 -> 
09:10:57.605 -> +UUSOCL: 0
09:10:57.605 -> [8084] ### URC Sock Closed:  0
09:10:57.605 ->  [8255] < 0 > 
09:10:57.871 -> AT
09:10:58.070 -> [8556] < 0 > 
09:10:58.170 -> AT
09:10:58.402 -> [8857] < 0 > 
09:10:58.502 -> AT
09:10:58.701 -> [9158] < 0 > 
09:10:58.801 -> AT
09:10:59.000 -> [9459] < 0 > 
09:10:59.100 -> AT
09:10:59.299 -> [9760] < 0 > 
09:10:59.399 -> AT
09:10:59.598 -> [10061] < 0 > 
09:10:59.698 -> AT
09:10:59.897 -> [10362] < 0 > 
09:10:59.997 -> AT
09:11:00.063 -> 
09:11:00.063 -> OK
09:11:00.063 -> [10524] < 1 > /OK/
09:11:00.063 -> ATE0
09:11:00.063 -> 
09:11:00.063 -> OK
09:11:00.063 -> [10532] < 1 > /OK/
09:11:00.063 -> AT+CMEE=2
09:11:00.063 -> 
09:11:00.063 -> OK
09:11:00.063 -> [10541] < 1 > /OK/
09:11:00.063 -> AT+CGMI
09:11:00.063 -> 
09:11:00.063 -> OK
09:11:00.096 -> [10550] < 1 > /OK/
09:11:00.096 -> AT+GMM
09:11:00.096 -> 
09:11:00.096 -> OK
09:11:00.096 -> [10559] < 1 > /OK/
09:11:00.096 -> [10559] ### Modem: /OK/ /OK/
09:11:00.096 -> AT+CPIN?
09:11:00.096 -> 
09:11:00.096 -> OK
[... 767 lines cut ...]
09:11:45.475 -> +CME ERROR:[55965] < 3 > /+CME ERROR:
09:11:45.475 ->  Operation not allowed
09:11:45.475 -> AT+USOCTL=0,10
09:11:45.509 -> 
09:11:45.509 -> +CME ERROR:[55995] < 3 > /+CME ERROR:
09:11:45.509 ->  Operation not allowed
09:11:45.542 -> [56020] ### Unhandled: Operation not allowed
09:11:45.542 -> [56020] < 0 > 
09:11:46.538 -> Response status code: -3
09:11:46.538 -> Response Headers:
09:11:46.538 -> AT+USORD=0,0
09:11:46.572 -> 
09:11:46.572 -> +CME ERROR:[57046] < 3 > /+CME ERROR:
09:11:46.572 ->  Operation not allowed
09:11:46.572 -> AT+USOCTL=0,10
09:11:46.605 -> 
09:11:46.605 -> +CME ERROR:[57076] < 3 > /+CME ERROR:
09:11:46.605 ->  Operation not allowed
09:11:46.638 -> [57101] ### Unhandled: Operation not allowed
09:11:46.638 -> [57101] < 0 > 
09:11:47.037 -> AT+USORD=0,0
09:11:47.070 -> 
09:11:47.070 -> +CME ERROR:[57547] < 3 > /+CME ERROR:
09:11:47.070 ->  Operation not allowed
```

jpmeijers, 2019-06-01:
https://github.com/ARMmbed/mbed-os/issues/9712 makes me also think it's a firmware version related issue, or just an issue with this modem.

My modem reports the following as response to both AT+GMR and AT+CGMR:
```
12:41:48.245 -> 09.02
12:41:48.245 -> 
12:41:48.245 -> OK
```

jpmeijers, 2019-06-02:
The library is searching for a carriage return - line feed before "+UUSORD:" on line 730: 
`} else if (data.endsWith(GF(GSM_NL "+UUSORD:"))) {`
But at that moment if you print out the data buffer there is no newline in front, but it starts with the `+`. So I edited it to look like this:
`} else if (data.endsWith(GF("+UUSORD:"))) {` 
and for good measure the +UUSOCL one too: 
`} else if (data.endsWith(GF("+UUSOCL:"))) {`
After this change `+UUSORD` is detected correctly.

Also during debugging it seems like the HTTP library sends a header `Connection: close\r\n`. When there is a long enough delay between opening the socket and sending this header the socket does not get closed before one can read the data. That's why it seems to work when sending the AT commands by hand using serial pass through. Changing the code to not send this header seems to keep the socket open and one can successfully make the HTTP GET. But now you need to manually close the socket or wait for it to time out.

This again is contradictory to what the AT command reference manual says about `+UUSOCL`:

> 25.10.1, Some notes about the TCP sockets, bullet 2:
> If the connection is closed by the remote host, the +UUSOCL URC is not sent until all received data is
> read using the AT+USORD command.

Small subset of code that does work:
```
  modem.sendAT("+USOCTL=0,10");
  modem.waitResponse();
  modem.sendAT("+USOCR=6");
  modem.waitResponse();
  modem.sendAT("+USOSO=0,6,1,1");
  modem.waitResponse();

// clear the serial read buffer so that we are sure we wait for the correct two "OK"s
  while(SerialAT.available()) { SerialAT.read(); }
  modem.sendAT("+USOCO=0,\"vsh.pp.ua\",80");
  modem.waitResponse(10000, "OK");
  modem.waitResponse(10000, "OK");

  String toSend = "GET /TinyGSM/logo.txt HTTP/1.1\r\n";
  modem.sendAT("+USOWR=0,"+String(toSend.length()));
  modem.waitResponse("@");
  SerialAT.print(toSend);
  modem.waitResponse();
  
  toSend = "Host: vsh.pp.ua\r\n";
  modem.sendAT("+USOWR=0,"+String(toSend.length()));
  modem.waitResponse("@");
  SerialAT.print(toSend);
  modem.waitResponse();
  
  toSend = "User-Agent: Arduino/2.2.0\r\n";
  modem.sendAT("+USOWR=0,"+String(toSend.length()));
  modem.waitResponse("@");
  SerialAT.print(toSend);
  modem.waitResponse();
  
//  toSend = "Connection: close\r\n";
//  modem.sendAT("+USOWR=0,"+String(toSend.length()));
//  modem.waitResponse("@");
//  SerialAT.print(toSend);
//  modem.waitResponse();
//  
  toSend = "\r\n";
  modem.sendAT("+USOWR=0,"+String(toSend.length()));
  modem.waitResponse("@");
  SerialAT.print(toSend);
  modem.waitResponse();
  
  modem.waitResponse(10000);
  modem.sendAT("+USORD=0,0");
```

SRGDamia1, 2019-06-03:
Removed the new-line in from the URC.

From what I've seen with the SARA U201 and SARA R410 (which follows a very similar but not _quite_ the same command manual) as soon as you get a ```+UUSOCL``` URC or send a ```+USOCL``` any further queries about the socket (connection status, data available, etc) return ```+CME ERROR: Operation not allowed```.  Anything left unread from the buffer is gone.  So if the ```+USSOCL``` is returned immediately after the ```+UUSORD``` you're kind-of SOL.  I'm not sure how to work with that other than to recommend always using the keep-alive header.  Of the websites that I've tried, though, the only one that's given me trouble has been connecting to the NIST daytime server.  Every modem I've tried has trouble with that sometimes because NIST sends four characters and immediately closes.  Half the time they register the close before the characters come in.  (Happens with the XBee's, SIM800/900, Sequans, ESP8266, etc..)

Not one of the modules I've played with has actually followed their manual exactly, but yours is especially bad.

jpmeijers, 2019-06-04:
Also asked a question on the Ublox site:
https://portal.u-blox.com/s/question/0D52p00008WS6JuCAL/sarag450-at-responses-different-than-documented

jpmeijers, 2019-06-13:
Our local distributor was able to update my module to a newer "testing" firmware version.

```
14:47:24.701 -> AT+GMR
14:47:24.701 -> 
14:47:24.701 -> 09.03
14:47:24.701 -> 
14:47:24.701 -> OK
14:47:33.593 -> AT+CGMR
14:47:33.593 -> 
14:47:33.593 -> 09.03
14:47:33.593 -> 
14:47:33.593 -> OK
```

This solves many of the issues I had, but mainly the socket closing issue that was a killer for this product. Apparently the firmware file may not be distributed so the dealer had to do the upgrade for me.

closed 2019-06-13 by jpmeijers

clonedesign, 2020-10-27:
> Yes, the APN is correct.
>
> Issue 1:
> When sending `AT+UPSDA=0,3` the modem responds with `OK` directly. Then one should wait for the unsolicited `+UUPSDA: 0,"100.89.183.48"` before sending [...]

Sorry to post on closed but can you share your TinyGsmClientSaraG450.h? I have exactly same problems (and sadly the 09.02 FW). Thanks.

jpmeijers, 2020-10-27:
You can try my fork (WIP before I PR) here: https://github.com/jpmeijers/TinyGSM

jpmeijers, 2020-10-27:
I would recommend you upgrade to at least 09.03, as the older versions have serious issues preventing data transfers via TCP.

clonedesign, 2020-10-27:
Thanks a lot!