vshymanskyy/TinyGSM #297: Keeps on retrying to read socket even when closed

jpmeijers, 2019-06-03:
TinyGSM version: 0.7.9
Hardware: SARA-G450

Related issue: https://github.com/vshymanskyy/TinyGSM/issues/296

### Scenario, steps to reproduce

Use the HttpClient example. Make sure the AT+USORD fails.

### Expected result

The HttpClient library should stop retrying when the socket is closed or when any error was returned.

### Actual result

There is an indefinite retry so the firmware gets "stuck".

### AT command log

```
+USOWR:[86700] < 1 >  /+USOWR:
 0,2

OK
[87400] < 1 > /OK/
AT+USORD=0,0
+UUSORD: 0,633

+UUSOCL: 0
[90301] ### URC Sock Closed:  0

+CME ERROR:[91601] < 3 > /+CME ERROR:
 Operation not allowed
AT+USOCTL=0,10

+CME ERROR:[93103] < 3 > /+CME ERROR:
 Operation not allowed
[95603] ### Unhandled: Operation not allowed
[95603] < 0 > 

... (keeps on repeating AT+USOCTL and AT+USOCRD)
```

SRGDamia1, 2019-06-03:
Well, this *shouldn't* happen.  Once the sock is closed, maintain should ignore it.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1