vshymanskyy/TinyGSM #298: cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClientSecure'

khvalov, 2019-06-06:
TinyGSM version: Latest
Hardware: esp8266 + SIM800L
Trying to run example but it's fail with: 

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Client.h:38:22: note: 	virtual bool Client::stop(unsigned int)

         virtual bool stop(unsigned int maxWaitMs = 0) = 0;

                      ^

exit status 1
cannot declare variable 'client' to be of abstract type 'TinyGsmSim800::GsmClientSecure'

SRGDamia1, 2019-06-06:
Bah.  I missed this when fixing the other virtual functions.  I'll try to get to it.

It's another face of several other issues related to the core of the ESP's not properly following the "official" Arduino API.

SRGDamia1, 2019-06-11:
I went back to look at this and I don't need to fix this.  You need to update your ESP8266 core from 2.5.0 to 2.5.2 (or 2.5.1).

closed 2019-06-11 by SRGDamia1

p-efford, 2025-01-04:
I updated ESP32 core (3.1.0) and it has the same error.
I have had to revert to version 3.0.7 to get it working properly again.