vshymanskyy/TinyGSM #299: Cannot connect to GPRS when using roaming SIM

tkralj, 2019-06-12:
Hello, it seems i cannot establish a connection when using roaming SIM. If i use my local SIM it works. Here's debugger output of AT commands. Please can someone help ?

Initializing GSM modem...
AT+CPIN?
02"

OK

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN=""

ERROR
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","wm"

OK
AT+CGDCONT=1,"IP","wm"

OK
AT+CGACT=1,1

ERROR
AT+SAPBR=1,1

ERROR
AT+SAPBR=2,1

+SAPBR: 1,3,"0.0.0.0"

OK
AT+CGATT=1

ERROR
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","wm"

OK
AT+CGDCONT=1,"IP","wm"

OK
AT+CGACT=1,1

ERROR
AT+SAPBR=1,1

ERROR
AT+SAPBR=2,1

+SAPBR: 1,3,"0.0.0.0"

OK
AT+CGATT=1

ERROR
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","wm"

OK
AT+CGDCONT=1,"IP","wm"

OK
AT+CGACT=1,1

ERROR
AT+SAPBR=1,1

ERROR
AT+SAPBR=2,1

+SAPBR: 1,3,"0.0.0.0"

OK
AT+CGATT=1

ERROR
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","wm"

OK
AT+CGDCONT=1,"IP","wm"

OK
AT+CGACT=1,1

ERROR
AT+SAPBR=1,1

ERROR
AT+SAPBR=2,1

+SAPBR: 1,3,"0.0.0.0"

OK
AT+CGATT=1

ERROR
AT+CGATT?

+CGATT: 0

OK
Cannot establish GPRS connection. Please check SIM card and APN name.
AT+CGATT?
02"

OK

+CGATT: 0

OK

SRGDamia1, 2019-06-12:
What modem and code are you using?  Did you try waiting for the network? You should wait until you are registered on the network before trying to attach to GPRS.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1