vshymanskyy/TinyGSM #303: How to decrease MTU size

tkralj, 2019-06-13:
Hello, can someone give me some insight on what is the MTU size tinyGSM uses to send packets of data ? i am having problems with SSL and my network provider adviced me to decrease MTU size to 1400.

SRGDamia1, 2019-06-13:
TinyGSM isn't really setting the MTU.  I believe each chip handles that on its own.  The longest "chunk" that TinyGSM sends at once is the size of the longest single String or string you send with a single write function.  Unless you're creating really long strings in your code, I doubt you're anywhere close to the 1400 limit.  Is your network provider actually watching traffic from your chip and seeing some unusually huge data units flying around or are they just throwing out random suggestions?

tkralj, 2019-06-14:
Hello, it may be the case during SSL handshake that MTU reaches the limit. I must check with provider if they limit the traffic. But interesting part is that it worked a week ago and now it wont work any more.

[softLabLogo87x25]
Tomislav Kralj
Voditelj razvoja
+385 091 724-0724 (mob)
tomislav@softlab.hr
Softlab obrt
S. Radića 16, 10430 Samobor
Croatia
www.softlab.hr

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1