vshymanskyy/TinyGSM #304: Bug in TinyGsmCommon.h

tkralj, 2019-06-21:
Hello, i've found a bug (probably) in TinyGsmCommon.h in function getSimCCID :

// Gets the CCID of a sim card via AT+CCID
#define TINY_GSM_MODEM_GET_SIMCCID_CCID() \
  String getSimCCID() { \
    sendAT(GF("+CCID")); \
    if (waitResponse(GF(GSM_NL "+CCID:")) != 1) { \
      return ""; \
    } \
    String res = stream.readStringUntil('\n'); \
    waitResponse(); \
    res.trim(); \
    return res; \
  }

The problem is in first waitResponse :
    if (waitResponse(GF(GSM_NL "+CCID:")) != 1) { \

this way function call allways returns empty string.

if we rewrite waitResponse line to:
      if (waitResponse(GF(GSM_NL)) != 1) { \

now function call returns ID as it should. Can you confirm the bug ?

SRGDamia1, 2019-06-22:
It should work as written. The expected response is +CCID and then the string, not just the naked string. Is it not working for you in tests? Please show code / output.

SRGDamia1, 2020-04-15:
Fixed SIM800 to wait only for the new line and then cut out the "CCID:" later if it's there.

closed 2020-04-15 by SRGDamia1