vshymanskyy/TinyGSM #308: Git Repository is too big

eabase, 2019-07-09:
Most probably because of the aged fork of this repository, it has grown huge. 
This not a real problem, but quite annoying when one just want to clone and run. 

Any ideas how to reduce the size? Do you really need to keep the entire git history?
I doubt anyone cares what happened in the code > 2 years ago...

SRGDamia1, 2019-07-10:
The size is some from age and some from pdf's hanging out in the docs folder.  Old commits could be pruned; the pdf's could all be cleaned out with something like BFG Repo Cleaner (https://rtyley.github.io/bfg-repo-cleaner/).  But doing so would mean everyone who has forked this repo and submitted (or plans to submit) a pull request would need to re-clone/re-base afterwards.  Given there are currently 264 forks of this repo, I'm not so sure that's a great idea.

If your problem is just that it takes to long to download and you don't want to fork and make changes, use a shallow clone,

eabase, 2019-07-16:
Yep, should benefit from shallow clone. 
Nothing to do here...

closed 2019-07-16 by eabase