vshymanskyy/TinyGSM #309: Simcom SIM7600(H)

shajek, 2019-07-12:
Is any plan to implement this modem ? is compatibile witch sim7000 ?
thank you

SRGDamia1, 2019-07-12:
I don't have one.  If you want to mail me a USA/North America brand break-out I might be able to take a look at it, but I'm not planning to get one otherwise.

You can try running it as a SIM7000 or a SIM800 and post back if it works.

eabase, 2019-07-12:
@shajek Where did you buy it? 
If the chipset is Qualcomm based, it will be compatible with 7100E, 7500E, 7600E and probably also with 5360E, 800, 900 and some others.

shajek, 2019-07-16:
yes is Qualcomm  MDM9607. I get it from aliexpress in [http://www.and-global.com/index.php/product/SIM7600E%20H%20Breakout.html](url) this breakboard. I will try it as 800/7000, or another simcom stuff. Thank you

eabase, 2019-07-16:
@shajek  I've already tried as 800/7000 and will post some of the work tomorrow.  Don;t waste your time on those, and rather try to help with https://github.com/vshymanskyy/TinyGSM/issues/189

eabase, 2019-07-17:
BTW. The SIM5360 is based on the very old Qualcomm MDM6200... Gobi stuff

shajek, 2019-07-18:
here is my today trying ... [https://pastebin.com/L0Uj7wXE](url)
i take @SRGDamia1 fork with 5360, delete som unnessery stuff for 7600 .... it is little step, because i cannot get work httpclient with this output [https://pastebin.com/gZXN4hyb](url)

what can be with it ?

i would like to help with 5360, but i dont have one :(

SRGDamia1, 2019-07-19:
It wasn't waiting long enough for the TCP connection to complete.  Everything else looked like it worked.  (Yay!)  Update and try again.

shajek, 2019-07-20:
Yes and no ... I was completly out of mind from that http client, until i try to connect with AT commands from SIMCOM (tcp/ip datasheet) - again unsuccessful. All end with +CIPCLOSE : 0,4.  Then i try all possible and impossible commands without success. Then i check FW version, i have B8 and newest was B12... after update magically all AT commands from SIMCOM starts working and i successfully connect to my server. Now is time to tune TinyGSM, because there is still errors

AT+NETCLOSE  (+NETCLOSE: 2)
AT+CSOCKSETPN=1 (OK)
AT+CIPMODE=0 (OK)
AT+NETOPEN (OK          +NETOPEN: 0)
AT+CIPOPEN=0,"TCP","some.domain.com",80 (OK    +CIPOPEN: 0,0)
AT+CIPSEND=0,22 (OK   +CIPSEND: 0,22,22   xyz      +IPCLOSE: 0,1)
AT+NETCLOSE

shajek, 2019-07-20:
still same problem 
```
AT+CIPRXGET=1

OK
AT+CIPOPEN=0,"TCP","X.X.X.X",80
AT+CIPOPEN=0,"TCP","X.X.X.X",80

OK

+CIPOPEN: 0,0
[69816] ### Unhandled: 0,0
failed to connect

```

eabase, 2019-07-22:
Try the FileDownload example. I just got this working for the SIM5360 so you should have a look at that. The problem for me, was that there were too many settings being executed, while the module was smart enough to use the right ones without messing with them. So I got connection with only 3-4 lines. (AT commends that is.)

shajek, 2019-07-22:
yes, as i say,  I write 4 AT command and i was allready connect , but tinygsm have toughts. Anyway ... here is [FileDownload](https://pastebin.com/ni3pzqBb)
But still no luck with MQTT/HTTP wich is my goal (LTE GW)

PS: here is [http client](https://pastebin.com/6vqLcFXT) with same result

eabase, 2019-07-22:
@SRGDamia1  Could this be a similar HTTP Client issue as you mentioned the other day? 

@shajek What HTTP client library are you using? (It may not be compatible with the ESP32 if that's what you have.)

shajek, 2019-07-22:
[arduino-libraries/ArduinoHttpClient](https://github.com/arduino-libraries/ArduinoHttpClient) for now with 8266 (2.4.2 and latest 2.5.x)

shajek, 2019-07-22:
some progress... is this [helpful ](https://pastebin.com/546ch7V1)?

eabase, 2019-07-23:
Looks pretty good, you should post your code so that it can be implemented and added to supported modems. PR?

shajek, 2019-07-24:
OK. I am making on PR (i guess to [EnviroDIY/TinyGSM](https://github.com/EnviroDIY/TinyGSM) because i based it on 5360 work). For now here is [MQTTclient](https://pastebin.com/DBzZEGk4) log, that make me sad...

shajek, 2019-07-25:
[success with MQTT ](https://pastebin.com/azBuGxTu)!!!! 

but it is little trickie

```
  bool modemGetConnected(uint8_t mux) {
    // Read the status of all sockets at once
    sendAT(GF("+CIPCLOSE?"));
    if (waitResponse(GF("+CIPCLOSE:")) != 1)	
      //return false;  // THIS IS THE PROBLEM
    for (int muxNo = 0; muxNo <= TINY_GSM_MUX_COUNT; muxNo++) {
      // +CIPCLOSE:<link0_state>,<link1_state>,...,<link9_state>
      sockets[muxNo]->sock_connected = stream.parseInt();
	  streamSkipUntil(',');
    }
    waitResponse();  // Should be an OK at the end
    return sockets[mux]->sock_connected;
  }
```

as you see, i have to comment that return, because it cause crash of ESP, but i dont understand why. Because normal AT command and response look this:

```
AT+CIPCLOSE?
+CIPCLOSE: 1, 0, 0, 0, 0, 0, 0, 0, 0, 0
OK
```

if i commented out that return, it is all right, but when i try to debug that if with DBG, it crash again

PS: where i have to make PR ? because my work is based on SIM5360, which is not included here

SRGDamia1, 2019-07-25:
Make the PR over at EnviroDIY. I'm sorry, I'm vacationing right now and won't get to look through it until next week.

SRGDamia1, 2019-08-01:
@shajek Can you see if it's working for you as it stands?

haimiko, 2019-08-01:
It works but cannot maintain a MQTT connection after the first publish.  I get a 

```` +CIPEVENT: NETWORK CLOSED UNEXPECTEDLY ````

```` 
OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,1,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,1
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0

OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,1,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,1
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0

OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,1,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,1

+CIPEVENT: NETWORK CLOSED UNEXPECTEDLY
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0

OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,1
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0

OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK

SMS DONE
````

haimiko, 2019-08-02:
I think i fixed it buy modifying modemGetConected.  waitresponse just needed a timeout and  a GSM_NL needed to be added before the CIPCLOSE response.

````
 bool modemGetConnected(uint8_t mux) {
	  // Read the status of all sockets at once
	  sendAT(GF("+CIPCLOSE?"));
	  if (waitResponse(10000L, GF(GSM_NL "+CIPCLOSE:")) != 1)
			 return false;  // THIS IS THE PROBLEM
	  for (int muxNo = 0; muxNo <= TINY_GSM_MUX_COUNT; muxNo++) {
		// +CIPCLOSE:<link0_state>,<link1_state>,...,<link9_state>
		sockets[muxNo]->sock_connected = stream.parseInt();
		streamSkipUntil(',');
	  }
	  waitResponse();  // Should be an OK at the end
	  return sockets[mux]->sock_connected;
  }

````

haimiko, 2019-08-02:
Nope happened again.  Odd thing is it happens right after I publish a message

```
OK
TOPIC: TestMessage00/test_test001/TESTM 1234567777
AT+CIPSEND=1,36
AT+CIPSEND=1,36
>0,0,AT+CIPRXGET=4,1
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0

OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,1,0,0,0,0,0,0,0,0

OK

+CIPEVENT: NETWORK CLOSED UNEXPECTEDLY

SMS DONE
AT+CIPRXGET=4,1
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0

OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,1
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0

OK
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
````

shajek, 2019-08-02:
sorry! I was on vacation ... I am on the way to make PR to EnviroDIY repo.

shajek, 2019-08-02:
[PR is created](https://github.com/EnviroDIY/TinyGSM/pull/25)
As i write , little tweak is needed in handling errors/unexpected response in  modemGetConnected() in waiting for response after CIPCLOSE. But basically MQTT work quite stable

SRGDamia1, 2019-08-02:
Can you check one more time to make sure it's working and close this if it is?

haimiko, 2019-08-02:
So you're not seeing "+CIPEVENT: NETWORK CLOSED UNEXPECTEDLY"?    What broker are you using?  I'm using EMQX.

haimiko, 2019-08-02:
> [success with MQTT ](https://pastebin.com/azBuGxTu)!!!!
>
> but it is little trickie
>
> ```
>   bool modemGetConnected(uint8_t mux) {
>     // Read the status of all sockets at once [...]

@shajek 
BTW. You can't  just comment that return.  If you do then the "for" loop is never run and function becomes basically disabled.   The problem with that line is that there is no ```` GSM_NL ``` before the ``` +CPCLOSE: ``` in the waitresponse statement.   Can you check again with my updated code above?

shajek, 2019-08-05:
here is log when i use your code for [modemGetConnected](https://pastebin.com/6J1idbMG)

haimiko, 2019-08-08:
Ok turned out being a faulty modem.  I got a new one and that one is indeed working fine.  Now the only problem is pubsubclient.loop() is takes a few seconds to complete.  I don't have this issue with the SIM800 module.  Must be some waitresponse timing out in the SIM7600 code.

During a pubsubclient.loop()
I narrowed it down to a waitresponse somewhere.  If i time modemGetConnected function it takes 2s to run
```
modemGetConnected took 2012s
modemGetConnected took 2013s
modemGetConnected took 2012s
modemGetConnected took 2013s
```
removed  ``` waitResponse();  // Should be an OK at the end ``` line in modemGetConnected
the time is cut down to 1s with no ill effects.
```
modemGetConnected took 1013s
modemGetConnected took 1014s
modemGetConnected took 1013s
```

I then added a timer to waitResponse and saw that waitResponse is taking 8-10s to complete during the pubsubclient.loop()

```
OK
AT+CIPRXGET=4,0
AT+CIPRXGET=4,0
+CIPRXGET:waitResponse took 10s
 4,0,0

OK
waitResponse took 2s
AT+CIPCLOSE?
AT+CIPCLOSE?
+CIPCLOSE:waitResponse took 10s
 1,0,0,0,0,0,0,0,0,0

OK

```

Thoughts?

shajek, 2019-08-09:
please try this

` bool modemGetConnected(uint8_t mux) {
	  // Read the status of all sockets at once
    sendAT(GF("+CIPCLOSE?"));
	uint8_t res = waitResponse(GF("+CIPCLOSE:"));
    if (res != 1)
      return false;
  
	  for (int muxNo = 0; muxNo <= mux; muxNo++) {
		// +CIPCLOSE:<link0_state>,<link1_state>,...,<link9_state>
		//DBG("### SOM VO FORE OF MODEMCONNECT:", muxNo);
		sockets[muxNo]->sock_connected = stream.parseInt();
		streamSkipUntil(',');
	  }
	  waitResponse();  // Should be an OK at the end
	  return sockets[mux]->sock_connected;
  }
`

i am for now working with this stable as rock

haimiko, 2019-08-09:
That would theoretically work, however, you still need to check that mux is <= TINY_GSM_MUX_COUNT -1 or you will get an error if you say pass mux > 9

SRGDamia1, 2020-04-15:
I think the 7600 is now functional.  Re-open thisif it is not working.

closed 2020-04-15 by SRGDamia1

PatentLobster, 2021-10-10:
send SMS is broken with the 7600

SRGDamia1, 2021-11-24:
@PatentLobster - if you are still having issues, can you clarify what they are?