vshymanskyy/TinyGSM #310: TinyGSM disconnects from MQTT broker when it wants to make a GET request

Black3rror, 2019-07-16:
TinyGSM version: 0.7.9
Hardware: esp8266 using SIM800L

### Scenario, steps to reproduce
I want to be connected to my MQTT broker and in the same time having some HTTP GET requests.

### Expected result
no corruption in MQTT connection while performing GET request

### Actual result
when the esp8266 wants to execute the `http.get(resource);` line, I can see the _Offline_ message on the _Will_Topic_. so its important to note that it gets offline immediately after GET request and it doesn't depend on the _KeepAlive_ time

### esp8266 code
its combination of _MqttClient_ example and _HttpClient_ example
credentials are starred

```
#define TINY_GSM_MODEM_SIM800

#include <TinyGsmClient.h>
#include <PubSubClient.h>
#include <ArduinoHttpClient.h>

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// or Software Serial on Uno, Nano
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(4, 5); // RX, TX

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "******";
const char user[] = "";
const char pass[] = "";

// MQTT details
const char* broker = "******";

const char* topicLed = "GsmClientTest/led";
const char* topicInit = "GsmClientTest/init";
const char* topicLedStatus = "GsmClientTest/ledStatus";

// Server details
const char server[] = "******";
const char resource[] = "******";
const int  port = 80;

TinyGsm modem(SerialAT);
TinyGsmClient client(modem);
PubSubClient mqtt(client);
HttpClient http(client, server, port);

#define LED_PIN LED_BUILTIN
int ledStatus = LOW;

long lastReconnectAttempt = 0;

void setup() {
  pinMode(LED_PIN, OUTPUT);

  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
    while (true);
  }
  SerialMon.println(" OK");

  SerialMon.print("Connecting to ");
  SerialMon.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    SerialMon.println(" fail");
    while (true);
  }
  SerialMon.println(" OK");

  // MQTT Broker setup
  mqtt.setServer(broker, 1883);
  mqtt.setCallback(mqttCallback);

  mqttConnect();

  
  SerialMon.print(F("Performing HTTP GET request... "));
  int err = http.get(resource);
  if (err != 0) {
    SerialMon.println(F("failed to connect"));
    delay(10000);
    return;
  }

  int status = http.responseStatusCode();
  SerialMon.println(status);
  if (!status) {
    delay(10000);
    return;
  }

  while (http.headerAvailable()) {
    String headerName = http.readHeaderName();
    String headerValue = http.readHeaderValue();
    //SerialMon.println(headerName + " : " + headerValue);
  }

  int length = http.contentLength();
  if (length >= 0) {
    SerialMon.print(F("Content length is: "));
    SerialMon.println(length);
  }
  if (http.isResponseChunked()) {
    SerialMon.println(F("The response is chunked"));
  }

  String body = http.responseBody();
  SerialMon.println(F("Response:"));
  SerialMon.println(body);

  SerialMon.print(F("Body length is: "));
  SerialMon.println(body.length());

  // Shutdown

  http.stop();
  SerialMon.println(F("Server disconnected"));
}

boolean mqttConnect() {
  SerialMon.print("Connecting to ");
  SerialMon.print(broker);

  // Connect to MQTT Broker
  boolean status = mqtt.connect("GsmClientTest", "", "", topicInit, 2, false, "Offline");

  if (status == false) {
    SerialMon.println(" fail");
    return false;
  }
  SerialMon.println(" OK");
  mqtt.publish(topicInit, "GsmClientTest started");
  mqtt.subscribe(topicLed);
  return mqtt.connected();
}

void loop() {

  if (!mqtt.connected()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
    // Reconnect every 10 seconds
    unsigned long t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
    delay(100);
    return;
  }

  mqtt.loop();
}

void mqttCallback(char* topic, byte* payload, unsigned int len) {
  SerialMon.print("Message arrived [");
  SerialMon.print(topic);
  SerialMon.print("]: ");
  SerialMon.write(payload, len);
  SerialMon.println();

  // Only proceed if incoming message's topic matches
  if (String(topic) == topicLed) {
    ledStatus = !ledStatus;
    digitalWrite(LED_PIN, ledStatus);
    mqtt.publish(topicLedStatus, ledStatus ? "1" : "0");
  }
}
```

SRGDamia1, 2019-07-16:
Are your MQTT broker and your GET server the same?  That is, is the same TCP connection being used for both or are you trying to use multiple TCP connections?  The ESP8266 is capable of multiple simultaneous TCP connections, but to use more than one you must create more than one GSMClient instance.

If you are using the same server/TCP connection for both, you must tell the http client to keep the connection alive.  (Put ```http.connectionKeepAlive();``` in your setup _before_ ```mqttConnect();```.)

If connectionKeepAlive is not set, the HttpClient library (not this library) will attempt to create a new connection regardless of the current connection state (it won't even check).  When the HttpClient library passes off the job of creating the new connection to this library, the first thing this library (TinyGSM) does before creating a new connection on a client is to stop any connection on that client.  That causes the behavior you're seeing - the MQTT connection is broken.

Black3rror, 2019-07-16:
thank u @SRGDamia1 for your quick answering.
my HTTP server is different from MQTT server. and as u can see in the code, I was using just one instance of TinyGsmClient so I changed the code to use two instances:

```
TinyGsm modem(SerialAT);
TinyGsm modem_2(SerialAT);
TinyGsmClient client(modem);
TinyGsmClient client_2(modem_2);
PubSubClient mqtt(client);
HttpClient http(client_2, server, port);
```

but nothing changed

also in the mosquitto.log file I can see the below lines which I thought may help:

```
1563309694: New client connected from 5.126.161.54 as GsmClientTest (p2, c1, k15).
1563309695: Socket error on client GsmClientTest, disconnecting.
```

SRGDamia1, 2019-07-16:
When you create the two different clients, they must have different "mux" numbers:

```cpp
TinyGsm modem(SerialAT);
// TinyGsm modem_2(SerialAT);  // <- not needed
TinyGsmClient client(modem, 1);  // "default" mux is 1
TinyGsmClient client_2(modem, 2);  // must have a different mux number
PubSubClient mqtt(client);
HttpClient http(client_2, server, port);
```

If that doesn't work, could you post your AT log?

Black3rror, 2019-07-16:
Thank u @SRGDamia1. you solved the problem

closed 2019-07-16 by Black3rror

vandanoob, 2024-01-10:
i am facing same issue but with http and mqtt 
WiFiClient client1;
WiFiClient espClient;
WiFiClient httpClient;
PubSubClient client(espClient);

///////////////////////////////////////////
void send_request(char* webst)
              { 
               HTTPClient http;
               WiFiClient httpWiFiClient;
              //http.setTimeout(1000); //in ms
              
              
              http.begin(httpWiFiClient,webst);
              int httpCode = http.GET();
              http.end();
              Serial.println(httpCode);
            //  Serial.println("===========");
            //  Serial.println(webst);
              
              
              //client.publish(pub_token, webstr);
         }