vshymanskyy/TinyGSM #312: Arduino MKR GSM worked fine initially, now it stopped working

maxkraft7, 2019-07-31:
I tried using a Arduino MKR GSM 1500 with you library. When I compiled it for the first time it even worked without an additional battery and just the antenna wich really suprised me. But now for a while if I use the stnadard library it's not working because at some point the energy from the battey I assume is too low and it just stops producing AT-commands. When I try your code (with the U-BLOX enabled) it compiles but as soon as it's uploaded the serial-port in my Arduino IDE dusappears. I then have to press the double-reset button twice, uplaod a blink example and everything's working again. Do you have any idea what could be the reason for this? I really only made minor changes (set the APN and the Modem-Type).

Furthermore I'm using a 7.4 Volt Battery with 5000mAh wich I run through a UBEC 5V step-down converter to get 5 Volts ([this ](https://www.amazon.com/Ship-Hobbywing-Switch-mode-UBEC-Lowest/dp/B008ZNWOYY/ref=pd_lpo_sbs_147_t_0?_encoding=UTF8&psc=1&refRID=T2C0SAP5C5TPCSY11KQV) one). The antenna I'm using is [this ](https://www.amazon.de/gp/product/B07PJ6HYWT/ref=ox_sc_act_title_1?smid=A1TCLMQH7PKTWY&psc=1) one here

````

// Select your modem:

 #define TINY_GSM_MODEM_UBLOX

// Increase RX buffer to capture the entire response
// Chips without internal buffering (A6/A7, ESP8266, M590)
// need enough space in the buffer for the entire response
// else data will be lost (and the http library will fail).
#define TINY_GSM_RX_BUFFER 650

// See all AT commands, if wanted
//#define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
//#define TINY_GSM_DEBUG Serial
//#define LOGGING  // <- Logging is for the HTTP library

// Add a reception delay, if needed
//#define TINY_GSM_YIELD() { delay(1); }

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX

#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "hologram";
const char gprsUser[] = "";
const char gprsPass[] = "";
const char wifiSSID[]  = "YourSSID";
const char wifiPass[] = "SSIDpw";

// Server details
const char server[] = "vsh.pp.ua";
const char resource[] = "/TinyGSM/logo.txt";
const int  port = 80;

#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

TinyGsmClient client(modem);
HttpClient http(client, server, port);

void setup() {
  // Set console baud rate
  SerialMon.begin(115200);
  delay(10);

  // Set your reset, enable, power pins here
  pinMode(20, OUTPUT);
  digitalWrite(20, HIGH);

  pinMode(23, OUTPUT);
  digitalWrite(23, HIGH);

  SerialMon.println("Wait...");

  // Set GSM module baud rate
  SerialAT.begin(115200);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  modem.restart();
  // modem.init();

  String modemInfo = modem.getModemInfo();
  SerialMon.print("Modem: ");
  SerialMon.println(modemInfo);

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");
}

void loop() {

#if defined TINY_GSM_USE_WIFI && defined TINY_GSM_MODEM_HAS_WIFI
  SerialMon.print(F("Setting SSID/password..."));
  if (!modem.networkConnect(wifiSSID, wifiPass)) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");
#endif

#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_XBEE
  // The XBee must run the gprsConnect function BEFORE waiting for network!
  modem.gprsConnect(apn, gprsUser, gprsPass);
#endif

  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" OK");

  if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }

#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_HAS_GPRS
    SerialMon.print(F("Connecting to "));
    SerialMon.print(apn);
    if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
      SerialMon.println(" fail");
      delay(10000);
      return;
    }
    SerialMon.println(" OK");
#endif

  SerialMon.print(F("Performing HTTP GET request... "));
  int err = http.get(resource);
  if (err != 0) {
    SerialMon.println(F("failed to connect"));
    delay(10000);
    return;
  }

  int status = http.responseStatusCode();
  SerialMon.print(F("Response status code: "));
  SerialMon.println(status);
  if (!status) {
    delay(10000);
    return;
  }

  SerialMon.println(F("Response Headers:"));
  while (http.headerAvailable()) {
    String headerName = http.readHeaderName();
    String headerValue = http.readHeaderValue();
    SerialMon.println("    " + headerName + " : " + headerValue);
  }

  int length = http.contentLength();
  if (length >= 0) {
    SerialMon.print(F("Content length is: "));
    SerialMon.println(length);
  }
  if (http.isResponseChunked()) {
    SerialMon.println(F("The response is chunked"));
  }

  String body = http.responseBody();
  SerialMon.println(F("Response:"));
  SerialMon.println(body);

  SerialMon.print(F("Body length is: "));
  SerialMon.println(body.length());

  // Shutdown

  http.stop();
  SerialMon.println(F("Server disconnected"));

#if TINY_GSM_USE_WIFI
    modem.networkDisconnect();
    SerialMon.println(F("WiFi disconnected"));
#endif
#if TINY_GSM_USE_GPRS
    modem.gprsDisconnect();
    SerialMon.println(F("GPRS disconnected"));
#endif

  // Do nothing forevermore
  while (true) {
    delay(1000);
  }
}

```

SRGDamia1, 2019-07-31:
Does it still work with the MKR library with the antenna and battery? I'm not familiar with the board or library. Check to make sure that you have the right Serials selected. Also make sure that you have the right pins set high/low as needed to wake the modem.

Without a separate battery, the whole board *will brown out* (and stop working) when the modem attempts to connect to the network. The first time you set up with a new sim/network it may not connect automatically or it might take a very long time to do so; is usually faster after that.

SRGDamia1, 2020-04-15:
I'm going to close this issue.  Please re-open it if you're still having issues.

closed 2020-04-15 by SRGDamia1