vshymanskyy/TinyGSM #313: I'm working to add Simcom SIM5320 Support

wolfen351, 2019-08-01:
I'm intending to add support for the SIM5320 module and submit a PR to this library to add it. Are there any existing attempts that I can start with? If not are there any recent additions of new boards, perhaps I can use one of those as a template for adding the new board.. 

As requested, answers to the questions below:

## What type of issues is this?

[X] Request to support a new module
    - Please, consider forking and submitting a pull request!
[ ] Bug or problem compiling the library
    - NOTE:  If you are using an ESP32, you must will not be able to compile the HttpClient or HttpsClient examples if you are using ESP32 core >1.0.1.  Downgrade to version 1.0.1 or use the WebClient example.  Please comment on the issue on the ESP32 core, not in this library:  https://github.com/espressif/arduino-esp32/issues/2755
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Main processor board: WeMOS D1 Mini ESP8266 based board
Modem: SIMCOM SIM5230
TinyGSM version: Latest
Code: Just starting now

### Scenario, steps to reproduce
I'd like to add support for this module, as I bought one, and it will be useful for all. 

### Expected result
Code merged into the library

### Actual result
None so far

### AT command log
n/a

SRGDamia1, 2019-08-01:
Look over at the EnviroDIY fork:  https://github.com/EnviroDIY/TinyGSM.  There's not-quite-finished support for the 5360 which uses nearly identical AT commands to the 5230 (and 71xx).

Barnik12, 2019-08-05:
Hey for "http client" file have you added sim5320? I couldn't find it, Or do you know of any library where GSM/IoT works fine for SIM5320E?

SRGDamia1, 2019-08-05:
The 5320 and 5360 are almost identical; use it like the 5360.  

If you use ```#define TINY_GSM_MODEM_SIM5320``` it will call up the 5360.

Barnik12, 2019-08-06:
#define TINY_GSM_MODEM_SIM536 is giving the following error. Its same for #define TINY_GSM_MODEM_SIM5320.

 ```
C:\Users\barni\Documents\Arduino\libraries\TinyGSM-master\src/TinyGsmClientSIM5360.h: In member function 'sendAT.constprop':

C:\Users\barni\Documents\Arduino\libraries\TinyGSM-master\src/TinyGsmClientSIM5360.h:750:1: internal compiler error: Segmentation fault

 TINY_GSM_MODEM_STREAM_UTILITIES()

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Users\barni\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/barni/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

```

SRGDamia1, 2019-08-06:
Well that's a new error for me.

You're using the Arduino.cc IDE?  I think the fault might be in the IDE, not here:  https://forum.arduino.cc/index.php?topic=583509.0

wolfen351, 2019-08-07:
I've used the 5360 as a base as suggested, and I've made some progress.. the system gets to the point where it needs to open a TCP connection and then fails. The basics are there, the reboot commands etc

SRGDamia1, 2019-08-07:
@wolfen351 Logs?

SRGDamia1, 2020-04-15:
I think we have people successfully using the 5320, so I'm going to close this.

closed 2020-04-15 by SRGDamia1